lvgl/lvgl #7008: empty screen have flicker when set optimization from -O0 to -O3

woniulive, 2024-10-08:
### LVGL version

v9.2.0

### What happened?

I use STM32L4P5RG + external QSPI flash + external QSPI PSRAM + SPI DMA to driver TFT.
When I use -O0, the benchmark demo work correctly.
When I use -O3, empty screen have flicker when the benchmark demo run.

-O3:
[image]
https://github.com/user-attachments/assets/c4104dfe-01c4-4fb6-8ff1-f59d9b177d20

kisvegabor, 2024-10-14:
Hi,

What is color format of the display? Is it 24 bit (RGB888)?

woniulive, 2024-10-16:
Hi @kisvegabor :
I use RGB565.

kisvegabor, 2024-10-28:
Sorry for the late reply.  It seems only the screen sized fills are problematic. Maybe it's a RAM config issue. Without using LVGL, please try to fill the RAM 16 bit numbers increasing from 0 and ready back the result.

kissa96, 2024-10-28:
O3 optimization can result in unpredictable behavior if the screen is not properly handled with variables. Or better said, O3 needs code which is 100% standards compliant.

I am guessing you are never accessing the blank screen part, so it gets optimized out and some garbage is displayed instead

kisvegabor, 2024-10-29:
I use O3 all the time so I can say that LVGL itself is O3 compatible.

woniulive, 2024-11-04:
@kisvegabor @kissa96 :
After I confirmed with ST official, it seems that the reason is PSRAM, the data store in PSRAM will be wrong if I set optimization from -O0 to -O3. After I decrease the frequecy of PSRAM, it work correctly.
Thank you for your help.
This issue can be closed now.

kisvegabor, 2024-11-13:
Oh, interesting. Thank you for letting us know.

closed 2024-11-13 by kisvegabor