lvgl/lvgl #7009: Button Selected State Cleared on Key Indev Enter Pressing in lv_keyboard Widget

littepointR, 2024-10-08:
### LVGL version

v9.2.0

### What happened?

When I use a key input device (key indev) to control the lv_keyboard widget, pressing the Enter key once (without releasing it) on a selected button causes the button's selected state to be cleared. After inspecting the code in lv_buttonmatrix, I found that the selected state is cleared in the handling of the LV_EVENT_PRESSING event (lines 457 to 468). Since I am using a key indev, the get_button_from_point function always returns -1. After adding a check for the indev type, the issue was resolved. Is this the correct approach, and is this considered a bug?
[image]

### How to reproduce?

```
// lv_buttonmatrix.c line 457
// befor:
    else if(code == LV_EVENT_PRESSING) {
        /*If a slid to a new button, discard the current button and don't press any buttons*/
        if(btnm->btn_id_sel != LV_BUTTONMATRIX_BUTTON_NONE) {
            lv_indev_t * indev = lv_event_get_indev(e);
            lv_indev_get_point(indev, &p);
            uint32_t btn_pr = get_button_from_point(obj, &p);
            if(btn_pr != btnm->btn_id_sel) {
                invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the old area*/
                btnm->btn_id_sel = LV_BUTTONMATRIX_BUTTON_NONE;
            }
        }
    }
// after:
    else if(code == LV_EVENT_PRESSING) {
        /*If a slid to a new button, discard the current button and don't press any buttons*/
        if(btnm->btn_id_sel != LV_BUTTONMATRIX_BUTTON_NONE) {
            lv_indev_t * indev = lv_event_get_indev(e);
            lv_indev_type_t indev_type = lv_indev_get_type(indev);
            if(indev_type == LV_INDEV_TYPE_POINTER || indev_type == LV_INDEV_TYPE_BUTTON) {
                lv_indev_get_point(indev, &p);
                uint32_t btn_pr = get_button_from_point(obj, &p);
                if(btn_pr != btnm->btn_id_sel) {
                    invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the old area*/
                    btnm->btn_id_sel = LV_BUTTONMATRIX_BUTTON_NONE;
                }
            }
        }
    }
```

kisvegabor, 2024-10-14:
Hi,

I think it's a typo and instead of invalidation we should save `btn_pr`
```c
    else if(code == LV_EVENT_PRESSING) {
        /*If a slid to a new button, discard the current button and don't press any buttons*/
        if(btnm->btn_id_sel != LV_BUTTONMATRIX_BUTTON_NONE) {
            lv_indev_t * indev = lv_event_get_indev(e);
            lv_indev_get_point(indev, &p);
            uint32_t btn_pr = get_button_from_point(obj, &p);
            if(btn_pr != btnm->btn_id_sel) {
                invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the old area*/
                btnm->btn_id_sel = btn_pr;   /*CHANGED HERE*/
            }
        }
    }
```

Could you try it?

RamDav, 2024-10-16:
I think i Made a duplicated of the Same Bug #7074

kisvegabor, 2024-10-28:
Let's continue in https://github.com/lvgl/lvgl/issues/7074

closed 2024-10-28 by kisvegabor

RamDav, 2024-11-07:
should be resolved with #7125