lvgl/lvgl #7012: Better Gridnav use Elliptical distance instead of Circular distance

TCRnext, 2024-10-08:
### Introduce the problem

In Gridnav function ,we use circular distance to decide which object to jump into.However in the case ,between `FIND_LEFT`  and `FIND_TOP`  , or the situation between  `FIND_LEFT`  and `FIND_TOP` ,it may lack directional selectivity.

### Proposal

So ,I think using Elliptical distance instead of Circular distance may be a good idea.

By adding two parameter, `x_factor` and  `y_factor`,for example when `FIND_LEFT` we can have more weight find a object that close to the former object in x_anis,while have more weight find a object that close to the former object in y_anis when `FIND_TOP`

This is the original code at [https://github.com/lvgl/lvgl/blob/8d47460580c6e17ff800aaac2dbbb11c3520cc44/src/others/gridnav/lv_gridnav.c#L301](url)
```c
    for(i = 0; i < child_cnt; i++) {
        lv_obj_t * child = lv_obj_get_child(obj, i);
        if(child == start_child) continue;
        if(obj_is_focusable(child) == false) continue;

        int32_t x_err = 0;
        int32_t y_err = 0;
        switch(mode) {
            case FIND_LEFT:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(x_err >= 0) continue;    /*It's on the right*/
                if(LV_ABS(y_err) > h_half) continue;    /*Too far*/
                break;
            case FIND_RIGHT:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(x_err <= 0) continue;    /*It's on the left*/
                if(LV_ABS(y_err) > h_half) continue;    /*Too far*/
                break;
            case FIND_TOP:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(y_err >= 0) continue;    /*It's on the bottom*/
                break;
            case FIND_BOTTOM:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(y_err <= 0) continue;    /*It's on the top*/
                break;
            case FIND_NEXT_ROW_FIRST_ITEM:
                y_err = get_y_center(child) - y_start;
                if(y_err <= 0) continue;    /*It's on the top*/
                x_err = lv_obj_get_x(child);
                break;
            case FIND_PREV_ROW_LAST_ITEM:
                y_err = get_y_center(child) - y_start;
                if(y_err >= 0) continue;    /*It's on the bottom*/
                x_err = obj->coords.x2 - child->coords.x2;
                break;
            case FIND_FIRST_ROW:
                x_err = get_x_center(child) - x_start;
                y_err = lv_obj_get_y(child);
                break;
            case FIND_LAST_ROW:
                x_err = get_x_center(child) - x_start;
                y_err = h_max - lv_obj_get_y(child);
        }

        if(guess == NULL ||
           (y_err * y_err + x_err * x_err < y_err_guess * y_err_guess + x_err_guess * x_err_guess)) {
            guess = child;
            x_err_guess  = x_err;
            y_err_guess  = y_err;
        }
    }
``` 

And this is my better version of gridnav:
(in lv_conf.h)
```c
/** 1: Enable grid navigation */
#define LV_USE_GRIDNAV 0
#if LV_USE_GRIDNAV 
    #define GRIDNAV_X_FACTOR 1 //default to 1 to keep the original function,can set to 1,4,9,16 or any number to get different directional selectivity
    #define GRIDNAV_Y_FACTOR 1 //default to 1 to keep the original function,
#endif
```
(in gridnav.c)
```c
    int32_t x_factor = 1;
    int32_t y_factor = 1;
    for(i = 0; i < child_cnt; i++) {
        lv_obj_t * child = lv_obj_get_child(obj, i);
        if(child == start_child) continue;
        if(obj_is_focusable(child) == false) continue;

        int32_t x_err = 0;
        int32_t y_err = 0;
        
        switch(mode) {
            case FIND_LEFT:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(x_err >= 0) continue;    /*It's on the right*/
                y_factor = GRIDNAV_Y_FACTOR ;
                //if(LV_ABS(y_err) > h_half) continue;    /*Too far,now use Elliptical distance instead*/
                break;
            case FIND_RIGHT:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(x_err <= 0) continue;    /*It's on the left*/
                //if(LV_ABS(y_err) > h_half) continue;    /*Too far,now use Elliptical distance instead */
                y_factor = GRIDNAV_Y_FACTOR ;
                break;
            case FIND_TOP:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(y_err >= 0) continue;    /*It's on the bottom*/
                x_factor = GRIDNAV_X_FACTOR ;
                break;
            case FIND_BOTTOM:
                x_err = get_x_center(child) - x_start;
                y_err = get_y_center(child) - y_start;
                if(y_err <= 0) continue;    /*It's on the top*/
                x_factor = GRIDNAV_X_FACTOR ;
                break;
            case FIND_NEXT_ROW_FIRST_ITEM:
                y_err = get_y_center(child) - y_start;
                if(y_err <= 0) continue;    /*It's on the top*/
                x_err = lv_obj_get_x(child);
                break;
            case FIND_PREV_ROW_LAST_ITEM:
                y_err = get_y_center(child) - y_start;
                if(y_err >= 0) continue;    /*It's on the bottom*/
                x_err = obj->coords.x2 - child->coords.x2;
                break;
            case FIND_FIRST_ROW:
                x_err = get_x_center(child) - x_start;
                y_err = lv_obj_get_y(child);
                break;
            case FIND_LAST_ROW:
                x_err = get_x_center(child) - x_start;
                y_err = h_max - lv_obj_get_y(child);
        }

        if(guess == NULL ||
           (y_err * y_err*y_factor  + x_err * x_err*x_factor  < y_err_guess * y_err_guess*y_factor  + x_err_guess * x_err_guess*x_factor )) {
            guess = child;
            x_err_guess  = x_err;
            y_err_guess  = y_err;
        }
    }
```

kisvegabor, 2024-10-09:
Hi,

 could you record a screen cast to demonstrate the difference?

TCRnext, 2024-10-09:
This is my version of gridnav,using the x,y factor of 4 = (2*2):
https://github.com/user-attachments/assets/29f91d82-31a4-4908-a6d3-8ce1dc3ed6b2

This is the orgin version of gridnav:

https://github.com/user-attachments/assets/d0580c27-d839-4240-a909-d27a89e4c035

TCRnext, 2024-10-09:
> Hi,
> 
> could you record a screen cast to demonstrate the difference?

I also have a image to explain it:

[image]

kissa96, 2024-10-28:
Hi. How much of a performance increase does it give us? Sounds useful, but since the current solution works fine, it would only make sense if significant increases are expected

liamHowatt, 2024-10-28:
> How much of a performance increase does it give us?

I think it's an improvement on accuracy rather than performance. Is that right?

kisvegabor, 2024-10-29:
@TCRnext I've looking into it in more detail again, and now I better see what the goal is. However, I don't think this the best solution to the problem:
- Different girdnavs might require different parameters (so factors can't be compile time settings but should be specific to a given gridnav)
- Even in a single gridnav we can have elements where the "jump" is more preferred. But it's an N:M connection. (E.g. button1 likes button2 and button3; button3 likes button8 and button10, etc). But it's hard to describe.

All things considered  I think we shouldn't add this feature, but wait for more flexible proposal instead.

Anyway, thank you for taking time to open the PR and explaining your idea! :100:

closed 2024-10-29 by kisvegabor

TCRnext, 2024-10-29:
@kisvegabor Thank you for taking time to consider my idea.

For the second idea, I think there is a ways to descibe:
Use a function to set manual navigation infomation:
`lv_gridnav_set_focused_next(lv_obj_t * cont, lv_obj_t * to_focus,lv_obj_t * to_focus_next,_lv_key_t key );`
In gridnav function handle function,we first check if the focused object has a manual navigation infomation,and the key input is match , if so ,it will focus the object we set.
The manual navigation infomation can be store as a linklist struct into `lv_gridnav_dsc_t`.

We can also store the direction selection factor into  `lv_gridnav_dsc_t`.So,we can give different parameters to different gridnavs.

kisvegabor, 2024-10-30:
> lv_gridnav_set_focused_next(lv_obj_t * cont, lv_obj_t * to_focus,lv_obj_t * to_focus_next,_lv_key_t key );

It sounds interesting. My only concern is when any of the referenced object are deleted they will leave a wild pointer behind. We can attache `LV_EVENT_DETETE` event to all the referenced widget but it starts to be a little too complicated. 

What about sending an event to the currently focused widget, telling the current key, and from the event we can get back a preferred widget if any. If there is a preferred widget use it, otherwise use the current  (non-elliptic) method.

What do you think?

TCRnext, 2024-11-01:
> It sounds interesting. My only concern is when any of the referenced object are deleted they will leave a wild pointer behind. We can attache `LV_EVENT_DETETE` event to all the referenced widget but [...]

In gridnav's event_cb we can detect `LV_EVENT_CHILD_DELETED` event and compare the deleted object with the manual navigation infomation,if any informations' `lv_obj_t * to_focus` or `lv_obj_t * to_focus_next` is deleted then remove the deleted objects .

TCRnext, 2024-11-02:
> It sounds interesting. My only concern is when any of the referenced object are deleted they will leave a wild pointer behind. We can attache `LV_EVENT_DETETE` event to all the referenced widget but [...]

@kisvegabor I've another idea now.

Instead of using function.
> lv_gridnav_set_focused_next(lv_obj_t * cont, lv_obj_t * to_focus,lv_obj_t * to_focus_next,_lv_key_t key );

We can have a function to manually set the "logic position" of the object.
Using` lv_gridnav_set_focused_obj_logic_pos(lv_obj_t * cont, lv_obj_t * to_focus,uint32_t logic_x,uint32 logic_y);

For example ,obj1 has logic pos (5,5) if press right arrow key it will first search a obj has the logic position(6,5),if it fail to search ,it will use traditional ways to search the next object.

TCRnext, 2024-11-02:
[image]
To better suit the situation above ,may be two logic position is need .
So maybe` lv_gridnav_set_focused_obj_logic_pos(lv_obj_t * cont, lv_obj_t * to_focus,uint32_t logic_x1,uint32 logic_y1,uint32_t logic_x2,uint32 logic_y2);`is better.

kisvegabor, 2024-11-06:
Where would we store the logical positions of each widget?