lvgl/lvgl #7014: Custom Font Code Doesn't Compile

dana2048, 2024-10-09:
### LVGL version

9.1.0

### What happened?

Compiler errors processing the font .c file in Arduino IDE. **error: field name not in record or union initializer** repeats multiple times. This happens with output from both the online and the local converter.

lv_font_conv 1.5.3 is installed locally
Arduino IDE 2.3.3

The errors begin partway into the `static const lv_font_fmt_txt_glyph_dsc_t glyph_dsc[] = {` section and repeat for every field name in every line.

### How to reproduce?

```c 11890 static const lv_font_fmt_txt_glyph_dsc_t glyph_dsc[] = {
11891 {.bitmap_index = 0, .adv_w = 0, .box_w = 0, .box_h = 0, .ofs_x = 0, .ofs_y = 0},

.
.
.

12013 {.bitmap_index = 58360, .adv_w = 672, .box_w = 42, .box_h = 10, .ofs_x = 0, .ofs_y = 13},
12014 {.bitmap_index = 58570, .adv_w = 864, .box_w = 54, .box_h = 36, .ofs_x = 0, .ofs_y = 0},
12015 {.bitmap_index = 59542, .adv_w = 960, .box_w = 60, .box_h = 48, .ofs_x = 0, .ofs_y = -6},

Error points to .ofs_y
montserrat_48.c:12015:56: note: (near initialization for 'glyph_dsc[124]')
montserrat_48.c:12015:69: error: field name not in record or union initializer
     {.bitmap_index = 58570, .adv_w = 864, .box_w = 54, .box_h = 36, .ofs_x = 0, .ofs_y = 0},

First error is reported at line 12014. Errors repeat, for the same line, pointing successively to .bitmap_index, .adv_w, .box_w, etc

Errors repeat for lines 12015 to 12048 which are the rest of the lines in the lv_font_fmt_txt_glyph_dsc_t glyph_dsc[] initializer. 

After that there are many more errors for the rest of the file.```

dana2048, 2024-10-13:
I did my initial test using the default character range 0x20 to 0x7E.

Since then I have tested with a smaller range (0x2F-0x3A) and this has worked with no problems.

kisvegabor, 2024-10-25:
Hi, 
I'm sorry for the delay. I have no idea what could be the problem. Can you add the generated font file?

closed 2024-11-24 by lvgl-bot