lvgl/lvgl #7015: Dave2D missing include headers

Imrekorf, 2024-10-09:
### LVGL version

v9.2.0

### What happened?

setting "LV_USE_DRAW_DAVE2D" to 1 causes around 450 errors to occur, caused by "incomplete type" errors.

It seems that the necessary private struct type definitions have not been included.
This can be fixed by including the following headers in lv_draw_dave2d.h after "the initial "lv_draw.h" include.
```c
#include "../../lv_draw_private.h"
#include "../../sw/lv_draw_sw.h"
#include "../../lv_draw_label_private.h"
#include "../../lv_image_decoder_private.h"
```

relevant build log snippet:
```
In file included from ../lib/lvgl-9.2.0/src/draw/renesas/dave2d/lv_draw_dave2d_arc.c:1:
../lib/lvgl-9.2.0/src/draw/renesas/dave2d/lv_draw_dave2d.h:33:20: error: field 'base_unit' has incomplete type
   33 |     lv_draw_unit_t base_unit;
      |                    ^~~~~~~~~
../lib/lvgl-9.2.0/src/draw/renesas/dave2d/lv_draw_dave2d.h:53:68: error: unknown type name 'lv_draw_image_dsc_t'
   53 | void lv_draw_dave2d_image(lv_draw_dave2d_unit_t * draw_unit, const lv_draw_image_dsc_t * draw_dsc,
      |                                                                    ^~~~~~~~~~~~~~~~~~~
....
```

### How to reproduce?

You can reproduce this issue by setting up the [ra8d1 lvgl](https://github.com/lvgl/lv_port_renesas_ek-ra8d1_gcc) project and setting the "LV_USE_DRAW_DAVE2D" to 1 in the lv_conf.h file. ( Project is build with [FSP version 5.3.0 in e2studio version 2024-04](https://github.com/renesas/fsp/releases/tag/v5.3.0)

kisvegabor, 2024-10-09:
Oh shoot, I fixed it locally but haven't sent a PR :facepalm:

Could you test it on your end too? #7016

Imrekorf, 2024-10-09:
The fix works on my end aswell, thanks for the quick reply!

PR #7016 (2024-10-12): fix(dave2d): fix dave2d private include issue
Fixes #7015

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12`  [... cut]
closed 2024-10-12 by FASTSHIFT