lvgl/lvgl #7026: 2bpp/4bpp font glyph format support

junminacts, 2024-10-10:
### Problem to solve

Both 1/2/4bpp font glyph formats are considered as A8 format during drawing in LVGL 9, we hope the renderer can honor the glyph format of glyph_dsc during label drawing. 

### Success criteria

When the returned format (also the real format) of glyph_dsc (lv_draw_glyph_dsc_t) is A1/2/4, the sw draw unit can draw it correctly.

### Solution outline

the sw draw unit support A1/2/4 glyph formats

### Rabbit holes

no

### Testing

Select 1 builtin font, like lv_font_montserrat_38, just return the original 4bpp bitmap in get_glyph_bitmap callback, then check the letter drawing is correct or not.

### Teaching

N/A

kisvegabor, 2024-10-10:
It was already requested by multiple vendors, so I agree that we should handle it somehow. IMO the main thing here is support blending A1, A2, A4 images. We can allocate time for that in the near future.

junminacts, 2024-10-14:
Yes, I want them handled separately, glyph 1 -> A1, 1 -> A2 and 4 -> A4.

kisvegabor, 2024-10-16:
I had some discussion about this topic with others as well. This is what I know so far:
1. Now the e.g. the A2 bitmaps are packed. It means when w=11, h=10, will require 110 x 2bits. However for the sake of simplicity it should be padded to byte boundary, resulting in w=11+5(padding) and h=10 = 160 * 2 bits.
2. To do this we need to update the [font converter](https://github.com/lvgl/lv_font_conv) and add a `padded = 1` or so flag in the font's description for backward compatibility. 
3. `lv_font_get_glyph_bitmap()` now converts all bitmaps to A8 and its result is already set in 
```c
lv_draw_buf_t * draw_buf = glyph_draw_dsc->glyph_data;
blend_dsc.mask_buf = draw_buf->data;
```
So the draw unit can't access the original non-A8 bitmap data. 
4. In light of that we should let the user call `lv_font_get_glyph_bitmap` in its `draw_letter_cb`.

junminacts, 2024-10-16:
font converter may also add a new option to generate byte boundary A2 bitmap to get better hardware which often require a byte stride.

kisvegabor, 2024-10-16:
Yes, that's what I meant by this
> To do this we need to update the [font converter](https://github.com/lvgl/lv_font_conv) and add a padded = 1 or so flag in the font's description for backward compatibility.

kisvegabor, 2024-10-31:
It's scheduled from me for next week.

kisvegabor, 2024-11-06:
PR opened: https://github.com/lvgl/lvgl/pull/7234

PR #7234 (2024-11-21): feat(font): allow using A1,2,4 bitmaps + handle byte aligned fonts
Fixes #7026 (Let the draw unit use the raw A1,2,4 bitmaps)

You can get the raw (non A8 converted) bitmaps like:
```c
      glyph_draw_dsc->g->req_raw_bitap = 1; //g is lv_font_glyph_dsc_t
      glyph_draw_dsc->glyph_data = lv_font_get_glyph_bitmap(glyph_draw_dsc->g, glyph_draw_dsc->_draw_buf);
```

To let the draw unit decide how to get the bitmap, `lv_font_get_glyph_bitmap` is moved to the draw units. 

Also requires https://github.com/lvgl/lv_font_conv/pull/127 to convert bytes aligned A1,2,4 fonts. 

cc @FASTSHIFT @junminac [... cut]
closed 2024-11-21 by liamHowatt