lvgl/lvgl #7027: ARGB8565 color format full support

junminacts, 2024-10-10:
### Problem to solve

SW draw unit donot support LV_COLOR_FORMAT_ARGB8565 in LVGL 9, which is useful to evaluate the memory usage in simulator and some special case (like recolor) unsupported by VG-Lite GPU. 

### Success criteria

SW render support LV_COLOR_FORMAT_ARGB8565 image and destination draw buffer.

### Solution outline

bring back LV_COLOR_FORMAT_ARGB8565 support

### Rabbit holes

No

### Testing

Testing LV_COLOR_FORMAT_ARGB8565 image with LV_COLOR_FORMAT_ARGB8565 display.

### Teaching

No spec doc

kisvegabor, 2024-10-10:
Why is `LV_COLOR_FORMAT_ARGB8565` needed as a destination format? What supports it?

junminacts, 2024-10-11:
VG-Lite can render LV_COLOR_FORMAT_ARGB8565 (RGB565A8 unsupported by vglite yet) dest buffer instead of ARGB8888 to save memory,  if transparent display or snapshot required. But some complex cases, like image color, which are not supported by VG-Lite, require sw rendering.

kisvegabor, 2024-10-16:
I see. I made a little bit of research and I'm ok with adding `ARGB8565` as both destination and source format. 
Have you already done some initial work on it? If so  can you send a PR?

junminacts, 2024-10-16:
I have just ported hardware support on LVGL9 and not started sw argb8565 support yet.

kisvegabor, 2024-10-28:
FYI, In the meantime I'm having meetings with some companies to better understand why and how we need this feature.

kisvegabor, 2024-11-13:
Not stale, it's on our radar. Can work on it in a month.

kisvegabor, 2026-09-09:
Sorry for the long silence. In past years it didn't came up as a need so close this issue right now. Let's reopen it if needed.

closed 2026-09-09 by kisvegabor