lvgl/lvgl #7029: lv anim support pause/resume

junminacts, 2024-10-10:
### Problem to solve

anim requires pause/resume in some cases to keep continuity

### Success criteria

pause the anim, sleep several seconds, the resume it started with the same anim value.

### Solution outline

add new control API, like lv_anim_pause/resume

### Rabbit holes

No

### Testing

pause the anim, sleep several seconds, the resume it started with the same anim value.

### Teaching

Doc the new API

kissa96, 2024-10-16:
Already working on it.

uLipe, 2024-12-11:
Not stale - It reminds me to go back on this

kisvegabor, 2025-01-03:
Not stale.

uLipe, 2025-01-09:
@AndreCostaaa sure, no problem I also already paid an initial review on your PR.

Thank you for your contribution!

PR #7583 (2025-01-15): feat(anim): add a pause method
Closes #7029 

This PR adds the ability to pause and resume animations
closed 2025-01-15 by kisvegabor