lvgl/lvgl #7030: Poor performance of roller widget

junminacts, 2024-10-10:
### LVGL version

LVGL 9

### What happened?

the roller draw performance is poor  with many options, especially with LV_ROLLER_MODE_INFINITE modes.

kisvegabor, 2024-10-10:
Hi,

Please send a code snippet to reproduce the issue. The interesting parts are:
- the options
- the font
- the roller's size

Billername, 2024-10-11:
Hi,
    At present, the stop position of the list after scrolling cannot be determined, and the control needs to be positioned at the specified position after scrolling.  You can refer to the following video (the middle row will remain in the middle or other fixed positions after each scrolling).  It is hoped that the rolling stop point can be returned to the user after sliding, and the user will calculate and set the stop point after obtaining it.

Uploading 123.mp4…

Billername, 2024-10-12:
Hi,
This is the code snippet.  Hope want to calculate only the display area , Thanks.

//opts the longer it is, the less efficient it is.
const char * opts = "50\n100\n150\n200\n250\n300\n350\n400\n\
					450\n500\n550\n600\n650\n700\n750\n800\n\
					850\n900\n950\n1000\n1050\n1100\n1150\n1200\n\
					1250\n1300\n1350\n1400\n1550\n1600\n1650\n1700\n\
					1750\n1800\n1850\n1900\n1950\n2000\n2050\n2100";
lv_obj_t *roller = lv_roller_create(data->tv);
lv_roller_set_options(roller,opts,LV_ROLLER_MODE_INFINITE);
lv_obj_set_width(roller,80);
lv_obj_set_style_text_font(roller,&data->font,LV_PART_SELECTED);
lv_obj_set_style_text_font(roller,&data->font,LV_PART_MAIN);
lv_obj_set_style_text_color(roller,lv_color_white(),LV_PART_SELECTED);
lv_obj_set_style_text_color(roller,lv_color_hex(0x333333),LV_PART_MAIN);
lv_obj_set_style_text_align(roller,LV_TEXT_ALIGN_RIGHT,LV_PART_SELECTED);
lv_obj_set_style_text_align(roller,LV_TEXT_ALIGN_RIGHT,LV_PART_MAIN);
lv_roller_set_visible_row_count(roller,3);
lv_obj_align(roller,LV_ALIGN_CENTER,0,0);

kissa96, 2024-10-17:
Hi @Billername Can you please provide a more detailed snippet with the exact fonts and everything, so we can reproduce it 1:1? Thanks in advance!

Billername, 2024-10-21:
Hi,
     Sorry，Here's what's been fixed。
        //opts the longer it is, the less efficient it is.
	const char * opts = "50\n100\n150\n200\n250\n300\n350\n400\n\
		450\n500\n550\n600\n650\n700\n750\n800\n\
		850\n900\n950\n1000\n1050\n1100\n1150\n1200\n\
		1250\n1300\n1350\n1400\n1550\n1600\n1650\n1700\n\
		1750\n1800\n1850\n1900\n1950\n2000\n2050\n2100";
	lv_obj_t *roller = lv_roller_create(lv_scr_act());
	lv_roller_set_options(roller,opts,LV_ROLLER_MODE_INFINITE);
	lv_obj_set_width(roller,80);
	lv_obj_set_style_text_font(roller,&data->font,LV_PART_SELECTED);
	lv_obj_set_style_text_font(roller,&data->font,LV_PART_MAIN);
	lv_obj_set_style_text_color(roller,lv_color_white(),LV_PART_SELECTED);
	lv_obj_set_style_text_color(roller,lv_color_hex(0x333333),LV_PART_MAIN);
	lv_obj_set_style_text_align(roller,LV_TEXT_ALIGN_RIGHT,LV_PART_SELECTED);
	lv_obj_set_style_text_align(roller,LV_TEXT_ALIGN_RIGHT,LV_PART_MAIN);
	lv_roller_set_visible_row_count(roller,3);
	lv_obj_align(roller,LV_ALIGN_CENTER,0,0);

kissa96, 2024-10-21:
Hi, can you please support the data struct as well, for the font? Thx in advance

Billername, 2024-10-21:
Hi, can use "lv_font_montserrat_48" font , the main problem is that the longer the "opts", the more difficult it is.
lv_obj_set_style_text_font(roller,&lv_font_montserrat_48 , LV_PART_SELECTED);
lv_obj_set_style_text_font(roller,&lv_font_montserrat_48 , LV_PART_MAIN);

kissa96, 2024-10-21:
I've tested it and everything seems to be working just fine. Which MCU are you using? I ran the code in simulator, so..

kissa96, 2024-10-21:
I assume you have debugged everything else. Is anything else in LVGL slow, or its only the roller?

Billername, 2024-10-22:
For wearing this is more commonly used, currently only found this control, if found later I will bring it up. Thanks.

lizhaoming634, 2024-10-23:
> Low efficiency compared with other controls，Actions 3085s cpu192Mhz。

Has the new SDK for Actions 3085s upgraded to LVGL 9? I remember it was using LVGL 8.3.

kisvegabor, 2024-11-13:
Not stale.

xuanmumy, 2024-12-04:
I also meet this issue . I use freetype font with a size 60, and it is worse than montserrt 48 font.

xuanmumy, 2024-12-04:
> I also meet this issue . I use freetype font with a size 60, and it is worse than montserrt 48 font.

my environment can be found at this link. https://github.com/lvgl/lvgl/issues/7410
the problem of this issue is that draw post event of roller_label take 15ms in my case. 
and the time is consumed by scrollbar part. 
I think this scrollbar is not useful in this case?
```
static void lv_roller_constructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
    LV_UNUSED(class_p);
    lv_roller_t * roller = (lv_roller_t *)obj;

    roller->mode = LV_ROLLER_MODE_NORMAL;
    roller->option_cnt = 0;
    roller->sel_opt_id = 0;
    roller->sel_opt_id_ori = 0;

    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLLABLE);
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN_VER);

    LV_LOG_INFO("begin");
    lv_obj_t * label = lv_obj_class_create_obj(&lv_roller_label_class, obj);
    lv_obj_class_init_obj(label);
    lv_roller_set_options(obj, "Option 1\nOption 2\nOption 3\nOption 4\nOption 5", LV_ROLLER_MODE_NORMAL);
    lv_obj_clear_flag(label, LV_OBJ_FLAG_SCROLLABLE); // this line cut off this 15ms of my code.
    LV_LOG_TRACE("finshed");
}
```

xuanmumy, 2024-12-04:
> my environment can be found at this link. #7410 the problem of this issue is that draw post event of roller_label take 15ms in my case. and the time is consumed by scrollbar part. I think this [...]

The static frame rate has been increased after this modification, but when scrolling up and down , the fps reduce a lot.  33fps -> 20fps. some other things still consuming time.

kisvegabor, 2024-12-22:
>     lv_obj_clear_flag(label, LV_OBJ_FLAG_SCROLLABLE); // this line cut off this 15ms of my code.

Great finding! Please send a Pull request with this simple fix.

In case of FreeType, have you enabled FreeType's cache in lv_conf.h?

kisvegabor, 2025-04-29:
I've just checked that the roller is fast with a built-in font and FreeType cache should make FreeType fast as well.

As it's not an issue with roller itself I close this issue now.

closed 2025-04-29 by kisvegabor