lvgl/lvgl #7032: 3D draw API plan

junminacts, 2024-10-10:
### Problem to solve

LVGL have any plan to add 3D draw API ? Small 3D GPU (with the similar features of OpenGL ES 2.0) will be integrated to IoT devices soon.

### Success criteria

Support 3D draw API to draw 3D model, glTF descripted models (https://www.khronos.org/gltf/), etc.

### Solution outline

can draw 3D models

### Rabbit holes

No

### Testing

draw a 3D box or glTF models 

### Teaching

Not yet

kisvegabor, 2024-10-10:
I'm not familiar enough with 3D to comment in detail, but I also agree that 3D on MCUs will be important in the future. 

`glTF` sounds great! We already have a built-in driver to create OpenGL windows and add textures there. If I understand correctly we need a `glTF` renderer (similar to a PNG decoder) which can render to an OpenGL or other texture.

junminacts, 2024-10-11:
Yes, we need a glTF renderer. Beside the glTF, 3d draw API, like vector API declared in lv_draw_vector.h, should be added.
The 3D GPU will be powered by Verisilicon and be similar to OpenGL ES 2.0.

kisvegabor, 2024-10-26:
It's not forgotten. In the background I'm having and still organizing meetings with some related parties.

PR #8033 (2025-04-16): feat(widget): Add `lv_3dtexture` widget and 3D draw task type
fixes #7032

Read the docs and see the example repo: https://github.com/lvgl/lv_example_3dtexture

[image]

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [... cut]