lvgl/lvgl #7035: Cannot seem to get flex wrapping to work

plueschpruem, 2024-10-10:
### LVGL version

9.2

### What happened?

- I create a screen with an object.
- Set object to LV_FLEX_FLOW_ROW_WRAP
- Place a bunch of buttons in there with min_width set

No matter what I try, the row of buttons never wraps.
Tried setting parent width, max-width, no widths, scroll modes…

Styles of the parent:
```
lv_style_set_width(&style_bootgroup, lv_pct(100));
lv_style_set_height(&style_bootgroup, lv_pct(100));
lv_style_set_layout(&style_bootgroup, LV_LAYOUT_FLEX);
lv_style_set_flex_flow(&style_bootgroup, LV_FLEX_FLOW_ROW_WRAP);
lv_style_set_flex_main_place(&style_bootgroup, LV_FLEX_ALIGN_SPACE_EVENLY);
```
Styles of the buttons:
```
lv_style_set_height(&style_btn_basic, LV_SIZE_CONTENT);
lv_style_set_margin_bottom(&style_btn_basic, 3);
lv_style_set_pad_left(&style_btn_basic, 0);
lv_style_set_pad_right(&style_btn_basic, 0);
lv_style_set_min_width(&style_btn_basic, 50);
```
Styles of the buttons labels:
```
lv_style_init(&style_btn_label_basic);
lv_style_set_align(&style_btn_label_basic, LV_ALIGN_CENTER);
lv_style_set_text_align(&style_btn_label_basic, LV_TEXT_ALIGN_CENTER);
lv_style_set_text_font(&style_btn_label_basic, &lv_font_montserrat_14_de);
lv_style_set_text_color(&style_btn_label_basic, lv_color_black());
```

Since there seems to be no way of breaking a row manually (consistent with CSS flexbox), I was hoping to use min-width to have a more uniform button size.
Any ideas?

egonbeermat, 2024-10-10:
You probably need to post more of a working example that shows the issue; all this shows is you created styles and not some of the more common issues, like you forgot to apply the style to the parent, or you didn't set the parent of the buttons to to be the flex container, etc. This code in the simulator, taking your style definitions and creating objects and applying the styles:

```
  static lv_style_t style_bootgroup;
  static lv_style_t style_btn_basic;
  static lv_style_t style_btn_label_basic;

  lv_style_init(&style_bootgroup);
  lv_style_init(&style_btn_basic);
  lv_style_init(&style_btn_label_basic);

  lv_style_set_width(&style_bootgroup, lv_pct(100));
  lv_style_set_height(&style_bootgroup, lv_pct(100));
  lv_style_set_layout(&style_bootgroup, LV_LAYOUT_FLEX);
  lv_style_set_flex_flow(&style_bootgroup, LV_FLEX_FLOW_ROW_WRAP);
  lv_style_set_flex_main_place(&style_bootgroup, LV_FLEX_ALIGN_SPACE_EVENLY);

  lv_style_set_height(&style_btn_basic, LV_SIZE_CONTENT);
  lv_style_set_margin_bottom(&style_btn_basic, 3);
  lv_style_set_pad_left(&style_btn_basic, 0);
  lv_style_set_pad_right(&style_btn_basic, 0);
  lv_style_set_min_width(&style_btn_basic, 50);

  lv_style_set_align(&style_btn_label_basic, LV_ALIGN_CENTER);
  lv_style_set_text_align(&style_btn_label_basic, LV_TEXT_ALIGN_CENTER);
  lv_style_set_text_font(&style_btn_label_basic, &lv_font_montserrat_16);
  lv_style_set_text_color(&style_btn_label_basic, lv_color_black());

  lv_obj_t * screen = lv_screen_active();
  lv_obj_t * cont = lv_obj_create(screen);
  lv_obj_add_style(cont, &style_bootgroup, 0);

   for (int i = 0; i < 20; i++) {
    lv_obj_t * button = lv_button_create(cont);
    lv_obj_t * label = lv_label_create(button);
    lv_obj_add_style(button, &style_btn_basic, 0);
    lv_obj_add_style(label, &style_btn_label_basic, 0);
   }
```
Produced this:

[image]

So, by itself, on the information you posted, there doesn't seem to be an issue with your styles or LVGL? It may be some issue in your wider code base, parent objects of the flex container, etc?

plueschpruem, 2024-10-10:
OK, I just figured it out!
`lv_obj_set_flex_grow(button_obj, grow);` leads to the wrapping to no take place.
When I don't set any grow, it wraps as intended.
I added that to each button individually. Values were between 1 and 4 in this case. and display correctly.
But did prohibit the wrapping.

So I guess in that instance lvgl flex differs from CSS…
Or is it a bug?

plueschpruem, 2024-10-11:
Alright, I did some more testing.
When adding the wraps manually with `lv_obj_add_flag(child, LV_OBJ_FLAG_FLEX_IN_NEW_TRACK)`, it works as intended.
But the question remains: Is `grow` incompatible with `wrap`, or is it just a bug?

kissa96, 2024-10-17:
@plueschpruem Is this resolved? If so, please close the issue, thanks!

plueschpruem, 2024-10-17:
> @plueschpruem Is this resolved? If so, please close the issue, thanks!

Well, I solved my problem – or circumvented it?
Question remains: Is "grow breaks wrap" the intended behaviour?
I will close it for now tho.

closed 2024-10-17 by plueschpruem

kisvegabor, 2024-10-17:
> Question remains: Is "grow breaks wrap" the intended behaviour?

Oh I see. Yes, they are really not working together nicely now. As workaround you can manually add the `LV_OBJ_FLAG_FLEX_IN_NEW_TRACK` flag to force a new line.

plueschpruem, 2024-10-17:
> As workaround you can manually add the `LV_OBJ_FLAG_FLEX_IN_NEW_TRACK` flag to force a new line.

Yes, but then you start managing your own rows again.
I was hoping to circumvent that by using wrapping.
Will this "feature request" somehow magically appear on some to-do list, or how does that work with LVGL?

kisvegabor, 2024-10-28:
I've just opened a PR to fix it :slightly_smiling_face: 
https://github.com/lvgl/lvgl/pull/7168

PR #7168 (2024-11-21): fix(flex): make min-width, grow, and wrap work together
fixes #7035

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle [... cut]