lvgl/lvgl #7047: List scrolling expansion

Billername, 2024-10-11:
### Problem to solve

At present, the stop position of the list after scrolling cannot be determined, and the control needs to be positioned at the specified position after scrolling.  You can refer to the following video (the middle row will remain in the middle or other fixed positions after each scrolling). 

https://github.com/user-attachments/assets/e35e866a-fa67-4bd3-996a-47f9060ca96a

### Success criteria

 It is hoped that the rolling stop point can be returned to the user after sliding, and the user will calculate and set the stop point after obtaining it.

### Solution outline

no

### Rabbit holes

no

### Testing

no

### Teaching

no

kissa96, 2024-10-16:
Interesting scenario for sure. So far I have not came across GUIs which strictly support this, as the user will quickly learn how much to scroll in order to get the desired outcome. These are just my 0.02USD. @kisvegabor are we going to support a feature like this?

kisvegabor, 2024-10-18:
I think what you need the [scroll snapping](https://docs.lvgl.io/master/overview/scroll.html#snapping) which is already supported. Or am I missing something?

Billername, 2024-10-21:
Hi,  
    Something different  ,when the list starts scrolling animation, need to know the stop destination and then re-change the end destination of the scrolling animation. This scrolling animation is not triggered by the user, it is triggered by TP put up. 
     Scroll snapping, I only know the obj position, but I don't know what the purpose is, I don't know how long it takes to stop.
For example:
    TP put up, scroll anim self-triggering y to 33 pixel , But because of the layout I have to move y / 10, 3 pixel need to abandon or 
y to 40pixel.

kisvegabor, 2024-10-31:
Assuming that TP means touchpad I think it's really the scroll snap. See
- https://docs.lvgl.io/master/details/base-widget/scroll.html#id2
- https://docs.lvgl.io/master/details/base-widget/scroll.html#translate-on-scroll

closed 2024-11-28 by lvgl-bot