lvgl/lvgl #7050: Platformio does not use lv_conf.h

Grey-Lancaster, 2024-10-11:
### LVGL version

v8.4.0

### What happened?

what I do not understand is if I do not put lv_conf.h next to the lvgl folder or if I use `#define LV_CONF_INCLUDE_SIMPLE ` and do not put it in the lib folder I will get an error lv_conf.h not found
But when it is properly in place it will build but ignores any changes I make to it.  I need to make it smaller I am running out of space on my 4mb esp32.

So I removed the fonts, but they still compile.

I even tried to edit kconfig and the other lv_conf_internal and it still compiles un-needed fonts

### How to reproduce?

 ```

/* clang-format off */
#if 1 /*Set it to "1" to enable content*/

#define LV_FONT_MONTSERRAT_34 1
#define LV_FONT_MONTSERRAT_36 1
#define LV_FONT_MONTSERRAT_38 0
#define LV_FONT_MONTSERRAT_40 0
#define LV_FONT_MONTSERRAT_42 0
#define LV_FONT_MONTSERRAT_44 0
#define LV_FONT_MONTSERRAT_46 0
#define LV_FONT_MONTSERRAT_48 0
```

Grey-Lancaster, 2024-10-12:
a better example is I use
#define LV_FONT_MONTSERRAT_22 1
if it is set to 0 it gives an error and will not build
If i set it to 1 it does build
The problem is is compiles all fonts, even the ones set to zero

So it does see the lv_conf.h "sometimes"

kissa96, 2024-10-14:
Have you copied it from the lv_conf_template.h? Have you enabled its contents?
LVGL uses an internal config header, if lv_conf is not in place to define the default values for the symbols.
I'd suggest looking into it.

kisvegabor, 2024-10-14:
I think it's related to PlatformIO. Can you remove `lvgl` from the build library (I don't know the exact location), change the configs and build again?

Grey-Lancaster, 2024-10-14:
Thanks, I have.  In Platformio, when you "clean" it erases all of the downloaded "stuff" (lvgl, Tft_espi)  The instructions say to place lv_conf next to the lvgl folder.  And that works.  But when you do  Clean in Platformio it also erase the lv_conf.h file so you have to make your edits again. Copy template, rename, edit.
So what you can do is make a lib folder and place the lvgl folder in there and it will not get erased and downloaded each time..  It does not work with either approach.

Like I said it does read lv_conf.h when it builds, but not when it compiles.

first picture is an error when I do not include the font.  Second picture is when it is included, but it also compiles all the excluded fonts.

[image]

[image]

I can even whack them out in lv_conf_internal.h and it will still add them

[image]

the only way to kill a font is to delete it or rename it in the src>font folder.  That sid I have no clue why it still builds the 28 point font.

Again, thanks for looking. Does no one really use Platformio?  I like it 

[image]

egonbeermat, 2024-10-23:
I use PlatformIO in VSCode on MacOS. I cannot say I am fully versed in the nuances of configuration for it, or can troubleshoot it as it can be very finiky, but this is how I have it setup, so that I don't lose config or LVGL folders with 'clean':

- `lv_conf.h` is in my `<project_root>/config/ `folder
- I have a local folder containing LVGL that I download independently from Git Hub, rather than specify it in `lib_deps`
- In `platformio.ini`, I specify the LVGL folder under the `[env]` section using `lib_extra_dirs = <LVGL_folder_location>`
- In `platformio.ini`, I add `-D LV_LVGL_H_INCLUDE_SIMPLE` to the `build_flags` in the `[env]` section
- In `platformio.ini`, I add `-I config` to the `build_flags` in the `[env]` section, to make sure it includes the config folder

So part of my `platformio.ini` file looks like this (edited for brevity, there could be other things there):

```
[env]
platform = <platform>
framework = <framework>
lib_extra_dirs = ~/Documents/libraries/LVGL
build_flags = -D LV_LVGL_H_INCLUDE_SIMPLE
              -I config
              -I src
```

Grey-Lancaster, 2024-10-23:
Thanks, I tried the tweaks you suggested but it still compiles fonts it should not

```
lib_extra_dirs = ~/Documents/lib/LVGL	
build_flags = -D LV_LVGL_H_INCLUDE_SIMPLE
              -I config
              -I src

```

[image]

egonbeermat, 2024-10-23:
Yes, mine reports it is compiling all the lv_font files too, even though I don't use any of them (all defines set to 0 in my lv_conf.h), but that isn't an issue - setting a font define to 0 in lv_conf.h doesn't _exclude_ the font file from the compiler, the compiler still parses the file, and reports it is parsing the file, but _inside_ the font file, the define excludes the content from being compiled. So, you should _expect_ to see the compiler compiling all the font files.

Grey-Lancaster, 2024-10-23:
Well I cannot seem to find the file like your example.  But I did enable some fonts and compare the sizes.  It does look like what you say is the gospel, even if it is confusing LOL.

Thanks so much.
with more fonts enabled

[image]

Disabled

[image]

egonbeermat, 2024-10-23:
For additional peace of mind, you can check the platformIO build folder, find the LVGL lib and look for the object files referenced above as the compiler output (eg, `lv_font_montserrat_26.c.o`). For this font, if it was compiled (define set to 1) the obj file is around 37KB, but when the define was set to 0, it is 724 bytes. The linker may still not include the font object file even if compiled, if there is no reference to it in your source....

Grey-Lancaster, 2024-10-23:
I appreciate your help. As I said setting to zero does seem to work.  So I guess I am good

Well I think I see what you are trying to point out.  That said it looks like only the fonts used in my project are not greyed out. Whether it is set to zero or not.  Go figure 
For example is not used in my code
`#define LV_FONT_MONTSERRAT_10 1`

shows

[image]

egonbeermat, 2024-10-23:
VSCode means well, but it isn't always the best at finding origin source files for definitions, so I wouldn't read too much into it. I know it works well for LVGL files opened in VSCode under my project, as I have the conf file location well defined, but not so much if I open the same files under a different project.....

Grey-Lancaster, 2024-10-23:
Thanks again.  I will close this not as fixed, but understood :-)

closed 2024-10-23 by Grey-Lancaster