lvgl/lvgl #7051: May I ask if there is another implementation of the lv_canvas-blur'hor function in lvgl9

jinsc123654, 2024-10-12:
### Introduce the problem

I used the blur function on the drawing board of LVGL8 and now I want to migrate to LVGL9. Is there a similar method for LVGL9

`    lv_area_t blur_area;
    blur_area.x1 = 0;
    blur_area.y1 = 0;
    blur_area.x2 = 100;
    blur_area.y2 = 150;

    lv_canvas_blur_hor(canvas, &blur_area, 50);
    lv_canvas_blur_ver(canvas, &blur_area, 50);`

kissa96, 2024-10-14:
As far as I know blurring will be a general feature in an upcoming release.

kisvegabor, 2024-10-14:
I can confirm @kissa96's comment. As any other drawing functions it will work on a canvas too. 

Do you have any special requirements such as:
- selectively blur a smaller area
- blur only inside a rounded rectangle/circle
- animate blurring (requires very high speed)
- large area blur

jinsc123654, 2024-10-14:
That is to say, the current lvgl9 does not currently support the canvas blur function.

Do I need to wait for the new version to be released

jinsc123654, 2024-10-15:
I want to achieve this kind of graphic effect
[image]

kisvegabor, 2024-10-25:
Oh, this a complicated one (large, rounded, on smaller area). 

Anyway, It's really not supported now, but it will be added in v9.3.

liamHowatt, 2024-10-28:
This might not be the most helpful comment, but for your exact example you may get the best performance by using a static image of the already-blurred scene (blurred by image editing software) and draw other objects (i.e. the clock hands) on top.

GorgonMeducer, 2024-10-28:
If no content changes dynamically below the blur layer, it is highly recommended to use a static image rather than a blur effect. It is very costly.

closed 2024-11-19 by lvgl-bot