lvgl/lvgl #7053: Cannot rotate screen

ryanjtr, 2024-10-12:
### LVGL version

9.3.0

### What happened?

i use board stm32f746G-DISCO, when i use macro lv_disp_set_rotation(lv_disp_get_default(), LV_DISP_ROTATION_180), i cannot rotate the screen, all button is disable, i cannot press it. Can anyone tell me why? Previously, i use version 8.2, this macro used successfully.

ryanjtr, 2024-10-12:
This is my code

int main(void)
{

  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  /* USER CODE BEGIN 2 */
  /* Enable I-Cache */
    SCB_EnableICache();

    /* Enable D-Cache */
    SCB_EnableDCache();

  	lv_init();
  	tft_init();

  	touchpad_init();
//  	ui_init();
//  	lv_disp_set_rotation(lv_disp_get_default(), LV_DISP_ROTATION_180);
  	drop_down_window();

//  lv_example_dropdown_1();
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  HAL_Delay(5);
	  lv_task_handler();
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}

MacDragon, 2024-10-12:
Are you handling the rotation in flush callback? > https://docs.lvgl.io/9.2/porting/display.html

ryanjtr, 2024-10-13:
> Are you handling the rotation in flush callback? > https://docs.lvgl.io/9.2/porting/display.html as noted in the docs, this behaviour changed majorly in v9

How can i rotate my screen ?

kissa96, 2024-10-14:
Hi, have you set up all buffers and callbacks as required? Can you post some of the code?

kisvegabor, 2024-10-14:
You can take look at [this one](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L213-L240) as a reference implementation for software rotation

keerthansys22, 2024-10-14:
Hi @kissa96, even I'm facing the same issue that I can't rotate my screen using v9.2.0 but in earlier v8.3.9 it was working fine.

The code for v9.2.0 is -

/ * @brief Function to initialize lvgl drivers */
void lvgl_init(void)
{
    // LVGL initialization
      lv_init();

      // Create a display
      disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);

      // Set the draw buffers
      lv_display_set_buffers(disp, buf_1, buf_2, (MY_DISP_HOR_RES * BYTES_PER_PIXEL), LV_DISPLAY_RENDER_MODE_PARTIAL);

      // Rotate the screen 
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);

      // Set the flush callback function
      lv_display_set_flush_cb(disp, disp_flush);

}

/* @brief   Flush the content of the internal buffer the specific area on the display */
static void disp_flush(lv_disp_t* display, const lv_area_t * area, uint8_t * color_p)
{
    lv_draw_sw_rgb565_swap(buf_1, (MY_DISP_HOR_RES * MY_DISP_VER_RES));
    lcd_send_data(area->x1, area->y1, area->x2, area->y2, (uint8_t *)color_p);

    lv_display_flush_ready(display);
}

kisvegabor, 2024-10-15:
In v9 there is no `disp_drv.sw_rotate = 1;` but you need to cal la rotate function manually in the `flush_cb` as mentioned [here](https://github.com/lvgl/lvgl/issues/7053#issuecomment-2410502776).

kissa96, 2024-10-17:
@keerthansys22 How are you doing with this issue?

Keerthan22-sys, 2024-10-20:
Hi @kissa96, I'm not getting the exact implementation!! I tried and got an error, can you kindly look into this?

keerthansys22, 2024-10-22:
Hi @kissa96, I tried on my side but I'm facing some error!! Is there any way to go with it?

kissa96, 2024-10-22:
Hi @keerthansys22, if you can supply us with the errors, you are facing, we might be able to help!

keerthansys22, 2024-10-23:
Hi @kissa96, I have attached the code snippet and I'm getting segmentation fault at lv_draw_sw_rotate() function. Thanks.

```c
static void disp_flush(lv_disp_t* display, const lv_area_t * area, uint8_t * color_p)
{
    /*SPI transmit data without DMA*/
	
	// screen rotation
	lv_color_format_t cf = lv_display_get_color_format(display_lvgl);
	lv_area_t rotated_area = *area;
	lv_display_rotate_area(display_lvgl, &rotated_area);

	int32_t px_map_w = lv_area_get_width(area);
	int32_t px_map_h = lv_area_get_height(area);
	uint32_t px_map_stride = lv_draw_buf_width_to_stride(lv_area_get_width(area), cf);
	uint32_t px_size = lv_color_format_get_size(cf);

	int32_t fb_stride = lv_draw_buf_width_to_stride(MY_DISP_HOR_RES, cf);
	uint8_t * fb_start = (uint8_t *) area->x1;
	fb_start += rotated_area.y1 * fb_stride + rotated_area.x1 * px_size;
	lv_display_rotation_t rotation = lv_display_get_rotation(display_lvgl);

	lv_draw_sw_rotate(color_p, fb_start, px_map_w, px_map_h, px_map_stride, fb_stride, rotation, cf);
	
	
	// swap to RGB_565
    lv_draw_sw_rgb565_swap(buf_1, (MY_DISP_HOR_RES * MY_DISP_VER_RES));

	// Send data to LCD
    lcd_send_data(area->x1, area->y1, area->x2, area->y2, (uint8_t *)color_p);

    lv_display_flush_ready(display);
}
```

kissa96, 2024-10-25:
Hi, does display_lvgl point to the display that the flush callback provides, or is this a separate display?

keerthansys22, 2024-10-25:
Hi @kissa96, my bad it was a typo error, I have updated my code snippet now. Actually, it is pointing to the same display not a separate display. 

I found the similar issue here - https://github.com/lvgl/lvgl/issues/6885, this might help us. Thanks.

keerthansys22, 2024-10-25:
I tried what was mentioned in the https://github.com/lvgl/lvgl/issues/6885, by downloading the latest LVGL v9.2.1 but the same error is happening!!

nikunj8780, 2024-10-25:
If you are using LVGL v9.2.1  define display roatation like :  lv_disp_set_rotation(disp, LV_DISPLAY_ROTATION_90);

kisvegabor, 2024-10-25:
This line is not correct for sure: `uint8_t * fb_start = (uint8_t *) area->x1;`

If you are using an external display controller:
1. Rotate the area
2. You should allocate the buffer to store the rotated image
3. Rotate the image
4. Send the rotated area to the rotated image

See this code as [an example](https://github.com/lvgl/lvgl/blob/release/v9.2/src/drivers/sdl/lv_sdl_window.c#L220-L250).

keerthansys22, 2024-10-28:
Thanks for your message @nikunj8780, I even tried defining display roatation like : lv_disp_set_rotation(disp, LV_DISPLAY_ROTATION_90); it is not working!

nikunj8780, 2024-10-28:
`#include <lvgl.h>

#if LV_USE_TFT_ESPI
#include <TFT_eSPI.h>
#endif

#define TFT_HOR_RES   320
#define TFT_VER_RES   240
#define TFT_ROTATION  LV_DISPLAY_ROTATION_0

/*LVGL draw into this buffer, 1/10 screen size usually works well. The size is in bytes*/
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
uint32_t draw_buf[DRAW_BUF_SIZE / 4];

#if LV_USE_LOG != 0
void my_print( lv_log_level_t level, const char * buf )
{
    LV_UNUSED(level);
    Serial.println(buf);
    Serial.flush();
}
#endif

/* LVGL calls it when a rendered image needs to copied to the display*/
void my_disp_flush( lv_display_t *disp, const lv_area_t *area, uint8_t * px_map)
{

    lv_display_flush_ready(disp);
}

/*Read the touchpad*/
void my_touchpad_read( lv_indev_t * indev, lv_indev_data_t * data )
{

static uint32_t my_tick(void)
{
    return millis();
}

void setup()
{
    String LVGL_Arduino = "Hello Arduino! ";
    LVGL_Arduino += String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();

    Serial.begin( 115200 );
    Serial.println( LVGL_Arduino );

    lv_init();

    /*Set a tick source so that LVGL will know how much time elapsed. */
    lv_tick_set_cb(my_tick);

    /* register print function for debugging */
#if LV_USE_LOG != 0
    lv_log_register_print_cb( my_print );
#endif

    lv_display_t * disp;
#if LV_USE_TFT_ESPI
    /*TFT_eSPI can be enabled lv_conf.h to initialize the display in a simple way*/
    disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, sizeof(draw_buf));
    lv_display_set_rotation(disp, TFT_ROTATION);

#else
    /*Else create a display yourself*/
    disp = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
    lv_display_set_flush_cb(disp, my_disp_flush);
    lv_display_set_buffers(disp, draw_buf, NULL, sizeof(draw_buf), LV_DISPLAY_RENDER_MODE_PARTIAL);
#endif

    /*Initialize the (dummy) input device driver*/
    lv_indev_t * indev = lv_indev_create();
    lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER); /*Touchpad should have POINTER type*/
    lv_indev_set_read_cb(indev, my_touchpad_read);

    lv_obj_t *label = lv_label_create( lv_screen_active() );
    lv_label_set_text( label, "Hello Arduino, I'm LVGL!" );
    lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );

    Serial.println( "Setup done" );
}

void loop()
{
    lv_timer_handler(); /* let the GUI do its work */
    delay(5); /* let this time pass */
}
`

see this code is proper work in Arduino ide and also work DISPLAY_ROTATION is correctly if you want to demo code for Espressif ide also tell me

kissa96, 2024-10-28:
You need to rotate the px_map. There are working functions for it, like draw_sw_rotate_rgb565()

keerthansys22, 2024-10-28:
Thanks @nikunj8780 for sharing the project example, but just to make sure I'm getting this issue in V9.2.0

nikunj8780, 2024-10-28:
Ok I understood but I have not working in Espressif IDE currently if you comfortable with Arduino IDE tell me because I have worked with Arduino IDE and LVGLv9.2.0. 
 @keerthansys22

keerthansys22, 2024-10-28:
Thanks for the message @nikunj8780, I'm switching my RGB565 to RGB888 for better presentation and visualization. I'll try for RGB888, I'll let you know how this works. Thanks.

keerthansys22, 2024-11-07:
Even for RGB888, the software screen rotation API is not working!! However, I will do rotate from the LCD driver itself. Thanks.

closed 2024-12-03 by lvgl-bot