lvgl/lvgl #7055: [v9] Text selection and recoloring

C47D, 2024-10-13:
### Introduce the problem

When testing #6966 I noticed that the text selection and text recoloring features don't like each other.
The text selection takes the recolor "command" (`#XXXXXX`) into consideration when counting the characters, so we have 'unexpected' results when trying to select text that is recolored.

```c
  lv_text_decor_t decor = LV_TEXT_DECOR_NONE;
  lv_color_t color = lv_palette_main(LV_PALETTE_BLUE);
  lv_color_t sel_bg_color = lv_palette_lighten(LV_PALETTE_RED, 4);
  lv_color_t sel_color = lv_palette_darken(LV_PALETTE_RED, 4);

  lv_obj_t * label = lv_label_create(lv_screen_active());
  lv_label_set_recolor(label, true);
  lv_label_set_text(label, "Hi,\nTesting the #00ff00 colored \nlabels.");
  lv_obj_set_style_text_decor(label, decor, 0);
  lv_obj_set_style_text_color(label, color, 0);
  lv_obj_set_style_text_opa(label, LV_OPA_COVER, 0);
  lv_obj_set_style_text_align(label, LV_TEXT_ALIGN_LEFT, 0);
  lv_obj_set_style_bg_color(label, sel_bg_color, LV_PART_SELECTED);
  lv_obj_set_style_text_color(label, sel_color, LV_PART_SELECTED);
  lv_label_set_text_selection_start(label, 10);
  lv_label_set_text_selection_end(label, 30);
```

[image]

### Proposal

I think we have two options:

1. Document this behavior, text selection doesn't work as expected when a recolor command is inside the selection range.
2. Teach the text selection 'algorithm' how to detect and don't consider the text recolor command into consideration when counting.

@kisvegabor What do you think?

C47D, 2024-10-16:
> So you have time and interest to look into it?

Interest yes, time more or less, next month I will have a heavy day job couple of weeks, do you think this will be needed for v9.3 or v9.4?

kisvegabor, 2024-10-17:
9.3 will be released only at 2025 Q1 So we have a lot of time :)

closed 2024-10-26 by C47D

C47D, 2025-01-09:
test

kisvegabor, 2025-01-09:
test worked. 🙂