lvgl/lvgl #7056: Does anyone use lvgl to create mini game applications?

wanghui1983, 2024-10-14:
### Introduce the problem

Currently, it is very popular to implement mini games on watches, most of which may be based on JS frameworks. Do any developers develop mini games based on LVGL? Do you have any relevant resources?

### Proposal

We want to find some resources to use lvgl for making small games, which can help us develop some small games for watches

lizhaoming19980614, 2024-10-14:
https://github.com/100askTeam/lv_lib_100ask/tree/master/src/lv_100ask_2048

kissa96, 2024-10-14:
Hi @wanghui1983, although core LVGL is not made for creating games, it is certainly capable of it, with its own widget limitations. Go get creative :)

kissa96, 2024-10-17:
@liamHowatt Can you close this issue please? Thx

closed 2024-11-12 by lvgl-bot