lvgl/lvgl #7057: Scroll events for objects in container

fatiihyildizz, 2024-10-14:
### Introduce the problem

Hello,
I am using a container containing objects. When the scrolling, There is a slowdown due to animation. I want to remove animation time for the slowless. For this, created a `LV_EVENT_SCROLL_BEGIN` event function for container. 
But this event only occurs at the beginning and end of the container when the area is scrolled out. I want this to occur when scrolling objects in the container up/down. This way, I will be able to control all animation times. How do I do this? Do I need to make all objects scrollable?

Active container codes for the scrolling :
```
lv_obj_add_flag(ps_cont_list, LV_OBJ_FLAG_SCROLL_ON_FOCUS);
lv_obj_set_scroll_dir(ps_cont_list, LV_DIR_VER); // Sadece dikey kaydırma
lv_obj_set_scroll_snap_y(ps_cont_list, LV_SCROLL_SNAP_NONE);

lv_obj_add_event_cb(ps_cont_list, v_TOUCHEVENT_ContainerScrollBegin, LV_EVENT_SCROLL_BEGIN, NULL);
```

kissa96, 2024-10-16:
Can you please provide code for the problem, so we can reproduce it 1:1? Thx in advance.

kisvegabor, 2024-10-16:
Maybe you just need to remove `LV_OBJ_FLAG_SCROLL_MOMENTUM`.

fatiihyildizz, 2024-10-16:
This work @kisvegabor . After remove LV_OBJ_FLAG_SCROLL_MOMENTUM, event generated for all scrolls, . Thank you