lvgl/lvgl #7060: LV_COLOR_16_SWAP define missing from config

cosmindanielradu19, 2024-10-14:
### LVGL version

v9.2.0

### What happened?

For LVGL support on LPC platforms, it seems that we need to set LV_COLOR_16_SWAP as 1, or the panel output color is not correct.

The LV_COLOR_16_SWAP was enabled in the lv_conf.h on LVGL v8.3.10, and the define is still used in [lv_refr.c](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L1260) . Shall we create a PR to add it in the current config?

Without LV_COLOR_16_SWAP 

[image]

With LV_COLOR_16_SWAP 

[image]

cosmindanielradu19, 2024-10-14:
@kisvegabor 
We already have the patch for this fix. If the changes looks good to you, we can create a PR.

[0001-fix-config-bring-back-LV_COLOR_16_SWAP-define.patch](https://github.com/user-attachments/files/17362621/0001-fix-config-bring-back-LV_COLOR_16_SWAP-define.patch)

kisvegabor, 2024-10-14:
Hi,
We have kept `LV_COLOR_DEPTH_16_SWAP` only for backward compatibility. The v9-ish was of doing it is calling `lv_draw_sw_rgb565_swap` in the `flush_cb`. See the details [here](https://docs.lvgl.io/master/porting/display.html#swap-endianness).

keerthansys22, 2024-10-15:
Hi @cosmindanielradu19, I'm attaching the working code snippet of lv_draw_sw_rgb565_swap in my flush_cb. Thanks.

```c
static void disp_flush(lv_disp_t* display, const lv_area_t * area, uint8_t * color_p)
{
    lv_draw_sw_rgb565_swap(buf_1, (MY_DISP_HOR_RES * MY_DISP_VER_RES));
    lcd_send_data(area->x1, area->y1, area->x2, area->y2, (uint8_t *)color_p);

    lv_display_flush_ready(display);
}

cosmindanielradu19, 2024-10-17:
Thank you for the clarification

closed 2024-10-17 by cosmindanielradu19