lvgl/lvgl #7065: Source control: what files/folders to include

focussing, 2024-10-14:
### LVGL version

v9.2.0 / all versions

### What happened?

Posted in Forum but no reactions.

My development environment is Arduino using Sketchbook location. Raspberry Pi RP2040 is the controller.
The Sketchbook location holds the source files (.ino and .h), and also the libraries folder.
I want to put the whole workspace including source files and also libraries under version control.
However the number of files and folders are quite high, so I want only to put all necessary files and folders under version control.
Which can I delete without problems?

kissa96, 2024-10-16:
For LVGL to work properly, you need ./src, lv_conf.h and lvgl.h in the project directory

kisvegabor, 2024-10-18:
And `lvgl.h`. :slightly_smiling_face:

focussing, 2024-11-10:
I was out of office for a week, next week I will come back to this issue.
Thanks a lot so far!

closed 2024-12-06 by lvgl-bot