lvgl/lvgl #7066: Using Kconfig for custom memory settings (malloc, free)

jag3705, 2024-10-14:
### LVGL version

v9.2.0

### What happened?

Hello everyone,

I have attempted to build a first project with lvgl, targeting ESP32-S3. I quickly ran out of memory, as hinted in issues #3742, #3244 and #2328.

I tried using the Kconfig environment to instruct Lvgl to use a custom/externally provided malloc/free function. However, I realized that the Kconfig file in Lvgl does only support the boolean whether or not to use Lvgl's built-in memory functions. It lacks the possibility to define a custom include header and set the function names. This should be supported by Lvgl on its 'lvgl_conf.h' approach already. Especially from the 8.4 to the 9.X release, there was a significant change of Kconfig for memory settings, however no changelog explains the matter; see also https://github.com/lvgl/lvgl/blob/release/v8.4/Kconfig 

My question is: how do you correctly instruct Lvgl 9.2 to make use of custom memory functions, leveraging Kconfig.

Thank you all for your work and best,
Jan

### How to reproduce?

```
	menu "Memory Settings"
		choice
			prompt "Malloc functions source"
			default LV_USE_BUILTIN_MALLOC

		config LV_USE_BUILTIN_MALLOC
			bool "LVGL's built in implementation"

		config LV_USE_CLIB_MALLOC
			bool "Standard C functions malloc/realloc/free"

		config LV_USE_MICROPYTHON_MALLOC
			bool "MicroPython functions malloc/realloc/free"

		config LV_USE_RTTHREAD_MALLOC
			bool "RTThread functions malloc/realloc/free"

		config LV_USE_CUSTOM_MALLOC
			bool "Implement the functions externally"

		endchoice # "Malloc functions"
```

C47D, 2024-10-21:
Hi,
By choosing the `LV_USE_CUSTOM_MALLOC` option it's not possible to use custom functions? It's been a while since I used the Espressif chips with LVGL, so maybe I'm missing something :smile_cat:

kissa96, 2024-10-21:
Hi, if you are using custom malloc, then you have to implement and pass the handle inside LVGL, otherwise the lib will never know which function to call.

liamHowatt, 2024-10-22:
I think the correct way to use `LV_USE_CUSTOM_MALLOC` is to define your own global functions that match the function prototypes in lv_mem.h and they will be linked by the linker automatically and used throughout LVGL.

closed 2024-11-18 by lvgl-bot

lanistor, 2025-02-08:
@kissa96 @kisvegabor It seems that you cannot define memory functions outside of lvgl, or linker will fail.

Do We have to put it into lvgl's source code?

liamHowatt, 2025-02-28:
Can you please send your linker error? Did you set `LV_USE_STDLIB_MALLOC` to `LV_STDLIB_CUSTOM`?

lanistor, 2025-03-02:
> Can you please send your linker error? Did you set `LV_USE_STDLIB_MALLOC` to `LV_STDLIB_CUSTOM`?

Yeah, even i changed source code `LV_USE_STDLIB_MALLOC` at "lv_conf_internal.h", linker will also failed.
```cpp
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_CUSTOM
``` 

And here is the log:
```
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/lv_conf_internal.h:65:120: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                                                                                                                        ^
[ 99%] Linking C static library libmain.a
[100%] Built target __idf_main
[100%] Generating ld/sections.ld
[100%] Built target __ldgen_output_sections.ld
Consolidate compiler generated dependencies of target FlickerList.elf
[100%] Linking CXX executable FlickerList.elf
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_init.c.obj):(.literal.lv_init+0x8): undefined reference to `lv_mem_init'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_init.c.obj): in function `lv_global_init':
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/lv_init.c:115: undefined reference to `lv_mem_init'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj):(.literal.lv_malloc+0x4): undefined reference to `lv_malloc_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj):(.literal.lv_free+0x0): undefined reference to `lv_free_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj):(.literal.lv_realloc+0x0): undefined reference to `lv_realloc_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj): in function `lv_malloc':
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/stdlib/lv_mem.c:70: undefined reference to `lv_malloc_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj): in function `lv_malloc_zeroed':
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/stdlib/lv_mem.c:100: undefined reference to `lv_malloc_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj): in function `lv_free':
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/stdlib/lv_mem.c:125: undefined reference to `lv_free_core'
/Users/Shared/SDK/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/lvgl/liblvgl.a(lv_mem.c.obj): in function `lv_realloc':
/Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display_2/components/lvgl/src/stdlib/lv_mem.c:137: undefined reference to `lv_realloc_core'
collect2: error: ld returned 1 exit status
make[2]: *** [FlickerList.elf] Error 1
make[1]: *** [CMakeFiles/FlickerList.elf.dir/all] Error 2
make: *** [all] Error 2
make failed with exit code 2
```

liamHowatt, 2025-03-12:
Is the file that defines the custom `lv_malloc` etc. being compiled?

lanistor, 2025-03-13:
> Is the file that defines the custom `lv_malloc` etc. being compiled?

I add this file in my CMakeLists.txt, but the lvgl will compile itself, utile linked to main project.

liamHowatt, 2025-03-13:
Sorry, the functions you have to define yourself are these (not `lv_malloc`):
```c
void lv_mem_init(void);
void lv_mem_deinit(void);
lv_mem_pool_t lv_mem_add_pool(void * mem, size_t bytes);
void lv_mem_remove_pool(lv_mem_pool_t pool);
void * lv_malloc_core(size_t size);
void * lv_realloc_core(void * p, size_t new_size);
void lv_free_core(void * p);
void lv_mem_monitor_core(lv_mem_monitor_t * mon_p);
lv_result_t lv_mem_test_core(void);
```

Use this as a template: https://github.com/lvgl/lvgl/blob/master/src/stdlib/clib/lv_mem_core_clib.c

giantbing, 2025-03-28:
i copy https://github.com/lvgl/lvgl/blob/master/src/stdlib/clib/lv_mem_core_clib.c this file in to the  main  and add `#include "src/stdlib/lv_mem.h"` this include  ,but not work `/Users/giantbing/project/esp32/matrix_idf/managed_components/lvgl__lvgl/src/lv_init.c:165:(.text.lv_init+0x15): undefined reference to `lv_mem_init'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj):(.literal.lv_malloc+0x4): undefined reference to `lv_malloc_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj):(.literal.lv_free+0x0): undefined reference to `lv_free_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj):(.literal.lv_realloc+0x0): undefined reference to `lv_realloc_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj): in function `lv_malloc':
/Users/giantbing/project/esp32/matrix_idf/managed_components/lvgl__lvgl/src/stdlib/lv_mem.c:70:(.text.lv_malloc+0x7): undefined reference to `lv_malloc_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj): in function `lv_malloc_zeroed':
/Users/giantbing/project/esp32/matrix_idf/managed_components/lvgl__lvgl/src/stdlib/lv_mem.c:100:(.text.lv_malloc_zeroed+0xa): undefined reference to `lv_malloc_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj): in function `lv_free':
/Users/giantbing/project/esp32/matrix_idf/managed_components/lvgl__lvgl/src/stdlib/lv_mem.c:125:(.text.lv_free+0xe): undefined reference to `lv_free_core'
/Users/giantbing/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lvgl__lvgl/liblvgl__lvgl.a(lv_mem.c.obj): in function `lv_realloc':
/Users/giantbing/project/esp32/matrix_idf/managed_components/lvgl__lvgl/src/stdlib/lv_mem.c:137:(.text.lv_realloc+0x24): undefined reference to `lv_realloc_core'
collect2: error: ld returned 1 exit status`