lvgl/lvgl #7067: Color Format I1 vtile

faxe1008, 2024-10-14:
### Problem to solve

Currently the I1 sw renderer assumes a htiled display. Either add a mode support of sorts to the renderer or provide method which takes a buffer and a temporary buffer and perform the conversion.

`lv_i1_convert_to_vtiled(uint8_t* buf, uint32_t width, uint32_t height, uint8_t* tmp, uint32_t tmp_buf_size)` 

Allowing the user to provide the temporary buffer allows for more flexibility on where to place the buffer. 

### Success criteria

Using vtiled monochrome displays is possible.

### Solution outline

See description 

### Rabbit holes

None that I know of?

### Testing

In CI using unit tests.

### Teaching

Using documentation in the spot where color formats are explained.

faxe1008, 2024-10-16:
Hmm good question. Would not this be handeled sufficiently if the temporary buffer the user provides is rounded? So for your example this would mean

1 byte * 8 bytes = 8 byte buffer

Or am I missing something?
This is also why I added the temporary buffer size as a parameter so we can put asserts into place to check for this.

kisvegabor, 2024-10-16:
I mean let's say this is rendered now:
```
   00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15
00 x  x  x  x  x  x  x  x
01 x  x  x  x  x  x  x  x
02
03
04
05
06
07
08
09
```

If we transform, how will have e.g. row 3, column 4?
```
   00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15
00 x  x  x  x  x  x  x  x
01 x  x  x  x  x  x  x  x
02 ?  ?  ?  ?  ?  ?  ?  ?
03 ?  ?  ?  ?  ?  ?  ?  ?
04 ?  ?  ?  ?  ?  ?  ?  ?
05 ?  ?  ?  ?  ?  ?  ?  ?
06 ?  ?  ?  ?  ?  ?  ?  ?
07 ?  ?  ?  ?  ?  ?  ?  ?
08
09
```

IMO we can do 2 things:
1. Round the invalidated area's y coordinate to Nx8
2. Use the FULL rendering mode

As monochrome displays are usually small I think we shouldn't implement a dedicated VTILED renderer. Converting the rendered image shouldn't take too long.   

What do you think?

faxe1008, 2024-10-16:
Yes we have to round for this to work. Conversion times should be also manageable I think.

kisvegabor, 2024-10-18:
Okay, great! Now the only question is how to tell LVGL to round the y coordinates too?
1. Add a new color format as you have suggested and do the vertical rounding [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L294-L299)
2. If a display controller cannot be configured to horizontal mode add an `LV_EVENT_INVALIDATE_AREA` and do the rounding there. 

Which seems better to you?

faxe1008, 2024-10-18:
I'd go with option 2, similar to how 16bit color swap is handled right now.

kisvegabor, 2024-10-19:
I agree! 👍 

I believe `lv_i1_convert_to_vtiled` could be added to LVGL with a detailed comment about the rounding requirements.

PR #7129 (2024-10-30): feat(sw): Add method to convert a htiled I1 buffer to vtiled
Adds a method to convert a monochrome draw buffer from htiled to vtiled orientation.

Resolves #7067.

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvg [... cut]
closed 2024-10-30 by liamHowatt