lvgl/lvgl #7074: Buttonmatrix with Keypad Indev

RamDav, 2024-10-15:
### LVGL version

v9.2.0

### What happened?

Hello,
i tried setting up a buttonmatrix controlled by a Keypad Device with UP/LEFT/DOWN/RIGHT  ENTER and Cancel Keys.
I created a buttonmatrix with two buttons. To detect the press on one button i added an event callback and check it for the `LV_EVENT_VALUE_CHANGE`.
I have set the button control for my buttons via 
`lv_buttonmatrix_set_ctrl_map` to `LV_BUTTONMATRIX_CTRL_CLICK_TRIG` to only send this event when the items has been clicked.

The problem is that as soon as the event handler of the buttonmatrix receives a pressing event the selected button is invalidated.
`buttonmatrix.c` line 457ff 
```c
   else if(code == LV_EVENT_PRESSING) {
        /*If a slid to a new button, discard the current button and don't press any buttons*/
        if(btnm->btn_id_sel != LV_BUTTONMATRIX_BUTTON_NONE) {
            lv_indev_t * indev = lv_event_get_indev(e);
            lv_indev_get_point(indev, &p);
            uint32_t btn_pr = get_button_from_point(obj, &p);
            if(btn_pr != btnm->btn_id_sel) {
                invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the old area*/
                btnm->btn_id_sel = LV_BUTTONMATRIX_BUTTON_NONE;
            }
        }
    }
```
The event checks for a point where the button is pressed but the point is invalid as the active indev is no pointer device.
In the LV_EVENT_PRESSED branch a check is performed prior to getting a the point of the indev. 
But here this receives an invalid point, this point does not match the active button so the button is invalidated.
Shouldn't this check only work if there is an pointer device active?
```c
    else if(code == LV_EVENT_PRESSING) {
        lv_indev_t * indev = lv_event_get_indev(e);
        lv_indev_type_t indev_type = lv_indev_get_type(indev);
        if(indev_type == LV_INDEV_TYPE_POINTER || indev_type == LV_INDEV_TYPE_BUTTON) {
            /*If a slid to a new button, discard the current button and don't press any buttons*/
            if(btnm->btn_id_sel != LV_BUTTONMATRIX_BUTTON_NONE) {
                lv_indev_get_point(indev, &p);
                uint32_t btn_pr = get_button_from_point(obj, &p);
                if(btn_pr != btnm->btn_id_sel) {
                    invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the old area*/
                    btnm->btn_id_sel = LV_BUTTONMATRIX_BUTTON_NONE;
                }
            }
        }
    }
```

### How to reproduce?

A Keyboard indev has to be created.

```c
lv_group_t *group = lv_group_create();
lv_group_set_default(group);
lv_indev_set_group(lv_indev_get_next(NULL), group);
lv_obj_t *buttons = lv_buttonmatrix_create(lv_screen_active());
lv_obj_set_width(buttons, lv_pct(100));
static const char * _Items[] = {
  "Save",
  "Create\nFile",
  NULL};
const lv_buttonmatrix_ctrl_t _ItemsCtrl[] = {
  LV_BUTTONMATRIX_CTRL_NO_REPEAT | LV_BUTTONMATRIX_CTRL_CLICK_TRIG,
  LV_BUTTONMATRIX_CTRL_NO_REPEAT | LV_BUTTONMATRIX_CTRL_CLICK_TRIG,
};
lv_buttonmatrix_set_map(buttons, _Items);
lv_buttonmatrix_set_ctrl_map(buttons, _ItemsCtrl);

lv_obj_add_event_cb(buttons, _buttons_handle, 0, NULL);
lv_group_focus_obj(buttons);

lv_buttonmatrix_set_selected_button(buttons, 0);
```
```c
static void _buttons_handle(lv_event_t *e) { lv_event_code_t c = lv_event_get_code(e);
  if(LV_EVENT_CANCEL == c)
    ApplGUI_LeaveState();
  else if(LV_EVENT_VALUE_CHANGED == c) {
    uint32_t *sel_butt= (uint32_t *)lv_event_get_param(e);
    switch(*sel_butt) {
      case 0:
         // Do Something
         break;
      case 1:
         // Do Something
        break;
    }
  }
}
```

kissa96, 2024-10-16:
Im not sure if I got your problem right.
Technically a pressing event means that the button has been clicked, just the click/press was not released yet.
I've set up a keyboard as indev, and tested your code. For me, it seems to be working exactly the way it is supposed to. Whenever I press the enter, to select a button, i get branched to the correct path in the callback.

Can you elaborate please?

RamDav, 2024-10-16:
Seems to be a duplicated of #7009

RamDav, 2024-10-17:
I replicated the behaviour with the Windows PC Simulation. https://github.com/lvgl/lv_port_pc_visual_studio.
The problem here is that the windows implementation seems to create a lot of Pressed and Released events not pressing events.
In the implementation i have the events generated are LV_PRESSED then LV_PRESSING as long as you keep the ENTER button pressed. And the LV_RELEASED event as the button is released.
This has the issue that on the pc port the LV_VALUE_CHANGED event  callback is called multiple times.

kisvegabor, 2024-10-28:
> Shouldn't this check only work if there is an pointer device active?

That's true! We have points only with pointer and button indev types. 

> This has the issue that on the pc port the LV_VALUE_CHANGED event callback is called multiple times.

Ah, the SDL driver has the same issue. I think it's because we are using a higher level text input API which handles long press repeat. So if you keep pressing the `f` key after a short time it will emulate a lot of `f` clicks.

RamDav, 2024-11-04:
> Ah, the SDL driver has the same issue. I think it's because we are using a higher level text input API which handles long press repeat. So if you keep pressing the f key after a short time it will emulate a lot of f clicks.
Isn't the expected behaviour to repeat the pressed button just as Windows does when a button ispresed?
It just seems that the windows simulation doesn't send them fast enough.
I tried to have long presses in my embedded project. I had to change my input driver to repeatedly send the pressed buttons to get the system to register this as a long press. How is input long pressed supposed to work?

kisvegabor, 2024-11-06:
> How is input long pressed supposed to work?

LVGL needs to sense the PRESSED state for 300 ms.

RamDav, 2024-11-07:
resolved with #7125