lvgl/lvgl #7081: Color Distortion on SPI Display

Spooks4576, 2024-10-15:
### LVGL version

v8.4.1

### What happened?

Currently I'm trying to make a simple splash screen using a image however while the image itself displays correctly the colors inside the image look like "color vomit" I'm using the latest version of ESP IDF what I expect to happen is for the colors of the image to display properly 

also I've tried converting my image to png bmp and jpg with no success using the RGB565A8 format to convert it using the image converter website

below is 2 images . One is the distorted image and the other is the original image

[image]
[image]

### How to reproduce?

```void splash_create(void) {

    lv_obj_t *black_screen = lv_obj_create(lv_scr_act());
    lv_obj_set_style_bg_color(black_screen, lv_color_black(), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(black_screen, 0, LV_PART_MAIN);
    lv_obj_set_style_pad_all(black_screen, 0, LV_PART_MAIN);
    lv_obj_set_style_radius(black_screen, 0, LV_PART_MAIN);
    lv_obj_set_size(black_screen, LV_HOR_RES, LV_VER_RES);

   
    splash_screen = lv_obj_create(black_screen);
    lv_obj_set_size(splash_screen, LV_HOR_RES, LV_VER_RES);
    lv_obj_clear_flag(splash_screen, LV_OBJ_FLAG_SCROLLABLE);
    lv_obj_set_style_bg_opa(splash_screen, LV_OPA_TRANSP, LV_PART_MAIN);
    lv_obj_set_style_border_width(splash_screen, 0, LV_PART_MAIN);
    lv_obj_set_style_pad_all(splash_screen, 0, LV_PART_MAIN);
    lv_obj_set_style_radius(splash_screen, 0, LV_PART_MAIN);

    img = lv_img_create(splash_screen);
    lv_img_set_src(img, &Ghost_ESP);
    lv_obj_align(img, LV_ALIGN_CENTER, 0, 0);
    
    lv_anim_t zoom_anim;
    lv_anim_init(&zoom_anim);
    lv_anim_set_var(&zoom_anim, img);
    lv_anim_set_values(&zoom_anim, 256 / 2, 512 / 2);
    lv_anim_set_time(&zoom_anim, 500);
    lv_anim_set_playback_time(&zoom_anim, 500);
    lv_anim_set_repeat_count(&zoom_anim, 1);
    lv_anim_set_exec_cb(&zoom_anim, zoom_anim_cb);
    lv_anim_set_ready_cb(&zoom_anim, fade_out_cb);
    lv_anim_start(&zoom_anim);
}```

this is currently how I'm adding it to the screen / animating it I'm not sure if this is a config issue, logic issue, or hardware related

Spooks4576, 2024-10-15:
Update: Just Tried Several Different Drivers With No Success ILI9341 ILI9481 86 and 88 all fail to fix the issue with ILI9341 being the most stable but still a issue

kmtlan, 2024-10-16:
The high and low levels of the data are reversed,Changes need to be driven from the bottom up

kmtlan, 2024-10-16:
数据的高8位和低8位反了，需要从底层驱动修改

RamDav, 2024-10-16:
Is the Output which is written to your device in big or little endian? Conversions from libraries little endian to big endian Screens have to be done in the write callback manually

Spooks4576, 2024-10-17:
> Is the Output which is written to your device in big or little endian? Conversions from libraries little endian to big endian Screens have to be done in the write callback manually

i switched the driver to ILI9481 and it fixes it with the exception of artifacts / dead pixels on the right side of the screen 
[image]

Spooks4576, 2024-10-17:
Closing this issue as it was Big Endian has to swap the order for my images Thanks to everyone who commented trying to help

closed 2024-10-17 by Spooks4576