lvgl/lvgl #7090: missing Combining Diacritical Marks render as missing characters

bryghtlabs-richard, 2024-10-16:
### LVGL version

v8.4.0

### What happened?

I would like to render the test phrase "B̶̹̌e̷̘͝ ̵͙̀N̴̹̂o̷̻͛t̴̰̎ ̵̱̓Ą̶͒f̶̳̕r̵̿ͅa̴̧͝ì̷̬d̵̩́", but instead, LVGL's label widget displays:
[image]

Each combining mark is rendered as a missing character. If a combining mark is missing from the font, could we instead skip drawing it?

### How to reproduce?

Render "B̶̹̌e̷̘͝ ̵͙̀N̴̹̂o̷̻͛t̴̰̎ ̵̱̓Ą̶͒f̶̳̕r̵̿ͅa̴̧͝ì̷̬d̵̩́"

bryghtlabs-richard, 2024-10-16:
This might be related to normalization. For the Polish test:

NFC/NFKC: "Zażółć gęślą jaźń" renders as expected:
[image]

NFD/NFKD "Zażółć gęślą jaźń" renders improperly:
[image]

kissa96, 2024-10-17:
Hi, LVGL does not support these characters by default. You have to add them with the font converter.
https://lvgl.io/tools/fontconverter
They are getting rendered, but since nothing is defined for these symbols in the default font, a rectange is made by default.
I don't believe we want to skip rendering for these.

kisvegabor, 2024-10-17:
You can set `LV_USE_FONT_PLACEHOLDER 0` to skip the unknown characters.

bryghtlabs-richard, 2024-10-17:
What I was thinking I wanted was the ability to hide the placeholder only for combining diacritical marks, so it didn't fail as badly, but I would like our product to render these properly if it can, so I think I have two choices:

1. Normalize incoming strings into the same format for which my font is compiled(NFC/NFKC).
2. Update my font configuration to include the corresponding combining diacritical marks.

@kissa96 , you were correct - my font had the pre-combined Polish characters but was missing the combining diacritical marks. So I added them. But it renders wrongly, differently:

[Polish Wikipedia Test Phrase](https://pl.wikipedia.org/wiki/Za%C5%BC%C3%B3%C5%82%C4%87_g%C4%99%C5%9Bl%C4%85_ja%C5%BA%C5%84) in Chrome:
[image]

NFC/NFKC: "Zażółć gęślą jaźń" renders as expected:
[image]

NFD/NFKD "Zażółć gęślą jaźń" renders improperly - the combining marks have shifted to the right by about 1 character.
[image]

bryghtlabs-richard, 2024-10-17:
If you prefer, I can create a new ticket, or rename the title.

kissa96, 2024-10-18:
If your font is NFC compiled, then do not expect with NFD to work correctly. AFAIK LVGL does not handle transformations.

bryghtlabs-richard, 2024-10-18:
The images are with a font compiled with NFC + NFD. I think this is a bug in the typesetter.

kisvegabor, 2024-10-28:
@bryghtlabs-richard could you share your current font file and a very short text (2-3 characters) that I can test with? Please also add an image as reference to see how the text should look like.

bryghtlabs-richard, 2024-10-28:
Here you go @kisvegabor

[LVGL-7090.zip](https://github.com/user-attachments/files/17543647/LVGL-7090.zip)

The test string is the last two letters of "Zażółć gęślą jaźń". I've included a screenshot of it rendered both ways("źń\nźń"), and the UTF8 bytes in hex, as some tools seem to 'fix' it into one form or the other.

In this case, the top line's diacritics appear to be shifted right 1 character, while the bottom line is correct
[image]

kisvegabor, 2024-10-28:
Thank you! I think I could make it work with this change:
```diff
diff --git a/src/draw/lv_draw_label.c b/src/draw/lv_draw_label.c
index d2e5f2a11..b0a73336e 100644
--- a/src/draw/lv_draw_label.c
+++ b/src/draw/lv_draw_label.c
@@ -292,6 +292,7 @@ void lv_draw_label_iterate_characters(lv_draw_unit_t * draw_unit, const lv_draw_
         const char * bidi_txt = dsc->text + line_start;
 #endif
 
+               int32_t letter_w_prev = 0;
         while(next_char_offset < line_end - line_start) {
             uint32_t logical_char_pos = 0;
 
@@ -428,11 +429,22 @@ void lv_draw_label_iterate_characters(lv_draw_unit_t * draw_unit, const lv_draw_
                 draw_letter_dsc.color = dsc->color;
             }
 
+            /*In case of "Combining Diacritical Marks" go back to the previous letter
+             *as these are accent which are drawn on top of the previous letter*/
+            if(letter >= 0x0300 && letter <= 0x036F) {
+               pos.x -= letter_w_prev;
+            }
             draw_letter(draw_unit, &draw_letter_dsc, &pos, font, letter, cb);
 
+            /*"Combining Diacritical Marks" have zero width, so go back to the normal position*/
+            if(letter >= 0x0300 && letter <= 0x036F) {
+               pos.x += letter_w_prev;
+            }
+
             if(letter_w > 0) {
                 pos.x += letter_w + dsc->letter_space;
             }
+            letter_w_prev = letter_w;
         }
 
 #if LV_USE_BIDI

```
[image]

Could you give it a try?

bryghtlabs-richard, 2024-10-29:
I am unfamiliar with how to render unicode CDMs, but this looks good to me and backporting to 8.4.0 fixes the horizontal spacing on my display for both ZN and the whole phrase.

kisvegabor, 2024-10-30:
Great! PR opened: https://github.com/lvgl/lvgl/pull/7190
Please take a final look.

bryghtlabs-richard, 2024-10-30:
Separately, I do wonder if we'll need something similar for capital letters - instead of resetting the X, we might need to adjust the Y based on some characteristic of the previous letter - currently the CDMs seem to always be at the same Y location regardless of base character W/H. But I don't want to hold this up for that.

kisvegabor, 2024-10-30:
You are right! I've just added a test for upper case and it's not perfect. For better alignment we would need more info from the fonts, but that's all we have now  🙁

So I think this was the low hanging fruit that we could add with minimal effort. The next step would be the addition of a full features text shaping engine which is huge a work and was already requested. IT might be added in v9.4.

bryghtlabs-richard, 2024-10-31:
I'm ok with fixing CDMs with lower-case for now, and addressing the upper-case ones later.

kisvegabor, 2024-11-14:
Closing in light of [this](https://github.com/lvgl/lvgl/pull/7190#issuecomment-2467570801).

closed 2024-11-14 by kisvegabor