lvgl/lvgl #7094: How to set more than two fonts in a label,also supports long mode such as LV_LABEL_LONG_SCROLL_CIRCULAR and LV_LABEL_LONG_DOT etc.

Pzhaowei, 2024-10-17:
### Problem to solve

a label can set more than two fonts, and compatible with all original properties

### Success criteria

a label can set more than two fonts, and compatible with all original properties

### Solution outline

None

### Rabbit holes

None

### Testing

None

### Teaching

None

Pzhaowei, 2024-10-17:
> I think you need 2 different labels for 2 different fonts, as you can only have 1 text inside 1 label [Fonts](https://docs.lvgl.io/master/overview/font.html) You can use multiple fonts, and display only the relevant label.
> 
> I'm not sure if I got your problem well. Could you elaborate?

for example, when the value and unit of data with different font on one line text, but I want them to scroll like a same label.
[image]

lizhaoming19980614, 2024-10-17:
I have an idea. You can use an obj as a container, and then use two label, adding animations to these label
this is an example
/******************************************************************************************************************/
static void anim_x_cb(void* var, int32_t v)
{
    lv_obj_set_x(var, v);
}

int app_label_1(lv_obj_t *parent)
{
    lv_obj_t* cont = lv_obj_create(parent);
    lv_obj_set_size(cont, 200, 30);
    lv_obj_clear_flag(cont, LV_OBJ_FLAG_SCROLLABLE);
    lv_obj_center(cont);

    lv_obj_t *label1 = lv_label_create(cont);
    lv_label_set_text(label1, "90");
    lv_obj_set_style_text_font(label1, &lv_font_montserrat_22, LV_PART_MAIN);
    lv_obj_align(label1, LV_ALIGN_LEFT_MID, 0, 0);
    lv_obj_update_layout(label1);
    lv_coord_t width_label1= lv_obj_get_width(label1);

    lv_obj_t *label2 = lv_label_create(cont);  
    lv_label_set_text(label2, "step frequency");
    lv_obj_set_style_text_font(label2, &lv_font_montserrat_14, LV_PART_MAIN);
    lv_obj_align_to(label2, label1, LV_ALIGN_OUT_RIGHT_BOTTOM, 0, 0);
    lv_obj_update_layout(label2);
    lv_coord_t width_label2 = lv_obj_get_width(label2);

    lv_anim_t anim_label1;
    lv_anim_init(&anim_label1);
    lv_anim_set_var(&anim_label1, label1);
    lv_anim_set_values(&anim_label1, lv_obj_get_x(label1), -(width_label1+ width_label2));
    lv_anim_set_time(&anim_label1, 3000);
    lv_anim_set_exec_cb(&anim_label1, anim_x_cb);
    lv_anim_set_path_cb(&anim_label1, lv_anim_path_ease_in);
    lv_anim_set_repeat_count(&anim_label1, LV_ANIM_REPEAT_INFINITE);
    lv_anim_start(&anim_label1);

    lv_anim_t anim_label2;
    lv_anim_init(&anim_label2);
    lv_anim_set_var(&anim_label2, label2);
    lv_anim_set_values(&anim_label2, lv_obj_get_x(label2), -width_label2);
    lv_anim_set_time(&anim_label2, 3000);
    lv_anim_set_exec_cb(&anim_label2, anim_x_cb);
    lv_anim_set_path_cb(&anim_label2, lv_anim_path_ease_in);
    lv_anim_set_repeat_count(&anim_label2, LV_ANIM_REPEAT_INFINITE);
    lv_anim_start(&anim_label2);

    return 0;
}

kisvegabor, 2024-10-17:
You can also 
1. use a [fallback font](https://docs.lvgl.io/master/overview/font.html#use-font-fallback)
2. or when you [generate the font](https://lvgl.io/tools/fontconverter) just add more fonts

Pzhaowei, 2024-10-18:
> I have an idea. You can use an obj as a container, and then use two label, adding animations to these label this is an example [...]

This method needs to calculate whether the length of the two labels exceeds the width of the container, which is not convenient, and it is not compatible with all long modes

MacDragon, 2024-10-20:
> > You can also
> > 
> > 1. use a [fallback font](https://docs.lvgl.io/master/overview/font.html#use-font-fallback)
> > 2. or when you [generate the font](https://lvgl.io/tools/fontconverter) just add more fonts
> 
> It's not a font problem, I hope it can set more than two fonts on a same label so that it can scroll like a whole

They meant that assuming the charecters want to use from the two different fonts aren't overlapping, they could be combined into a single font with superset.

closed 2024-11-12 by lvgl-bot