lvgl/lvgl #7098: v9.2.1 Release Discussion

kisvegabor, 2024-10-17:
### Introduce the problem

v9.2.1 will be released in Ocrober. Let's discuss all the related topic here.  

@espzav @nicusorcitu @cristian-stoica @faxe

kisvegabor, 2024-10-17:
Still need to resolve the performance issues because of recursive mutex. See [here](https://github.com/lvgl/lvgl/pull/6981#issuecomment-2417037919).

cederom, 2024-10-18:
@kisvegabor would it be possible to already create the `release/v9.2.1` branch so we could also merge code and perform tests on NuttX 12.7.0-RC (see https://github.com/apache/nuttx/issues/13825)? We are waiting with a release only for LVGL now. With that branch created we will be able to work further on NuttX 12.7.0 and LVGL 9.2.1 integration. Performance regression fixes in LVGL could be then backported to v9.2.1 branch and then we just switch commit while the rest of the work will be done :-)

kisvegabor, 2024-10-18:
@cederom we don't create new branches for patch versions. v9.2.1 will be released in the `release/v9.2` branch. 99% of the patches are already there so it should be good to test. :slightly_smiling_face:

sccoouut, 2024-10-21:
maybe a dumb question, but anyway. will #6686 part of 9.2.1?

cederom, 2024-10-21:
>@sccoouut: maybe a dumb question, but anyway. will #6686 part of 9.2.1?

This is part of LVGL project, you should ask there, from what I can see this commit landed in lvgl master but not release/v9.2, so probably no, and I dont think they will add it now, last bugfixes and release is planned within days..

cederom, 2024-10-21:
> @sccoouut: maybe a dumb question, but anyway. will #6686 part of 9.2.1?

I can see this commit in lvgl/master but not in lvgl/release/v9.2 branch so probably not in 9.2.1 but next release.. 9.2.1 is almost out within days :-)

nicusorcitu, 2024-10-21:
@kisvegabor We have two more PR opened: https://github.com/lvgl/lvgl/pull/7131 and https://github.com/lvgl/lvgl/pull/7071
Please let us know if these can get into 9.2.1 - in my opinion is no breakage of API neither a new functionality - just some fixes.
@cosmindanielradu19 @cristian-stoica

KlemenDEV, 2024-10-22:
In what release is NeoChrom planned? What is the ETA for this release (assuming such info is public)?

kisvegabor, 2024-10-23:
It will be part of v9.3 planned for early 2025.

nicusorcitu, 2024-10-23:
@kisvegabor when do you intend to make the release 9.2.1?

kisvegabor, 2024-10-23:
Yes, I reverted 53c846f4a. Please test it again, if it's ok I'll make the release tomorrow.

kisvegabor, 2024-10-24:
@nicusorcitu will you have time to test it today?

nicusorcitu, 2024-10-24:
Yes, @cosmindanielradu19 is on it. By today he will post the latest result from `release/9.2` branch so that everything is clear.

nicusorcitu, 2024-10-24:
@kisvegabor please take this PR as well.

cederom, 2024-10-24:
BIG THANK YOU @kisvegabor AND ALL CONTRIBUTORS!! =)

kisvegabor, 2024-10-25:
**KCONFIG ISSUES REPORTED**
A seemingly innocent last minute commit broke Kconfig on ESP and potentially on other platforms too. I'm sorry guys. We should take the feature freeze week more seriously... 
  
See https://github.com/lvgl/lvgl/pull/7131#issuecomment-2437551657

@cederom @faxe1008 Does it affect NuttX and Zephyr too?

faxe1008, 2024-10-25:
@kisvegabor this is part of the `lvgl_conf.h` on zephyr side. This should not break things on our side.

kisvegabor, 2024-10-25:
Great! Thank you for the quick answer @faxe1008.

cederom, 2024-10-25:
We have merged LVGL v9.2.1 to NuttX master and 12.7.0 release branch, @XuNeo was author of https://github.com/apache/nuttx-apps/pull/2592, did not report problems with branch releases/v9.2 and 9.2.1 release package.. I think we use our own configuration mechanism that is not affected, could you please verify and confirm @XuNeo ? :-)

XuNeo, 2024-10-26:
Confirmed that nuttx has no issue.

kisvegabor, 2024-10-26:
Thank you Neo! 

However I wonder why. Isn't LV_ATTRIBUTE_LARGE_CONST is exported as an empty string? If so it should cause build error. If not, why not?

XuNeo, 2024-10-26:
Hi Gabor,

The last-minute change is not included in nuttx's Kconfig file(which is copied out from lvgl/Kconfig)
I'll call it a lucky. It should be updated, but I didn't notice the last-minute change.

kisvegabor, 2024-10-26:
Oh, I see. Just to confirm: if it were updated, it were broken too, right?

XuNeo, 2024-10-26:
Yes. `LV_ATTRIBUTE_MEM_ALIGN` is defined as str in Kconfig, thus it cause issue in below code.

```
static const
LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST LV_ATTRIBUTE_IMG_BENCHMARK_LVGL_LOGO_ARGB
uint8_t img_benchmark_lvgl_logo_argb_map[]
```

The correct usage should be like this one, `LV_ATTRIBUTE_MEM_ALIGN` should be defined in CFLAG from `CONFIG_LV_ATTRIBUTE_MEM_ALIGN`.

https://github.com/apache/nuttx-apps/blob/1506c6f54dcf6f969e3e8e4e2fed805bfac0b4a1/graphics/lvgl/Makefile#L38-L44

cosmindanielradu19, 2024-10-28:
> @faxe1008 @nicusorcitu @XuNeo I've reverted the Kconfig change in `release/v9.2`. Please make a new test, we won't merge anything from now. If it works well, I'll release v9.2.2.

It works well on our side

XuNeo, 2024-10-28:
No issue from NuttX.

kisvegabor, 2024-10-29:
Thank you all. We have checked on ESP too and it worked well (@espzav).
Just waiting for the CMSIS PACK updated from @GorgonMeducer and we are good to go. 

Targeting tomorrow with v9.2.2.

GorgonMeducer, 2024-10-29:
Done @kisvegabor 
https://github.com/lvgl/lvgl/pull/7177

nicusorcitu, 2024-10-30:
> I wonder how it passed through as we have an ESP-IDF CI test running for MicroPython.
>
@kisvegabor I agree with you that:
1. We should have a feature freeze to avoid such situations. It worked on our case but broke others build.
2. We should have a better CI/CD - why only 9 checks have been done for PR https://github.com/lvgl/lvgl/pull/7131?! Just because there was no .c file changed?

Sorry to everyone for the trouble with that patch.

Thanks,
Nicu.

kisvegabor, 2024-10-30:
> We should have a feature freeze to avoid such situations. It worked on our case but broke others build.

Absolutely! 

> We should have a better CI/CD - why only 9 checks have been done for PR

Honestly, I don't know. :slightly_frowning_face: 
We will add a CI for ESP32 build, but I wonder if we can do the same for other vendors. @nicusorcitu can we build an MCUXpresso project from a CI Action?

kisvegabor, 2024-10-30:
I've just released v9.2.2

cosmindanielradu19, 2024-10-30:
> > We should have a feature freeze to avoid such situations. It worked on our case but broke others build.
> 
> Absolutely!
> 
> > We should have a better CI/CD - why only 9 checks have been done for PR
> 
> Honestly, I don't know. 🙁 We will add a CI for ESP32 build, but I wonder if we can do the same for other vendors. @nicusorcitu can we build an MCUXpresso project from a CI Action?

Hi @kisvegabor , unfortunately it's not possible to build an MCUXpresso project from an CI Action.

The good part is that we have the support for armgcc build integrated in our SDK examples, and that should work with the CI Action.

kisvegabor, 2024-10-30:
> The good part is that we have the support for armgcc build integrated in our SDK examples, and that should work with the CI Action.
 
 That's also good. We can prepare an "NXP reference project" repo which is using in CI.

GorgonMeducer, 2024-10-31:
Please have a look at #7201 @kisvegabor 
Thank you.

kisvegabor, 2024-11-05:
As v9.2.2 looks good, I'm closing this issue now.

closed 2024-11-05 by kisvegabor

cederom, 2024-11-05:
BIG THANK YOU @kisvegabor LVGL ROX!! =)