lvgl/lvgl #7105: No such file or directory: #include <process.h>

BenGeoSim, 2024-10-18:
### LVGL version

v8.3.10

### What happened?

Sorry I don't think true issue. I am using 8.3.10 and getting "no such file process.h". I am using old version to match tutorial, is there a dependancy I am missing? Thanks!

kissa96, 2024-10-18:
<process.h> is a predefined header, not an LVGL header.
Which function relies on it?

BenGeoSim, 2024-10-21:
Pardon my ignorance: it comes up as an error when I try and compile with the Arduino IDE.
This is for a Waveshare product so I think it will beyond the scope of this GitHub repo, as they provide library files which may break the compilation in unexpected ways. The error is called in the file lvgl/src/osal/lv_windows.c

kissa96, 2024-10-21:
Which board are you using?
If this is not LVGL related, can you please close the issue as resolved? Thx in advance

BenGeoSim, 2024-10-23:
I'll close it now, I'm using this board:
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-2.1#Overview

closed 2024-10-23 by BenGeoSim

liamHowatt, 2025-03-13:
Where is the library coming from? I can't find `#include <process.h>` in https://downloads.arduino.cc/libraries/github.com/lvgl/lvgl-8.3.10.zip

TheKerbecs, 2025-03-13:
It is the library from the demo code for Waveshare LCD screens.

https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4#Demo

liamHowatt, 2025-03-14:
Hmm when I download and extract https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-4/ESP32-S3-Touch-LCD-4_democode.zip and https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-4/ESP32-S3-Touch-LCD-4_democode_V2.zip and run `grep -r 'process\.r'` there are no results.

What's the full Arduino compiler error message?

TheKerbecs, 2025-03-14:
c:\Users\radle\OneDrive\Dokumente\Arduino\libraries\lvgl\src\osal\lv_windows.c:14:10: fatal error: process.h: No such file or directory
   14 | #include <process.h>
      |          ^~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

I gave that to ChatGPT and it told me to rename the file to .bak at the end. That solved the issue. 

Now when i wanted to look for the osal folder it was deleted, i dont remember if i did that or it was done automatically. I managed to recover it via OneDrive. 
Here is the link for it
https://1drv.ms/f/s!ApIWOyc1W43-7qdaQXsbZY1A7JLmhA?e=ccD03v