lvgl/lvgl #7114: _lv_cache_entry_t definition/declaration oddity

kdschlosser, 2024-10-19:
### Introduce the problem

the following code is in the public scope
```c
struct _lv_cache_entry_t {
    const lv_cache_t * cache;
    int32_t ref_cnt;
    uint32_t node_size;

    bool is_invalid;
};
```

and the following code is in the private scope

```c
struct _lv_cache_entry_t;

typedef struct _lv_cache_entry_t lv_cache_entry_t;
```

something got messed up when refactoring the API.

### Proposal

this should be declared in the public scope because `lv_cache_entry_t` is used in a slew of public headers
```c
typedef struct _lv_cache_entry_t lv_cache_entry_t;
```

and the following should be in the private scope
```c
struct _lv_cache_entry_t {
    const lv_cache_t * cache;
    int32_t ref_cnt;
    uint32_t node_size;

    bool is_invalid;
};
```

kdschlosser, 2024-10-19:
Now, in order for the above to be working properly that would mean there is a private header file included in a public header file somewhere and that should not be done. That is going to expose the private items into the public scope.

kdschlosser, 2024-10-19:
private includes in public headers

lv_global.h
```c
#include "../font/lv_font_fmt_txt_private.h"
```

lv_global.h
```c
#include "../misc/lv_timer_private.h"
#include "../misc/lv_anim_private.h"
#include "../tick/lv_tick_private.h"
#include "../draw/lv_draw_buf_private.h"
#include "../draw/lv_draw_private.h"
#include "../draw/sw/lv_draw_sw_private.h"
#include "../draw/sw/lv_draw_sw_mask_private.h"
#include "../stdlib/builtin/lv_tlsf_private.h"
#include "../others/sysmon/lv_sysmon_private.h"
#include "../layouts/lv_layout_private.h"
```

lv_draw_nema_gfx.h
```c
#include "../lv_draw_private.h"
#include "../lv_draw_buf_private.h"
#include "../lv_draw_image_private.h"
#include "../lv_image_decoder_private.h"
#include "../lv_draw_label_private.h"
#include "../lv_draw_mask_private.h"
#include "../lv_draw_rect_private.h"
#include "../lv_draw_triangle_private.h"
#include "../lv_draw_vector_private.h"

#include "../../misc/lv_area_private.h"
```

lv_draw_pxp.h
```c
#include "../../sw/lv_draw_sw_private.h"
#include "../../../misc/lv_area_private.h"
```

lv_draw_vglite.h
```c
#include "../../lv_draw_private.h"
#include "../../sw/lv_draw_sw_private.h"
#include "../../../misc/lv_area_private.h"
```

lv_draw_dave2d.h
```c
#include "../../lv_draw_private.h"
```

lv_draw_sw_arm2d.h
```c
#include "../../../misc/lv_area_private.h"
```

lv_draw_vg_lite_type.h
```c
#include "../lv_draw_private.h"
```

lv_svg_render.h
```c
#include "../../draw/lv_draw_vector_private.h"
```

lv_cache.h
```c
#include "lv_cache_private.h"
```

lv_cache_entry.h
```c
#include "lv_cache_private.h"
```

lv_cache_lru_rb.h
```c
#include "lv_cache_private.h"
```

The reason why you cannot import a private header file in a public one is because everything that is in that private header has now just been brought into the public scope. This causes an issue with bindings and exposing things that shouldn't be exposed. If there are any private header files that are included inside of a private header file now everything exposed in that private header is also in the public scope.

You have to remember that the code generation script recursively follows includes and reads the contents of the included files. so when a private header file gets included inside of a public one everything that is inside that private header file not gets read and parsed by the code generator when it really shouldn't be. The whole purpose to having the private header files is to not have code get generated for things that are for internal use. It increases the space it takes up, it also increases the memory use depending on the binding (MicroPython) and it can impact performance at the processor as well (MicroPython).

kdschlosser, 2024-10-19:
This is more than likely why there is an issue that has been brought up in #6418

kdschlosser, 2024-10-19:
and to comment on this a bit further.

if a structure is declared in a header file it should also be defined in a header file and not in a source file. 

lv_cache_private.h
```c
struct _lv_cache_entry_t;

typedef struct _lv_cache_entry_t lv_cache_entry_t;
```

lv_cache_entry.c 
```c
struct _lv_cache_entry_t {
    const lv_cache_t * cache;
    int32_t ref_cnt;
    uint32_t node_size;

    bool is_invalid;
};
```

Making something private is one thing. the private header files are able to be read and just omitted from a binding. In the case above `lv_cache_entry_t` is used in many places in the public API but the structure `_lv_cache_entry_t` is defined in a c source file. There is no way for a high level language binding script to be able to determine the size of `_lv_cache_entry_t` because it is not visible.

You end up with erroneous code generation like this...

```c
static void mp_lv_cache_entry_t_attr(mp_obj_t self_in, qstr attr, mp_obj_t *dest)
{
    mp_lv_struct_t *self = MP_OBJ_TO_PTR(self_in);
    GENMPY_UNUSED lv_cache_entry_t *data = (lv_cache_entry_t*)self->data;

    if (dest[0] == MP_OBJ_NULL) {
        // load attribute
        switch(attr)
        {
            ;
            default: call_parent_methods(self_in, attr, dest); // fallback to locals_dict lookup
        }
    } else {
        if (dest[1])
        {
            // store attribute
            switch(attr)
            {
                ;
                default: return;
            }

            dest[0] = MP_OBJ_NULL; // indicate success
        }
    }
}
```

`lv_anim_timeline_t` is another that is declared in the public scope but is defined in a source file. 

in the case of `lv_cache_entry_t`  and `_lv_cache_entry_t` bot are declared in a private header file and `_lv_cache_entry_t` is defined in a source file. `lv_cache_entry_t` is used in many places in the public api. How is the binding code supposed to be able to know what exactly `lv_cache_entry_t` is? It's not able to hence the bad code above,

liamHowatt, 2024-10-22:
I have to confess that I was aware of this type leaking into the public API. I could not see a way to address it at the time and it slipped off my radar. Your proposal looks good if you want to open a PR. It might not be trivial so I can take look if it's a lot of work.

kdschlosser, 2024-10-22:
I think that if there is a need to have the functions that take that type as a parameter or return that type have a specific need to be public then we can simply do a forward declaration of the type and define the type in a private header. This would take care of the issue. 

But we really need to have a look at those functions ans see if they really need to be in the public scope in the first place. I don't know enough about the internal mechanics of LVGL to be able to determine that. I personally don't see where there is a need to have public access to the cached items but perhaps @kisvegabor might be able to assist with this letting us know if there is a need to. Maybe you know if there is a need as well.

My concern is the cascading effect of including a private header inside a public header which could cause a whole slew of private types and functions to end up in the public scope.

kisvegabor, 2024-11-25:
Closing as #7220 is merged.

closed 2024-11-25 by kisvegabor