lvgl/lvgl #7117: There are things that have been made private that really shouldn't be.

kdschlosser, 2024-10-19:
### Introduce the problem

some of the functions listed below should probably be in the public scope.
`lv_text_get_next_line`
`lv_text_ins`
`lv_text_cut`
`lv_text_set_text_vfmt`
`lv_text_encoded_letter_next_2`
`lv_text_is_break_char`
`lv_text_is_a_word`
`lv_text_is_marker`
`lv_text_encoded_size`
`lv_text_unicode_to_encoded`
`lv_text_encoded_conv_wc`
`lv_text_encoded_next`
`lv_text_encoded_prev`
`lv_text_encoded_get_byte_id`
`lv_text_encoded_get_char_id`
`lv_text_get_encoded_length`

`lv_cache_entry_reset_ref`
`lv_cache_entry_inc_ref`
`lv_cache_entry_dec_ref`
`lv_cache_entry_set_node_size`
`lv_cache_entry_set_invalid`
`lv_cache_entry_release_data`
`lv_cache_entry_acquire_data`
`lv_cache_entry_set_cache`

`lv_image_buf_get_transformed_area`

`lv_inv_area`  private function but it is in the API map `#define _lv_inv_area lv_inv_area`???

`lv_refr_get_disp_refreshing`
`lv_display_refr_timer`

`lv_obj_style_create_transition`
`lv_obj_style_state_compare`

# these next 2 would be handy to have access to
`lv_indev_find_scroll_obj`
`lv_obj_scroll_by_raw`

`lv_obj_get_ext_draw_size`

# these would be very handy to have access to...
`lv_area_set_pos`
`lv_area_intersect`
`lv_area_diff`
`lv_area_join`
`lv_area_is_point_on`
`lv_area_is_on`
`lv_area_is_in`
`lv_area_is_out`
`lv_area_is_equal`

private structures but have public function to only do something with an empty structure because there are no functions to fill the structure with any data
=========

`lv_draw_mask_rect_dsc_t` is declared in the public scope

```c
typedef struct _lv_draw_mask_rect_dsc_t lv_draw_mask_rect_dsc_t;
```

`_lv_draw_mask_rect_dsc_t` is defined in the private scope

There are initialize functions and draw functions for the structure.
```c
void /* LV_ATTRIBUTE_FAST_MEM */ lv_draw_mask_rect_dsc_init(lv_draw_mask_rect_dsc_t * dsc);
void lv_draw_mask_rect(lv_layer_t * layer, const lv_draw_mask_rect_dsc_t * dsc);
```

But there is no way to set the field below that are in the structure??
```c
    lv_draw_dsc_base_t base;
    lv_area_t area;
    int32_t radius;
```

`lv_grad_t` is declared in the public scope
```c
typedef struct _lv_grad_t lv_grad_t;
```

`_lv_grad_t` is defined in the private scope

There is no way to get or set the fields below.

```c
    lv_color_t   *  color_map;
    lv_opa_t   *  opa_map;
    uint32_t size;
```

`lv_draw_sw_blend_dsc_t` is declared in the public scope
```c
typedef struct _lv_draw_sw_blend_dsc_t lv_draw_sw_blend_dsc_t;
```

`_lv_draw_sw_blend_dsc_t ` is defined in the private scope

There is a public function to draw the structure

```c
void lv_draw_sw_blend(lv_draw_unit_t * draw_unit, const lv_draw_sw_blend_dsc_t * dsc);
```

but there is no way to get or set the fields below
```c
    const lv_area_t * blend_area; 
    const void * src_buf;
    uint32_t src_stride;
    lv_color_format_t src_color_format;
    const lv_area_t * src_area;
    lv_opa_t opa;
    lv_color_t color;
    const lv_opa_t * mask_buf;
    lv_draw_sw_mask_res_t mask_res;
    const lv_area_t * mask_area;
    int32_t mask_stride;
    lv_blend_mode_t blend_mode; 
```

The list goes on and on....

I am thinking there was some over zealous moving of functions and structures into the private scope. there was not enough looking into each item to determine if there would be zero uses publicly.

liamHowatt, 2024-10-22:
The `text` API visibility has been discussed before. We don't want users of the library depending on it without being aware of the volatility of the API in the future. Same with `cache`. I could see `area` becoming public. At the time area public/private was just mapped directly based on whether the functions began with an underscore, as were most public/private decisions.

One I didn't see you mention that should be public: `lv_span_t` needs to be public or get a getter+setter for its `lv_style_t style` member, which is intended to be used by the user.

kdschlosser, 2024-10-22:
The list of functions that have been made private is a very long list so I only threw a bunch out there to use as an example. There are a mess of functions where I can easily see a use case for them being in the public scope. I have not personally used the text functions and they were at the top of the list so that is why they are listed here. Some of them I can see as possibly being helpful but not all of them..  The ones listed below are the ones that would possibly have some benefit being in the public scope. 

```
lv_text_get_next_line
lv_text_ins
lv_text_cut
lv_text_set_text_vfmt
 lv_text_is_a_word
```

while I know that a user would be able to implement these functions in their own code it ends up being duplicative code on a resource constrained device which makes it not ideal to have to do that. The smart thing to do is to make them public so the user is able to use the existing code and not have to duplicate it, do you agree?

I know you were jumping in blindly to the whole API reworking and you had not gotten your feet wet so to speak with the LVGL code base. LVGL's API is very large and can be very overwhelming at first which is the reason why there is the push to have the private and public scopes. How this was done prior was by adding the leading underscore and that was done because of the MicroPython binding and how Python makes things private. While the binding uses LVGL there should be a very minimal amount of things that are binding specific in the LVGL code base. That includes things like having the leading underscore. 

If you want I can put together a list of all of the private functions and we can post that list to allow users to comment on which ones might have some benefit being in the public scope. This might be the best way to go about identifying what should be moved. The area functions I feel should be in the public scope and I think you agree with that as well. This is something that can be done in steps so we don't end up with merge conflict issues when making the changes. I am sure the merge conflicts were a huge headache for you to deal with when you did the initial API change. I tip my hat to you for having to deal with that and how well you did with it.

liamHowatt, 2024-10-24:
As you say, it's not ideal to have multiple near-identical utf-8 handling libraries on a flash-constrained device. I think it's kept private because it's not a generally applicable utf-8 handling library. It works well for the scope of LVGL but cannot claim to be general. If the API becomes public, we lose the freedom refuse PRs that grow the text API beyond the scope of its usage within LVGL.

Just reiterating the original explanation.

> If you want I can put together a list of all of the private functions and we can post that list to allow users to comment on which ones might have some benefit being in the public scope. This might [...]

I think this is a good way to progress. Thanks.

kdschlosser, 2024-10-24:
again I have not use any of the text functions personally. I know that dealing with strings and different encoding is almost every developers worst nightmare because the entire encoding standard has been slapped together over a very long period of time without any thought to the future. So there really isn't a standard so to speak because it's a complete mess. I blame Microsoft for this because it is how they did things in the 80's and they are the ones that were ahead of everyone with internationalization of their software.  I swear that everything Microsoft has their fingers into ends up being a coding nightmare. 

My point is you may be correct about leaving the text API stuff private and letting the user "roll their own" because trying to cover all bases so that the functions would work for all use cases would bloat the hell out of LVGL big time.

closed 2024-11-18 by lvgl-bot