lvgl/lvgl #7126: Missing include file when enabling nema GPU support

hamztouiz, 2024-10-21:
### LVGL version

v9.3.0

### What happened?

When I compile with nema GPU on a stm32ua5a9j support enabled on a stm32ua5a9j, the compiler throws an error indicating that a specific include file is missing:

in  lv_draw_nema_gfx_hal.c : #include "config.h" at line 19

I checked [another project](https://github.com/lvgl/lv_port_riverdi_stm32u5/pull/1) that also uses this feature, and I found that the file in question exists in their setup : https://github.com/lvgl/lv_port_riverdi_stm32u5/blob/nema-test/STM32CubeIDE/Middlewares/NemaGFX/templates/config.h

I missing something ?

### How to reproduce?

enable on stm32

kissa96, 2024-10-21:
We expect config.h to reside in a predefined location relative to lv_draw_nema_gfx_hal.c
Is config.h in the correct path, or is it even in the project?

hamztouiz, 2024-10-21:
I can't find it in the project, nor a file that appears.

kissa96, 2024-10-21:
How did you create the project? Nema config files are not a part of the LVGL lib, we are just relying on them for the HAL and APIs

liamHowatt, 2024-10-21:
I think we may have made an incorrect assumption when releasing this feature that users will have some important files such as config.h already in their project. You can see in the `lv_port_riverdi_stm32u5` project you linked [where config.h is.](https://github.com/liamHowatt/lv_port_riverdi_stm32u5/blob/nema/STM32CubeIDE/Middlewares/NemaGFX/templates/config.h)

[This file, lv_draw_nema_gfx_hal.c](https://github.com/lvgl/lvgl/blob/master/src/draw/nema_gfx/lv_draw_nema_gfx_hal.c) is part of the LVGL source tree but it was moved from the nema GFX `templates` directory of `lv_port_riverdi_stm32u5`. We mention in the docs that if you already have a hal implemented in your `templates` folder, it will conflict with the one you may already have. You have the opposite problem where you don't have any files yet.

https://docs.lvgl.io/master/overview/renderers/nema_gfx.html

> "src/draw/nema_gfx/lv_draw_nema_gfx_hal.c" implements the HAL functionality required by Nema to allocate memory and lock resources (in this implementation, no locking is done). It may conflict with existing definitions if you have an existing Nema HAL implementation. You may simply be able to remove yours.

In your case, since you do not have config.h, you can try taking the one I linked (first link in this comment). Maybe a PR should be opened to move it into the LVGL source tree.

hamztouiz, 2024-10-22:
> How did you create the project? Nema config files are not a part of the LVGL lib, we are just relying on them for the HAL and APIs

So I created the project: 
1) from scratch by selecting my devkit (stm32u5g9j)
2) activated the gpu from the stm32 IDE
3) add lvgl as described here https://docs.lvgl.io/master/overview/renderers/nema_gfx.html

hamztouiz, 2024-10-22:
Thank you for your answer, so if Nema config files are not a part of the LVGL lib, I think it would be interesting to add it in a PR :

- add the config file as a template
- metion in the documentation 

If you agree, I can open the PR

liamHowatt, 2024-10-24:
Okay, I'm opening a PR with docs changes and configurations regarding the HAL file very soon anyways. I will move `config.h` into LVGL and name it `lv_draw_nema_gfx_config.h`.

PR #7174 (2024-11-06): chore(NemaGFX): add missing header, correct the docs, provided HAL is optional
Fixes #7126

- Add the missing `config.h` header. Rename and edit a STM32 one. Removed references to TouchGFX.
- Differentiate Nema GFX from ST's NeoChrom GPU in docs descriptions.
- Optionally disable the STM32-based HAL if the user has a different one.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
 [... cut]
closed 2024-11-06 by FASTSHIFT