lvgl/lvgl #7132: bug(file_explorer) use-after-free related to the table cells

C47D, 2024-10-22:
My of my own applications is not crashing anymore after these changes. Nice!

There is another issue I've been seeing. There's a use-after-free bug detected when the address sanitizer is enabled. It's related to the table cells. Don't try to fix it in this PR. Just fyi.

<details>
    <summary>Crash info</summary>

    ```
    ==51600==ERROR: AddressSanitizer: heap-use-after-free on address 0x6040000602e0 at pc 0x7f53c69fe070 bp 0x7ffcceb52870 sp 0x7ffcceb52018
    READ of size 14 at 0x6040000602e0 thread T0
        #0 0x7f53c69fe06f in __interceptor_strcpy ../../../../src/libsanitizer/asan/asan_interceptors.cc:430
        #1 0x5609a29bf291 in lv_strcpy /home/liam/lvgl/lvgl/src/stdlib/clib/lv_string_clib.c:81
        #2 0x5609a2a3f145 in copy_cell_txt /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:1057
        #3 0x5609a2a36a0f in lv_table_set_cell_value /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:119
        #4 0x5609a2b1a976 in exch_table_item /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:14
        #5 0x5609a2b1ab58 in sort_by_file_kind /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:36
        #6 0x5609a2b1ac83 in file_explorer_event_handler /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:60
        #7 0x5609a29b3ced in lv_event_send /home/liam/lvgl/lvgl/src/misc/lv_event.c:98
        #8 0x5609a2a5b21c in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:367
        #9 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #10 0x5609a2b1c9c6 in show_dir /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:617
        #11 0x5609a2b1bedd in browser_file_event_handler /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:531
        #12 0x5609a29b3ced in lv_event_send /home/liam/lvgl/lvgl/src/misc/lv_event.c:98
        #13 0x5609a2a5b21c in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:367
        #14 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #15 0x5609a2a3af00 in lv_table_event /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:595
        #16 0x5609a2a59f41 in lv_obj_event_base /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:89
        #17 0x5609a2a5b1b1 in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:364
        #18 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #19 0x5609a29a2bb3 in send_event /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1740
        #20 0x5609a29a01b9 in indev_proc_release /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1359
        #21 0x5609a299b742 in indev_pointer_proc /home/liam/lvgl/lvgl/src/indev/lv_indev.c:725
        #22 0x5609a2998a44 in lv_indev_read /home/liam/lvgl/lvgl/src/indev/lv_indev.c:251
        #23 0x5609a2992748 in lv_sdl_mouse_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_mouse.c:201
        #24 0x5609a2993a59 in sdl_event_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_window.c:272
        #25 0x5609a29bdd8f in lv_timer_exec /home/liam/lvgl/lvgl/src/misc/lv_timer.c:325
        #26 0x5609a29bcad9 in lv_timer_handler /home/liam/lvgl/lvgl/src/misc/lv_timer.c:107
        #27 0x5609a2961d60 in main /home/liam/lvgl/lvgl_workspace/main.c:3396
        #28 0x7f53c54e9082 in __libc_start_main ../csu/libc-start.c:308
        #29 0x5609a2961c2d in _start (/home/liam/lvgl/lvgl_workspace/build/lvgl_workspace+0x33c2d)

    0x6040000602e0 is located 16 bytes inside of 38-byte region [0x6040000602d0,0x6040000602f6)
    freed by thread T0 here:
        #0 0x7f53c6a6fc3e in __interceptor_realloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cc:163
        #1 0x5609a29bee5e in lv_realloc_core /home/liam/lvgl/lvgl/src/stdlib/clib/lv_mem_core_clib.c:69
        #2 0x5609a29bf54d in lv_realloc /home/liam/lvgl/lvgl/src/stdlib/lv_mem.c:150
        #3 0x5609a2a3685d in lv_table_set_cell_value /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:115
        #4 0x5609a2b1a976 in exch_table_item /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:14
        #5 0x5609a2b1ab58 in sort_by_file_kind /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:36
        #6 0x5609a2b1ac83 in file_explorer_event_handler /home/liam/lvgl/lvgl/examples/others/file_explorer/lv_example_file_explorer_3.c:60
        #7 0x5609a29b3ced in lv_event_send /home/liam/lvgl/lvgl/src/misc/lv_event.c:98
        #8 0x5609a2a5b21c in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:367
        #9 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #10 0x5609a2b1c9c6 in show_dir /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:617
        #11 0x5609a2b1bedd in browser_file_event_handler /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:531
        #12 0x5609a29b3ced in lv_event_send /home/liam/lvgl/lvgl/src/misc/lv_event.c:98
        #13 0x5609a2a5b21c in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:367
        #14 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #15 0x5609a2a3af00 in lv_table_event /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:595
        #16 0x5609a2a59f41 in lv_obj_event_base /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:89
        #17 0x5609a2a5b1b1 in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:364
        #18 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #19 0x5609a29a2bb3 in send_event /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1740
        #20 0x5609a29a01b9 in indev_proc_release /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1359
        #21 0x5609a299b742 in indev_pointer_proc /home/liam/lvgl/lvgl/src/indev/lv_indev.c:725
        #22 0x5609a2998a44 in lv_indev_read /home/liam/lvgl/lvgl/src/indev/lv_indev.c:251
        #23 0x5609a2992748 in lv_sdl_mouse_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_mouse.c:201
        #24 0x5609a2993a59 in sdl_event_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_window.c:272
        #25 0x5609a29bdd8f in lv_timer_exec /home/liam/lvgl/lvgl/src/misc/lv_timer.c:325
        #26 0x5609a29bcad9 in lv_timer_handler /home/liam/lvgl/lvgl/src/misc/lv_timer.c:107
        #27 0x5609a2961d60 in main /home/liam/lvgl/lvgl_workspace/main.c:3396
        #28 0x7f53c54e9082 in __libc_start_main ../csu/libc-start.c:308

    previously allocated by thread T0 here:
        #0 0x7f53c6a6fc3e in __interceptor_realloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cc:163
        #1 0x5609a29bee5e in lv_realloc_core /home/liam/lvgl/lvgl/src/stdlib/clib/lv_mem_core_clib.c:69
        #2 0x5609a29bf54d in lv_realloc /home/liam/lvgl/lvgl/src/stdlib/lv_mem.c:150
        #3 0x5609a2a37092 in lv_table_set_cell_value_fmt /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:183
        #4 0x5609a2b1c4ff in show_dir /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:585
        #5 0x5609a2b1bedd in browser_file_event_handler /home/liam/lvgl/lvgl/src/others/file_explorer/lv_file_explorer.c:531
        #6 0x5609a29b3ced in lv_event_send /home/liam/lvgl/lvgl/src/misc/lv_event.c:98
        #7 0x5609a2a5b21c in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:367
        #8 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #9 0x5609a2a3af00 in lv_table_event /home/liam/lvgl/lvgl/src/widgets/table/lv_table.c:595
        #10 0x5609a2a59f41 in lv_obj_event_base /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:89
        #11 0x5609a2a5b1b1 in event_send_core /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:364
        #12 0x5609a2a59cf8 in lv_obj_send_event /home/liam/lvgl/lvgl/src/core/lv_obj_event.c:67
        #13 0x5609a29a2bb3 in send_event /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1740
        #14 0x5609a29a01b9 in indev_proc_release /home/liam/lvgl/lvgl/src/indev/lv_indev.c:1359
        #15 0x5609a299b742 in indev_pointer_proc /home/liam/lvgl/lvgl/src/indev/lv_indev.c:725
        #16 0x5609a2998a44 in lv_indev_read /home/liam/lvgl/lvgl/src/indev/lv_indev.c:251
        #17 0x5609a2992748 in lv_sdl_mouse_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_mouse.c:201
        #18 0x5609a2993a59 in sdl_event_handler /home/liam/lvgl/lvgl/src/drivers/sdl/lv_sdl_window.c:272
        #19 0x5609a29bdd8f in lv_timer_exec /home/liam/lvgl/lvgl/src/misc/lv_timer.c:325
        #20 0x5609a29bcad9 in lv_timer_handler /home/liam/lvgl/lvgl/src/misc/lv_timer.c:107
        #21 0x5609a2961d60 in main /home/liam/lvgl/lvgl_workspace/main.c:3396
        #22 0x7f53c54e9082 in __libc_start_main ../csu/libc-start.c:308
    ```
</details>

_Originally posted by @liamHowatt in https://github.com/lvgl/lvgl/pull/7124#pullrequestreview-2382296918_

liamHowatt, 2024-10-22:
> Hi, thanks for the information, I'll try to take a look at it this week. It happens in example 3 demo? Also, do you have `LV_FILE_EXPLORER_QUICK_ACCESS` enabled?

Whether or not QUICK_ACCESS is enabled:

lv_example_file_explorer_1: crashes immediately
lv_example_file_explorer_2: unrelated strcmp inconsistent crash after clicking a few dirs and files
lv_example_file_explorer_3: crashes when going into a dir, then back to the parent

C47D, 2024-10-22:
This happens even before merging #7124 , is that right?

C47D, 2024-10-22:
I'm able to replicate it very easily, click in a directory, then in a empty space

C47D, 2024-10-23:
Here's another example, I'm editing the table example 1, removed a bunch of rows from the table and when I click outside of the valid rows, but inside of the table, let's call it "empty space", the table considers it as another row.

[Screencast from 22-10-24 21:34:24.webm](https://github.com/user-attachments/assets/299fccb5-c596-402f-ac40-ca8732b26011)

Then if I try to get the content of the selected (invalid) cell it happens the same problem as the file explorer.

C47D, 2024-10-23:
I've reverted my previous patches and I think this is the best I can came up with, calculating if the press was outside of the rows height, if so return `LV_RESULT_INVALID`

```patch
diff --git a/src/widgets/table/lv_table.c b/src/widgets/table/lv_table.c
index affb99b88..b431ad060 100644
--- a/src/widgets/table/lv_table.c
+++ b/src/widgets/table/lv_table.c
@@ -997,6 +1011,24 @@ static lv_result_t get_pressed_cell(lv_obj_t * obj, uint32_t * row, uint32_t * c
     lv_point_t p;
     lv_indev_get_point(lv_indev_active(), &p);
 
+    /* Calculate total height of rows so we know if the click was outside them */
+    lv_area_t area;
+    lv_obj_get_coords(obj, &area);
+
+    uint8_t idx = 0;
+    int32_t total_row_height = 0;
+
+    for(idx = 0; idx < table->row_cnt; ++idx) {
+        total_row_height += *(table->row_h);
+    }
+
+    /* Clicked outside rows on empty space */
+    if(p.y > (total_row_height + area.y1)) {
+        if(col) *col = LV_TABLE_CELL_NONE;
+        if(row) *row = LV_TABLE_CELL_NONE;
+        return LV_RESULT_INVALID;
+    }
+
     int32_t tmp;
     if(col) {
         int32_t x = p.x + lv_obj_get_scroll_x(obj);

```

[Screencast from 22-10-24 22:12:34.webm](https://github.com/user-attachments/assets/23d26fc9-89d7-403f-bb48-6c2a6a6f4d6b)

kisvegabor, 2024-10-23:
Thank you for tracking it. 

I agree with both changes:
1. Return `NULL` from `lv_table_get_cell_value`
2. Return `LV_RESULT_INVALID` from `get_pressed_cell`

C47D, 2024-10-23:
> Thank you for tracking it.
> 
> I agree with both changes:
> 
>     1. Return `NULL` from `lv_table_get_cell_value`
> 
>     2. Return `LV_RESULT_INVALID` from `get_pressed_cell`

Hi,

I did change 1, and then ran the tests and a lot failed, that's why i gave change 2 a try :smile: 
Is that the best way to calculate the row height? Should we also consider the columns width?

liamHowatt, 2024-10-24:
> This happens even before merging #7124 , is that right?

Correct.

> 2. `lv_table_get_cell_value` returns `""` instead of a pointer (I might be wrong here, please correct me if this is valid) when the row or column are invalid, then this returned value is passed to `lv_strcmp`, were @liamHowatt noticed the error.

This is okay because `""` is a pointer to a `'\0'` which is fine to pass to `lv_strcmp`.

Tried your patch from this comment https://github.com/lvgl/lvgl/issues/7132#issuecomment-2430845810 (with no other patches) and example 1 still crashes immediately with the original use-after-free dump. Are you building with `-fsanitize=address`?

Please open a PR to share the latest experiments :slightly_smiling_face:

I tried to fix it a few months ago and I had a hard time. There are some complicated allocations and re-allocations of table cell data. IIRC it `realloc`s some array which needs to change size, but something holds a pointer to a member of the array before it was reallocated, so the data is already "freed" when the reference to the old-size array member is now invalid, yet still accessed.

C47D, 2024-10-24:
Hi @liamHowatt ,

I tried example 3 only, will test it with example 1 later today. What I've also noticed is that after freeing something the pointer is not marked as NULL, which I think is a good practice, but I don't know if this is the solution to the issue.

C47D, 2024-10-24:
> Tried your patch from this comment https://github.com/lvgl/lvgl/issues/7132#issuecomment-2430845810 (with no other patches) and example 1 still crashes immediately with the original use-after-free dump. Are you building with -fsanitize=address?
Please open a PR to share the latest experiments 🙂

I've opened #7153 for it, I can't replicate the example 1 crash, and yes, I'm using the eclipse simulator, which is building with -fsanitize=address

C47D, 2024-10-25:
@liamHowatt is it possible for you to record the example 1 crash? Or get the callstack? I haven't been able to reproduce it :grimacing:

dpopov76, 2024-10-26:
> @liamHowatt is it possible for you to record the example 1 crash? Or get the callstack? I haven't been able to reproduce it 😬
> 
> **EDIT** I've also found this PR related to the table, I think it needs to be applied #7042

The bug in that PR is not directly affecting file_explorer (there was a similar bug in `lv_table_set_row_count(...)` that **was** causing crashes but it had been fixed).

On the current issue - I think `get_pressed_cell()` already "handles" clicking outside of the table, for examle if you have 5 rows (from 0 to 4) and you click in the white space below the last row it will return 5. So may be the fix should be something like this at the end of the function:
`if (*row >= table->row_cnt || *col >= table->col_cnt) return LV_RESULT_INVALID else return LV_RESULT_OK;`
and a check in `browser_file_event_handler`

C47D, 2024-10-26:
@dpopov76 I will check `get_pressed_cell` again, because I edited it in #7153 , I'm handling the click on a white space based on the row size, but I think your approach when checking the returned values of `get_pressed_cell` is cleaner.

liamHowatt, 2024-11-06:
I found the root-cause of the use-after-free bug.

The file explorer gets the text of a table cell with `lv_table_get_cell_value` and then sets another cell's text to that value with `lv_table_set_cell_value`. The crash happens when `i == j`. Before setting the new value, `lv_table_set_cell_value` frees the old value's memory, which is the memory as the new text that's being set, so when the new text gets strcpy'd into the new memory, the new text's memory was freed earlier in the function.

https://github.com/lvgl/lvgl/blob/6b982a3698d420bc9244a5806b92cea64d8e84df/src/others/file_explorer/lv_file_explorer.c#L653

Potential fix 1:

My preferred fix which I've opened a PR for: #7239. File explorer shall not set a table cell to its current value.

```diff
diff --git a/src/others/file_explorer/lv_file_explorer.c b/src/others/file_explorer/lv_file_explorer.c
index 7a1e69752..a0f7c3f29 100644
--- a/src/others/file_explorer/lv_file_explorer.c
+++ b/src/others/file_explorer/lv_file_explorer.c
@@ -647,6 +647,8 @@ static void strip_ext(char * dir)
 
 static void exch_table_item(lv_obj_t * tb, int16_t i, int16_t j)
 {
+    if(i == j) return;
+
     const char * tmp;
     tmp = lv_table_get_cell_value(tb, i, 0);
     lv_table_set_cell_value(tb, 0, 2, tmp);
```

Potential fix 2:

Makes it the table's responsibility to return early if the text being set is already the pointer of the cell being set.

```diff
diff --git a/src/widgets/table/lv_table.c b/src/widgets/table/lv_table.c
index 444406040..f50fc0747 100644
--- a/src/widgets/table/lv_table.c
+++ b/src/widgets/table/lv_table.c
@@ -95,11 +95,14 @@ void lv_table_set_cell_value(lv_obj_t * obj, uint32_t row, uint32_t col, const c
 
     lv_table_t * table = (lv_table_t *)obj;
 
+    uint32_t cell = row * table->col_cnt + col;
+
+    if(table->cell_data[cell]->txt == txt) return;
+
     /*Auto expand*/
     if(col >= table->col_cnt) lv_table_set_column_count(obj, col + 1);
     if(row >= table->row_cnt) lv_table_set_row_count(obj, row + 1);
 
-    uint32_t cell = row * table->col_cnt + col;
     lv_table_cell_ctrl_t ctrl = 0;
 
     /*Save the control byte*/
```

PR #7239 (2024-11-13): fix(file_explorer): table cell use after free
Fixes #7132

See: https://github.com/lvgl/lvgl/issues/7132#issuecomment-2459641965

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/ [... cut]
closed 2024-11-13 by kisvegabor