lvgl/lvgl #7137: Drawing a 3D rotated arc

kisvegabor, 2024-10-22:
### Introduce the problem

Can we draw and arc like this?
[image]

### Proposal

Rendering to a canvas with the vector API or SVG seems doable but probably you have better ideas for this than me.

 @FASTSHIFT @onecoolx could show a basic a example for this? If all goes well I can convert it to examples.

FASTSHIFT, 2024-10-23:
This effect can be achieved by using arc path fill + matrix transformation. You can use this tool: https://w-mai.github.io/vegravis/ to try to fine-tune the values ​​of the 3x3 matrix for preview.

FASTSHIFT, 2024-10-23:
[image]
This is the approximation I made.

path code:
```bash
MOVE, 186.97, 293.03, 
CUBI, 157.68, 263.74, 157.68, 216.26, 186.97, 186.97, 
CUBI, 216.26, 157.68, 263.74, 157.68, 293.03, 186.97, 
CUBI, 322.32, 216.26, 322.32, 263.74, 293.03, 293.03, 
LINE, 284.55, 284.55, 
CUBI, 309.15, 259.94, 309.15, 220.06, 284.55, 195.45, 
CUBI, 259.94, 170.85, 220.06, 170.85, 195.45, 195.45, 
CUBI, 170.85, 220.06, 170.85, 259.94, 195.45, 284.55, 
END, 
MOVE, 288.79, 282.79, 
CUBI, 292.10, 282.79, 294.79, 285.48, 294.79, 288.79, 
CUBI, 294.79, 292.10, 292.10, 294.79, 288.79, 294.79, 
CUBI, 285.48, 294.79, 282.79, 292.10, 282.79, 288.79, 
CUBI, 282.79, 285.48, 285.48, 282.79, 288.79, 282.79, 
END, 
MOVE, 191.21, 282.79, 
CUBI, 194.52, 282.79, 197.21, 285.48, 197.21, 288.79, 
CUBI, 197.21, 292.10, 194.52, 294.79, 191.21, 294.79, 
CUBI, 187.90, 294.79, 185.21, 292.10, 185.21, 288.79, 
CUBI, 185.21, 285.48, 187.90, 282.79, 191.21, 282.79, 
END, 
END, 
```

FASTSHIFT, 2024-10-23:
You can refer to the arc path generation method in `lvgl/src/draw/vg_lite/lv_draw_vg_lite_arc.c` to call `lv_vector_path_append_arc` to generate the path for filling. Call `lv_vector_dsc_set_transform` or `lv_vector_dsc_set_fill_transform` to set the matrix.

kisvegabor, 2024-10-23:
Thank you for the input and the useful link.

This is what I have now:
```c
void lv_example_canvas_9(void)
{
    /*Create a buffer for the canvas*/
    LV_DRAW_BUF_DEFINE_STATIC(draw_buf, 200, 200, LV_COLOR_FORMAT_ARGB8888);
    LV_DRAW_BUF_INIT_STATIC(draw_buf);

    /*Create a canvas*/
    lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_draw_buf(canvas, &draw_buf);
    lv_canvas_fill_bg(canvas, lv_color_hex3(0xccc), LV_OPA_COVER);
    lv_obj_set_pos(canvas, 10, 10);

    lv_layer_t layer;
    lv_canvas_init_layer(canvas, &layer);

    lv_vector_dsc_t * dsc = lv_vector_dsc_create(&layer);
    lv_vector_path_t * path = lv_vector_path_create(LV_VECTOR_PATH_QUALITY_MEDIUM);

    lv_fpoint_t start = {0, 0};
    lv_vector_path_move_to(path, &start);
    lv_fpoint_t center = {-50, 0};
	lv_vector_path_append_arc(path, &center, 50, 0, 350, false);

    lv_vector_dsc_set_stroke_color(dsc, lv_color_make(0x00, 0xff, 0x00));
    lv_vector_dsc_set_stroke_opa(dsc, LV_OPA_COVER);
    lv_vector_dsc_set_fill_opa(dsc, LV_OPA_0);
    lv_vector_dsc_set_stroke_width(dsc, 8.0f);

    lv_matrix_t mt;
    lv_matrix_identity(&mt);
    lv_matrix_skew(&mt, 20, 0);
    lv_matrix_translate(&mt, 150, 100);
    lv_vector_dsc_set_transform(dsc, &mt);

    lv_vector_dsc_add_path(dsc, path);

    lv_draw_vector(dsc);
    lv_vector_path_delete(path);
    lv_vector_dsc_delete(dsc);

    lv_canvas_finish_layer(canvas, &layer);
}
```

Resulting in:
[image]

There are a few things that I don't understand:
- It seems drawing an arc by knowing the start and end angles are complicated. Instead we should draw an arc with a sweep and rotate it. However I couldn't figure out how to rotate around the center.
- Setting only `skew_x` affects the arc vertically too. Is it because of the translation?

Could you help with these?

onecoolx, 2024-10-24:
Your question is actually a 2.5D perspective transformation problem, which cannot be directly obtained using 2D transformation. 
2D transformation actually only uses 6  factors of the 3x3 matrix. Perspective transformations require additional matrix operations, geometric transformations, and resampling support. Unfortunately, thorvg does not support these.

kisvegabor, 2024-10-24:
I think skew would be enough for what we need but I'm having trouble with setting the pivot for both the rotation and skew transformation.

onecoolx, 2024-10-24:
> I think skew would be enough for what we need but I'm having trouble with setting the pivot for both the rotation and skew transformation.

If you just want to rotate around a center point, you can do：
```
       lv_matrix_translate(&matrix,   pivot _x,  pivot _y);  // translate to pivot 
        lv_matrix_rotate(&matrix, rotation );   // set rotation
       // lv_matrix_scale(&matrix,  scale_x, scale_y);  // set scale if needed
        lv_matrix_translate(&matrix, -pivot _x, -pivot_y);  // translate back 
 
      .....
      lv_vector_dsc_set_transform(dsc, &matrix);
```

kisvegabor, 2024-10-25:
Thank you! I'm slowly understanding how it works :slightly_smiling_face: 

At https://w-mai.github.io/vegravis/ when I set m20 to 0.0002 it showed a perspective effect and it gave me the fals hope that it will be that simple. Now  I understand what you meant by:
> Your question is actually a 2.5D perspective transformation problem, which cannot be directly obtained using 2D transformation.
2D transformation actually only uses 6 factors of the 3x3 matrix. Perspective transformations require additional matrix operations, geometric transformations, and resampling support. Unfortunately, thorvg does not support these.

So this parameter is just ignored in ThorVG. :slightly_frowning_face:

kisvegabor, 2024-11-11:
I've made some research and it trued out our Software rendering doesn't support it, but it's possible by using VG-Lite's perspective transformation feature. 

We will prepare some demos/examples in the upcoming month. Closing this issue until then.

closed 2024-11-11 by kisvegabor