lvgl/lvgl #7145: arc could not see the lable!

xinsuinizhuan, 2024-10-24:
### LVGL version

8.3.10 and 8.4

### What happened?

[image]
could not see the lable value

### How to reproduce?

static void value_changed_event_cb(lv_event_t* e);

void lv_example_arc_1(void)
{
	lv_obj_t* label = lv_label_create(lv_scr_act());

	/*Create an Arc*/
	lv_obj_t* arc = lv_arc_create(lv_scr_act());
	lv_obj_set_size(arc, 150, 150);
	lv_arc_set_rotation(arc, 135);
	lv_arc_set_bg_angles(arc, 0, 270);
	lv_arc_set_value(arc, 10);
	lv_obj_center(arc);
	lv_obj_add_event_cb(arc, value_changed_event_cb, LV_EVENT_VALUE_CHANGED, label);

	/*Manually update the label for the first time*/
	lv_event_send(arc, LV_EVENT_VALUE_CHANGED, NULL);
}

static void value_changed_event_cb(lv_event_t* e)
{
	lv_obj_t* arc = lv_event_get_target(e);
	lv_obj_t* label = (lv_obj_t*)lv_event_get_user_data(e);

	lv_label_set_text_fmt(label, "%d%%", lv_arc_get_value(arc));

	/*Rotate the label to the current position of the arc*/
	lv_arc_rotate_obj_to_angle(arc, label, 25);
}

xinsuinizhuan, 2024-10-24:
but 9.3 is ok.

xinsuinizhuan, 2024-10-24:
[image]
9.3

kissa96, 2024-10-24:
Hi, it seems you have not created a label, so there is nothing which could display the text. LVGL functions return immediately if the supplied object is NULL

lizhaoming634, 2024-10-24:
#define LV_COLOR_SCREEN_TRANSP 1

xinsuinizhuan, 2024-10-24:
> #define LV_COLOR_SCREEN_TRANSP 1

is is not work, when set the lv_color_screen_transp, some where could see it, it seem the rect, int the rect cornner could see:
[image]

it is the issure:lv_arc_rotate_obj_to_angle
[image]

xinsuinizhuan, 2024-10-24:
> Hi, it seems you have not created a label, so there is nothing which could display the text. LVGL functions return immediately if the supplied object is NULL

[image]

kisvegabor, 2024-10-25:
If the label is created on the arc it will be clipped. Try creating it on a the screen.

xinsuinizhuan, 2024-10-29:
thank you. #define LV_COLOR_SCREEN_TRANSP 1 and the label is created on the screen, it is ok.

kisvegabor, 2024-10-29:
Great! I'm losing this issue then.

closed 2024-10-29 by kisvegabor