lvgl/lvgl #7147: Scrolling message is not working as expected

keerthansys22, 2024-10-24:
### LVGL version

v9.2.0

### What happened?

Hi, I'm using an LCD display (ST7291) on the custom board.

The scrolling message in the screen is getting additional border around the scrolling space which is not expected. I have been working on LVGL v8.3.9 and there it was working fine. I have checked in the documentation of V9.2.0 and there are no changes compared to V8.3.9 for calling scrolling text function.

One more observation, if a single scrolling message in a screen, then the border issue happens. If two scrolling message is displaying in a single screen then one of the scrolling messages gets the border issue and the other one works fine, I have attached the snapshot for the reference.

Two scrolling text - 

https://github.com/user-attachments/assets/d6c9ee12-dc70-4f29-a242-de4587f4d6b0

One scrolling text - 

https://github.com/user-attachments/assets/532301e3-906b-4946-a68f-2739f01d8f20

### How to reproduce?

```c
// For first scrolling message
lv_obj_t * label2 = lv_label_create(screenmain);
lv_label_set_long_mode(label2, LV_LABEL_LONG_SCROLL_CIRCULAR);     /*Circular scroll*/
lv_obj_set_size(label2, 120, 15);
lv_label_set_text(label2, "It is a circularly scrolling text. ");
lv_obj_align(label2, LV_ALIGN_DEFAULT, 4, 50);
lv_obj_set_style_text_font(label2, &lv_font_montserrat_12, LV_PART_MAIN);
lv_obj_set_style_text_color(label2, lv_color_white(), LV_PART_MAIN);

// For second scrolling message
lv_obj_t * label5 = lv_label_create(screenmain);
lv_label_set_long_mode(label5, LV_LABEL_LONG_SCROLL_CIRCULAR);     /*Circular scroll*/
lv_obj_set_size(label5, 120, 15);
lv_label_set_text(label5, "It is a circularly scrolling text. ");
lv_obj_align(label5, LV_ALIGN_DEFAULT, 4, 80);
lv_obj_set_style_text_font(label5, &lv_font_montserrat_12, LV_PART_MAIN);
lv_obj_set_style_text_color(label5, lv_color_white(), LV_PART_MAIN);
```

liamHowatt, 2024-10-25:
Thanks for sending code to reproduce. I could not reproduce the problem in the simulator unfortunately. It's interesting that there was no issue in v8.3.9.

If I were to guess, I would say that it's a buffer stride or alignment issue. Are you using `LV_COLOR_DEPTH 16` (RGB565) in partial mode? It could be that an odd-numbered area width is incorrectly handled in the flush callback.

keerthansys22, 2024-10-25:
Thanks for your message @liamHowatt, Yes, I'm using ``LV_COLOR_DEPTH 16`` (RGB565) in partial mode.

I have defined the horizontal and vertical resolution as:-
```c
#define MY_DISP_HOR_RES   130

#define MY_DISP_VER_RES    130
```

Please find my LVGL initialization code for your reference,

```c
#define BYTES_PER_PIXEL 256

lv_display_t * disp;
static uint32_t buf_1[MY_DISP_HOR_RES * 10 * BYTES_PER_PIXEL];
static uint32_t buf_2[MY_DISP_HOR_RES * 10 * BYTES_PER_PIXEL];

static void disp_flush(lv_disp_t* display, const lv_area_t * area, uint8_t * color_p);
extern void lcd_send_data(uint16_t x_start, uint16_t y_start, uint16_t x_end, uint16_t y_end, uint8_t *p);

void lvgl_init(void);

/*
 * @brief Function to initialize lvgl drivers
 * @param None
 */
void lvgl_init(void)
{
    // LVGL initialization
      lv_init();

      // Create a display
      disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);

      // Set the draw buffers
      lv_display_set_buffers(disp, buf_1, buf_2, (MY_DISP_HOR_RES * BYTES_PER_PIXEL), LV_DISPLAY_RENDER_MODE_PARTIAL);

      // Set the flush callback function
      lv_display_set_flush_cb(disp, disp_flush);

      // Set the rotation to 180
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);

}

/* @brief   Flush the content of the internal buffer the specific area on the display */

static void disp_flush(lv_disp_t* display, const lv_area_t * area, uint8_t * color_p)
{
    /*SPI transmit data without DMA*/
    lv_draw_sw_rgb565_swap(buf_1, (MY_DISP_HOR_RES * MY_DISP_VER_RES));
    lcd_send_data(area->x1, area->y1, area->x2, area->y2, (uint8_t *)color_p);

    lv_display_flush_ready(display);
}
```

Interestingly, this scrolling message issue and screen rotation is not working in v9.2.0!!

nikunj8780, 2024-10-25:
not use like line "lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180); "
using line  "lv_disp_set_rotation(disp, LV_DISPLAY_ROTATION_0);"

kisvegabor, 2024-10-25:
> not use like line "lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180); "

Good idea! Does it work without `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);`?

keerthansys22, 2024-10-28:
No, it is not working with lv_disp_set_rotation(disp, LV_DISPLAY_ROTATION_180); function also!!

nikunj8780, 2024-10-28:
> Good idea! Does it work without `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);`?

yes it's work proper without using line  lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);   if you not need to rotate your disply that not need to use this line without code is working properly

keerthansys22, 2024-10-28:
> > not use like line "lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180); "
> 
> Good idea! Does it work without `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);`?

No without this `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);` also the scrolling is not working!!

liamHowatt, 2024-10-30:
Is it an SPI display? Really not sure at all, but if you have control over `lcd_send_data`, try adding a 1 ms delay before de-asserting the CS pin.

keerthansys22, 2024-10-30:
Thanks for the reply @liamHowatt

Yes, it is SPI display, I will try to add delay and let you know the outcome.

However, I tried to switch from RGB565 to RGB888 in lvgl and my display supports only till 18 bit i.e RGB666. So, I made the switch from 565 to 888 and the scrolling message is working fine. This was my observation.

liamHowatt, 2024-10-30:
If changing the color format fixed it then my suggestion is probably not the right solution. If you do try though, actually add a 1 ms delay before _and_ after asserting CS and also before _and_ after de-asserting CS, just to be sure and eliminate this as a possibility.

keerthansys22, 2024-11-07:
I tried to add delay as per your suggestion @liamHowatt, I couldn't find any major difference. Anyways after changing from RGB555 to RGB888, it is working for me. 

I found this in changelog of V9.0 - In v9 lv_color_t is always RGB888 regardless of LV_COLOR_DEPTH. 

Maybe because of this we were facing issue in RGB565 I assume, I'm not sure.

Thanks for your time :)

closed 2024-12-03 by lvgl-bot