lvgl/lvgl #7151: Issue with straight line data rendering as zigzag pattern in chart

aleskdev, 2024-10-24:
### LVGL version

v9.2.0

### What happened?

I am encountering an issue with rendering a straight line in my chart using LVGL. Despite inputting straight line data into the buffer, the visual output on the screen appears as a zigzag pattern. This is unexpected, as the data is intended to represent a straight line.

**Steps to Reproduce:**
Construct a minimal example with LVGL to create a chart.
Populate the data buffer with values that should form a straight line.
Refresh the chart to display the data.

**Expected Behavior:**
The chart should render a smooth, straight line connecting the data points.

**Actual Behavior:**
The chart displays a zigzag pattern, with noticeable bumps along the intended straight line. However, when I reduce the number of data points to around 10, the line appears smooth and straight, suggesting that the issue may be related to the number of points used.

Sorry for bad quality pictures, but you can see zigzag pattern. All other demos and renderings are fine.

[image]
[image]

**Additional Information:**
LVGL Version: v9.2.0
Platform: STM32H7B3I-EVAL

**Questions:**
Are there any known issues or limitations with rendering straight lines in LVGL?
Is there a recommended approach to ensure smooth rendering of straight lines without unintended zigzag patterns?
Could this issue be related to the way data points are scaled or rendered on the chart?

I appreciate any guidance or insights into resolving this issue. Thank you!

### How to reproduce?

	lv_obj_t *chart2 = lv_chart_create(objects.second_screen);
	lv_obj_set_pos(chart2, 10, 10);
	lv_obj_set_size(chart2, 750, 450);

	lv_chart_set_type(chart2, LV_CHART_TYPE_LINE);
	lv_chart_set_point_count(chart2, 200);
	lv_chart_set_range(chart2, LV_CHART_AXIS_PRIMARY_Y, 0, 200);

	lv_obj_set_style_size(chart2, 1, 1, LV_PART_INDICATOR);

	lv_chart_series_t * ser_phases2 = lv_chart_add_series(chart2, lv_palette_main(LV_PALETTE_BLUE), LV_CHART_AXIS_PRIMARY_Y);
	int32_t * ser_phases_points2 = lv_chart_get_y_array(chart2, ser_phases2);

	uint32_t i;
	for(i = 10; i < 100; i++) {
			ser_phases_points2[i] = i;
	}

	lv_chart_refresh(chart2);

kissa96, 2024-10-25:
Does this issue show only in simulator, or does this apply on real HW too?
LVGL draws a straight line between 2 points on a line chart. Maybe the exact coordinates are off by 1 pixel, thus causing a jagged line. I'll look into it.

aleskdev, 2024-10-25:
This is on real HW as i stated. I have not tried in simulator yet. But it seams to be chart related, because all other renderings on HW are fine.

kisvegabor, 2024-10-25:
I think it's a limitation of the current line drawing algorithm. :slightly_frowning_face: 

Using ThorVG (LVGL's built-in vector graphics library) could produce better result. Although at this moment it's not integrated with the chart yet, I think it's worth a try. 

I'm also playing with it [in this issue](https://github.com/lvgl/lvgl/issues/7137).

closed 2024-11-18 by lvgl-bot