lvgl/lvgl #7152: indev: object does not receive LV_EVENT_RELEASED if an unrelated object state changes to LV_STATE_DISABLED

Viatorus, 2024-10-24:
### LVGL version

9.2.0

### What happened?

Take a look at the example below. Just after the `button` is pressed, the `unrelated_obj` changes it state to disabled. The indev-pointer (and all other indev's) looses the active object because it is reset.

This is annoying and I hope this shouldn't be the case. I think only indev's which do have something related to the `unrelated_obj` should be handled.

---

Why is this happen? Let's take a look at the code.

1. Assuming the button has received `LV_EVENT_PRESSED` after reaching:

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/indev/lv_indev.c#L728

2. Later, LV_STATE_DISABLED is added to the `unrelated_obj`...

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/core/lv_obj.c#L306-L314

3. ... which performs an `lv_indev_reset` over all indev's via `indev_reset_core`

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/indev/lv_indev.c#L412-L417

4. Which just resets any active indev object and adds it to the `reset_query`:

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/indev/lv_indev.c#L1685-L1691

5. If now the user releases the mouse and then `lv_indev_read` is called, `indev_proc_reset_query_handler` is called...

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/indev/lv_indev.c#L206-L218

6.  ...which just resets everything remaining

https://github.com/lvgl/lvgl/blob/a8d2cc05f116d08a1a0d58f9544e3d4ee9dab9d1/src/indev/lv_indev.c#L1510-L1516

7. Any active unrelated object? Forgotten. If the user wouldn't release the mouse, the button would receive the `LV_EVENT_PRESSED` again instead.

**History**
This functionality was introduced in https://github.com/lvgl/lvgl/pull/3483 and later improved in https://github.com/lvgl/lvgl/pull/4539. But still with issues as described above.

### How to reproduce?

```c
static void pressed_cb(lv_event_t* e)
{
    LV_LOG_USER("%d", lv_event_get_code(e));
}

static void released_cb(lv_event_t* e)
{
    LV_LOG_USER("%d", lv_event_get_code(e));
}

bool disable = true;
static void timer_cb(lv_timer_t* timer)
{
    lv_obj_t* obj = lv_timer_get_user_data(timer);
    if (disable) {
       lv_obj_add_state(obj, LV_STATE_DISABLED);
    } else {
       lv_obj_remove_state(obj, LV_STATE_DISABLED);
    }
    disable = !disable;
}

void obj_disable_test()
{
    lv_obj_t* button= lv_obj_create(lv_screen_active());
    lv_obj_add_event_cb(button, pressed_cb, LV_EVENT_PRESSED, NULL);
    lv_obj_add_event_cb(button, released_cb, LV_EVENT_RELEASED, NULL);

    lv_obj_t* unrelated_obj = lv_obj_create(lv_screen_active());
    lv_obj_set_pos(unrelated_obj , 100, 100);

    lv_timer_t* timer = lv_timer_create(timer_cb, 1, unrelated_obj );
    lv_timer_set_repeat_count(timer, -1);
}

```

kissa96, 2024-10-25:
Interesting scenario for sure. I have no idea why we need to reset indevs. @liamHowatt @kisvegabor are the gurus of LVGL's core.

kisvegabor, 2024-10-25:
Oh, I see the problem. It really could be improved. 

Resetting the input devices is really strong. Actually [now we handle](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1276) to not send any events to disabled widgets, so maybe the indev resets can be removed. Could you give it a try? Just remove `lv_indev_reset(NULL, obj);` from `lv_obj_add_state`. 
`

Viatorus, 2024-10-26:
I cannot try it before over next week. But since disabled objects are
already probably handled in lv_indev, the fix looks promising.

Viatorus, 2024-11-04:
@kisvegabor Looks good. PR created.

PR #7216 (2024-11-12): fix(indev): don't reset all indevs if an object is set disabled
Fixes #7152
closed 2024-11-12 by liamHowatt