lvgl/lvgl #7154: ESP32 lvgl project not compiling with #7131

Izumemori, 2024-10-25:
### LVGL version

9.2.1

### What happened?

Running `idf.py build` with lvgl version 9.2.1 results in the following error
```
{Project Path}/build/config/sdkconfig.h:761:41: error: expected identifier or '(' before string constant
  761 | #define CONFIG_LV_ATTRIBUTE_LARGE_CONST ""
      |                                         ^~
{Project Path}/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_internal.h:1324:42: note: in expansion of macro 'CONFIG_LV_ATTRIBUTE_LARGE_CONST'
 1324 |         #define LV_ATTRIBUTE_LARGE_CONST CONFIG_LV_ATTRIBUTE_LARGE_CONST
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
{Project Path}/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:24:8: note: in expansion of macro 'LV_ATTRIBUTE_LARGE_CONST'
   24 | static LV_ATTRIBUTE_LARGE_CONST const uint8_t glyph_bitmap[] = {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
{Project Path}/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:2155:21: error: 'glyph_bitmap' undeclared here (not in a function)
 2155 |     .glyph_bitmap = glyph_bitmap,
      |                     ^~~~~~~~~~~~
```
If I manually remove `LV_ATTRIBUTE_LARGE_CONST` from the glyph_bitmap definition then the same issue appears with `CONFIG_LV_ATTRIBUTE_MEM_ALIGN` instead.

Removing both configuration options from the Kconfig allows the project to compile again.

9.2.0 works fine, presumably since those options don't exist in the Kconfig.

nicusorcitu, 2024-10-25:
Sorry for that, please see this comment: https://github.com/lvgl/lvgl/pull/7131#issuecomment-2437651438

nikunj8780, 2024-10-25:
if you are using ESSPRESIF IDE than use this link and transfer your TTF/WOFF font  directly  : -https://lvgl.io/tools/fontconverter
and copy this font  .C file in you main folder and that use

liamHowatt, 2024-11-11:
#7131 was reverted by https://github.com/lvgl/lvgl/commit/932c14086b79aff2a27cd154441f680eb8257311 in `release/v9.2`. Closing for now.

closed 2024-11-11 by liamHowatt