lvgl/lvgl #7155: lv_line

kangkangyoung, 2024-10-25:
### LVGL version

v9.1.0

### What happened?

I tried running lv_example_scale_6() on the simulator and found that the effect was inconsistent with the v8.3, with the pointer display being abnormal. After troubleshooting, I discovered that the lines created by calling lv_line_create could not draw diagonal lines properly.
[image]

### How to reproduce?

void lv_example_line_1(void)
{
    /*Create an array for the points of the line*/
    static lv_point_precise_t line_points[] = { {5, 5}, {70, 70}, {120, 10}, {180, 60}, {240, 10} };

    /*Create style*/
    static lv_style_t style_line;
    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, 8);
    lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_rounded(&style_line, true);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(lv_screen_active());
    lv_line_set_points(line1, line_points, 5);     /*Set the points*/
    lv_obj_add_style(line1, &style_line, 0);
    lv_obj_center(line1);
}

C47D, 2024-10-25:
The ticks also look really bad, what display size are you using?

kisvegabor, 2024-10-25:
Are you using Visual Studio?  If so, this is a known bug that we track [here](https://github.com/lvgl/lv_port_pc_visual_studio/issues/84).

kangkangyoung, 2024-10-27:
Yes, I'm using Visual Studio

> Are you using Visual Studio? If so, this is a known bug that we track [here](https://github.com/lvgl/lv_port_pc_visual_studio/issues/84).

> The ticks also look really bad, what display size are you using?

size is 480*480

MouriNaruto, 2024-10-28:
This issue is caused by the LVGL mainline repository itself.

https://github.com/lvgl/lvgl/blob/c27760fb103dbcee94bbc2751e741fb55ffa8f35/src/draw/sw/lv_draw_sw_mask_private.h#L56
Change lv_draw_sw_mask_line_side_t to uint8_t because lv_draw_sw_mask_line_side_t will be seen as signed int. It actually causes LV_DRAW_SW_MASK_LINE_SIDE_TOP = -2 and LV_DRAW_SW_MASK_LINE_SIDE_BOTTOM = -3.

Kenji Mouri

kangkangyoung, 2024-10-29:
Thank you everyone.. The patch update issue has been resolved.

commit 253981ae (2024-10-29): fix: add +1 but for enum bitfields to avoid overflow to negative

fixes #7155