lvgl/lvgl #7156: LVGLV roller scrolls the whole screen vertically during scroll

TothFrank, 2024-10-25:
### LVGL version

v8.4.0

### What happened?

Hi, When I create a simple roller and scroll it vertically then It scrolls the whole screen around. I use ESP32-S3. The first picture shows how it should be. The second picture shows what is the issue. Sorry for the bad quality, but you get the idea. What the issue be?

[image]
[image]

### How to reproduce?

`#include <Arduino.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>

#include <lvgl.h>
#include "lvgl_port_v8.h"
#include <demos/lv_demos.h>

#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

#define LVGL_PORT_AVOID_TEAR 1

void setup()
{
  Serial.begin(115200);
  pinMode(GPIO_INPUT_IO_4, OUTPUT);

  Serial.println("Initialize IO expander");

  ESP_IOExpander *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS_000, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);

  expander->init();
  expander->begin();
  expander->multiPinMode(TP_RST | LCD_BL | LCD_RST | SD_CS | USB_SEL, OUTPUT);
  expander->multiDigitalWrite(TP_RST | LCD_BL | LCD_RST, HIGH);
  delay(150);

  expander->multiDigitalWrite(TP_RST | LCD_RST, LOW);
  delay(150);
  digitalWrite(GPIO_INPUT_IO_4, LOW);
  delay(150);
  expander->multiDigitalWrite(TP_RST | LCD_RST, HIGH);
  delay(200);

  Serial.println("Initialize panel device");
  ESP_Panel *panel = new ESP_Panel();
  panel->init();
  #if LVGL_PORT_AVOID_TEAR
   
    ESP_PanelBus_RGB *rgb_bus = static_cast<ESP_PanelBus_RGB *>(panel->getLcd()->getBus());
    rgb_bus->configRgbFrameBufferNumber(LVGL_PORT_DISP_BUFFER_NUM);
    rgb_bus->configRgbBounceBufferSize(LVGL_PORT_RGB_BOUNCE_BUFFER_SIZE);
  #endif
  panel->begin();

  Serial.println("Initialize LVGL");

  lv_init();
  lvgl_port_init(panel->getLcd(), panel->getTouch());

  Serial.println("Create UI");

  lvgl_port_lock(-1);
  ui_init();
  lvgl_port_unlock();
}

void loop()
{
  delay(30);
}
`

TothFrank, 2024-10-25:
[image]

kisvegabor, 2024-10-25:
Szia! :slightly_smiling_face: 

Please add `lv_obj_clear_flag(roller, LV_OBJ_FLAG_SCROLL_CHAIN_VER);`. It [should be added](https://github.com/lvgl/lvgl/blob/release/v8.4/src/widgets/lv_roller.c#L302) in v8.4 by default, but I can't think of anything else that can cause this problem.

TothFrank, 2024-10-26:
Szia!

Thanks, but it is already there. The flag is already there, and is automatically generated with EEZ Studio:

`{
  lv_obj_t *obj = lv_roller_create(parent_obj);
  lv_obj_set_pos(obj, -14, 109);
  lv_obj_set_size(obj, 139, 261);
  lv_roller_set_options(obj, "00:00\n01:00\n02:00\n03:00\n04:00\n05:00\n06:00\n07:00\n08:00\n09:00\n10:00\n11:00\n12:00\n13:00\n14:00\n15:00\n16:00\n17:00\n18:00\n19:00\n20:00\n21:00\n22:00\n23:00", LV_ROLLER_MODE_INFINITE);
  lv_obj_add_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN_VER);
  lv_obj_clear_flag(obj, LV_OBJ_FLAG_SNAPPABLE);
}
{
  lv_obj_t *obj = lv_roller_create(parent_obj);
  lv_obj_set_pos(obj, 145, 109);
  lv_obj_set_size(obj, 99, 261);
  lv_roller_set_options(obj, "Kikapcsolva\n16,0 C\n16,5 C\n17,0 C\n17,5 C\n18,0 C\n18,5 C\n19,0 C\n19,5 C\n20,0 C\n20,5 C\n21,0 C\n21,5 C\n22,0 C\n22,5 C\n23,0 C\n23,5 C\n24,0 C\n24,5 C\n25,0 C", LV_ROLLER_MODE_NORMAL);
  lv_obj_add_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN_VER);
  lv_obj_clear_flag(obj, LV_OBJ_FLAG_SNAPPABLE);
}`

TothFrank, 2024-10-26:
[screens.txt](https://github.com/user-attachments/files/17529220/screens.txt)
[screens_h.txt](https://github.com/user-attachments/files/17529221/screens_h.txt)

Just enclosed the whole screens.c file with screens.h file. Maybe EEZ studio generates something wrong.

kisvegabor, 2024-10-26:
This flag should be cleared, not added.

TothFrank, 2024-10-26:
I think, I found the solution. In lvgl_port_v8.h file the following setting were set:

`define LVGL_PORT_TICK_PERIOD_MS                (3) // The period of the LVGL tick task, in milliseconds`

`/*`
 `* Set the avoid tearing mode:`
` *      - 0: Disable avoid tearing function`
` *      - 1: LCD double-buffer & LVGL full-refresh`
` *      - 2: LCD triple-buffer & LVGL full-refresh`
` *      - 3: LCD double-buffer & LVGL direct-mode (recommended)`
` */`
`#define LVGL_PORT_AVOID_TEARING_MODE            (3)`

When I changed the above settings to 
`#define LVGL_PORT_AVOID_TEARING_MODE            (0)`
 and
`#define LVGL_PORT_TICK_PERIOD_MS                (10)`
then the issue suddenly disappeared. But it disappeared just when I set them both!

I'd like to emphasize that this issue is not cause by LVGL itself, but there is change made in the supporting "ESP32_Display_Panel" library provided by Waveshare (who is the maker of the display). The improper setting is in "ESP32_Display_Panel-master\examples\LVGL\v8\Porting\lvgl_port_v8.h" file (just for the case if anyone has the same issue like me) and looks like it does not cooperates well with ST7276 display.

Anyway, @kisvegabor thanks for quick feedback and helpfull guide, becuase it helped me to find the solution :)

TothFrank, 2024-10-26:
I think, I found the solution. In lvgl_port_v8.h file the following setting were set:

```
#define LVGL_PORT_TICK_PERIOD_MS                (3) // The period of the LVGL tick task, in milliseconds

/*
* Set the avoid tearing mode:
 *      - 0: Disable avoid tearing function
 *      - 1: LCD double-buffer & LVGL full-refresh
 *      - 2: LCD triple-buffer & LVGL full-refresh
 *      - 3: LCD double-buffer & LVGL direct-mode (recommended)
 */
#define LVGL_PORT_AVOID_TEARING_MODE            (3)

```
When I changed the above settings to
`#define LVGL_PORT_AVOID_TEARING_MODE            (0)`
and
`#define LVGL_PORT_TICK_PERIOD_MS                (10)`
then the issue suddenly disappeared. But it disappeared just when I set them both!

I'd like to emphasize that this issue is not cause by LVGL itself, but there is change made in the supporting "ESP32_Display_Panel" library provided by Waveshare (who is the maker of the display). The improper setting is in "ESP32_Display_Panel-master\examples\LVGL\v8\Porting\lvgl_port_v8.h" file (just for the case if anyone has the same issue like me) and looks like it does not cooperates well with ST7276 display.

Anyway, @kisvegabor thanks for quick feedback and helpfull guide, becuase it helped me to find the solution :)

closed 2024-10-26 by TothFrank

kisvegabor, 2024-10-29:
Very interesting. I can't see how these can be related. Anyway, thank you for the feedback!

mej558, 2024-12-28:
#define LVGL_PORT_AVOID_TEARING_MODE            (0)
and
#define LVGL_PORT_TICK_PERIOD_MS                (10)

immediately fixed my similar problem... Thanks!