lvgl/lvgl #7157: bug - possible dereference of NULL pointer from lv_subject_add_observer()

pfvogel, 2024-10-25:
### LVGL version

9.3.0

### What happened?

I ran static analysis tool (CodeChecker) that identified a possible NULL pointer dereference.

In src/others/observer/lv_observer.c:

lv_subject_add_observer() will pass NULL as the obj value to lv_subject_add_observer_obj(). 
 In that routine, obj is assigned to target, and then later observer->cb() will attempt to dereference target.

```
    observer->target = obj;
    observer->for_obj = 1;
    /* subscribe to delete event of the object */
    if(obj != NULL) {
        lv_obj_add_event_cb(obj, unsubscribe_on_delete_cb, LV_EVENT_DELETE, observer);
    }

    /* update object immediately */
    if(observer->cb) observer->cb(observer, subject);
```

`A possible fix would be to move the call inside the test to see if the obj is NULL.  The "for_obj" should probably also be set to 0 to indicate that target is not valid.

```
    observer->target = obj;
    observer->for_obj = 1;
    /* subscribe to delete event of the object */
    if(obj != NULL) {
        observer->for_obj = 0;

        lv_obj_add_event_cb(obj, unsubscribe_on_delete_cb, LV_EVENT_DELETE, observer);

       /* update object immediately */
       if(observer->cb) observer->cb(observer, subject);
    }
```

### How to reproduce?

N/A -- issue found through inspection

liamHowatt, 2024-10-28:
Hi, good find. Do you have a callback as an example which dereferences `target`? My first impression is that users of `lv_subject_add_observer` should know there isn't a target involved and would not try to access `target`.

pfvogel, 2024-10-28:
Yes, I think it would be reasonable to expect a user to know that the target is null and not dereference it, but I think that might be easy to overlook.  There are instances of this being done within the same source file (lv_observer.c). The lv_obj_bind_flag_if_eq() routine calls bind_to_bitfield() which creates a observer, passing the obj_flag_observer_cb() as the callback.  The callback calls the either the lv_obj_add_flag() or lv_obj_remove_flag() routine.  The add/remove flag routine() will assert when the obj is checked and found to be NULL.

It looks like there are cases where the target of the observer is expected to a pointer to an object (the "for_obj" flag is true).  Maybe part of the change would be to only allow invoking the callback with a NULL target would be when the for_obj flag is false?

pfvogel, 2024-10-29:
There doesn't appear to be a good way to change the code related to the lv_subjest_add_observer() without breaking backwards compatibility.  I was able to modify the callbacks being used to check that the target is not NULL before being used, and this change did not break any of the existing tests.

liamHowatt, 2024-10-30:
Looking at your [linked commit](https://github.com/pfvogel/lvgl/commit/ef7db8508271e592d5f592ee7d1ee314ddf8957a), I would say that this is exactly what `LV_USE_ASSERT_NULL` was made for. You could do

```c
LV_USE_ASSERT_NULL(subject);
LV_USE_ASSERT_NULL(obj);
```

What do you think?

pfvogel, 2024-10-30:
The LV_USE_ASSERT_NULL looks to be configuration control, set to 0 or 1.
I didn't use the LV_ASSERT_NULL macro because it looks like the routines are allowed to return NULL and the macro will block execution instead of returning NULL.  I can see where it might be a matter of preference on whether the routine blocked when it received invalid input or just returned NULL.  Since there were already other checks in the routine that returned NUL, I implemented it to also return NULL.  But I think using the macro would be reasonable as well.

kisvegabor, 2024-11-17:
The PR is merged, closing.

closed 2024-11-17 by kisvegabor