lvgl/lvgl #7158: Linker Error When Defining LV_DRAW_SW_SUPPORT_XRGB8888 without LV_DRAW_SW_SUPPORT_ARGB8888.

stachurr, 2024-10-25:
### LVGL version

v9.2.0

### What happened?

A function is not defined when `LV_COLOR_FORMAT_XRGB8888 == 1` and `LV_COLOR_FORMAT_ARGB8888 == 0`.

In lv_draw_sw.c, `lv_draw_sw_rotate()` uses the function `rotate90_argb8888()` for both `LV_COLOR_FORMAT_ARGB8888` and `LV_COLOR_FORMAT_XRGB8888`. However, said function is only defined if `LV_DRAW_SW_SUPPORT_ARGB8888 == 1`. 

If `LV_DRAW_SW_SUPPORT_XRGB8888` is allowed without `LV_DRAW_SW_SUPPORT_ARGB8888`, the function must be defined when `LV_DRAW_SW_SUPPORT_XRGB8888 || LV_DRAW_SW_SUPPORT_ARGB8888` (like the case statement in `lv_draw_sw_rotate()`).

Otherwise, a compile time warning should be issued.

### How to reproduce?

In lv_conf.c:
```
#define LV_DRAW_SW_SUPPORT_XRGB8888 1
#define LV_DRAW_SW_SUPPORT_ARGB8888 0
```

nilaygit-10721, 2024-10-26:
please assign this issue to me ?

kissa96, 2024-10-28:
Feel free to work on this! Much appriciated

stachurr, 2024-10-29:
@nilaygit-10721 I believe there are other combinations that can cause Linker errors as well. I stumbled across a different one after encountering this one, but unfortunately didn't make note of it.

kisvegabor, 2024-11-17:
https://github.com/lvgl/lvgl/pull/7185 is updated.

liamHowatt, 2024-11-21:
Fixed by #7185

closed 2024-11-21 by liamHowatt