lvgl/lvgl #7160: File explorer: Is the `.` element really useful?

C47D, 2024-10-27:
### Introduce the problem

The `.` element is really used/useful? I've never seen it on other file explorers.

Here's an example of the github explorer, there's only the `..` element.
[image]

### Proposal

Remove the `.` element from the file explorer.

kissa96, 2024-10-28:
.github and .devcontainer are hidden folders. In Unix, anything that has a . in its name, is considered a hidden folder

liamHowatt, 2024-10-29:
`.` is shown by the file explorer because `lv_fs_dir_read` finds it. You can see `.` in the output of `ls -a` on Linux or `dir` on Windows.

`..` means parent directory. `.` means current directory.

We could make the file explorer ignore it if we decide it's useless.

kisvegabor, 2024-10-29:
I can't see for what can we use `.`. If it's not to complicated we could replace `..` with `<- Back` or so.

liamHowatt, 2024-11-11:
> `.` is shown by the file explorer because `lv_fs_dir_read` finds it.

Sorry, I was incorrect. It actually gets added regardless of whether `lv_fs_dir_read` finds it, apparently.

liamHowatt, 2024-12-03:
Closing as #7270 was merged.

closed 2024-12-03 by liamHowatt