lvgl/lvgl #7164: We really need to get the drawing API documented.

kdschlosser, 2024-10-28:
### Introduce the problem

There is literally zero information on how to use the drawing API and layers when using the canvas widget. It only tell you how to fill the background color and that's it. This needs to be documented explaining how to use every single aspect of the drawing API. from outlines to shadows to text and the gradients as well. NONE of the drawing API is documented at all and that is crippling for anyone looking to make some kind of a custom widget using the canvas.

Now I also know there is a way to make a custom widget using the drawing events but that once again is not documented on how to do that either.

These are 2 things that should be covered in great detail as it is something that developers would use to create new widgets and features that might get submitted to be added to LVGL. but without the documentation on how to use those portions of LVGL that is not going to happen. widgets are something that can be turned on and off in LVGL so adding more of them isn't going to increase the compiled code size unless the user has them turned on and is using them, so there is no harm to having a large number of them a user is able to choose from.

lizhaoming634, 2024-10-28:
yes, I agree with your viewpoint. I also hope there is a detailed tutorial for custom widgets.

kdschlosser, 2024-10-28:
@Linjieqiang 

The reason why I have opened this thread is because I don't know how to use it which is why I have gone to the documentation to only find there is next to none on how to do it. I have spent several hours trying to get the gradients to render without any luck doing so. The program comes to a grinding halt when creating the gradient. It just sits there doing nothing.

Without the documentation the amount of time that would need to be invested to figure it out by trial and error is not going to be something that a user should have to go through.

Now seeing as how you made that comment I am guessing that you know how to use it and perhaps instead of spending the time to make those kinds of comments you correct the issue by writing the documentation for it.

lizhaoming634, 2024-10-28:
@kdschlosser 
Have you tried this example before？
https://docs.lvgl.io/master/details/widgets/canvas.html#drawing-on-the-canvas-and-rotate

lv_draw_rect_dsc_t rect_dsc;
  lv_draw_rect_dsc_init(&rect_dsc);
  rect_dsc.radius = 10;
  rect_dsc.bg_opa = LV_OPA_COVER;
  rect_dsc.bg_grad.dir = LV_GRAD_DIR_VER;
  rect_dsc.bg_grad.stops[0].color = lv_palette_main(LV_PALETTE_RED);
  rect_dsc.bg_grad.stops[0].opa = LV_OPA_100;
  rect_dsc.bg_grad.stops[1].color = lv_palette_main(LV_PALETTE_BLUE);
  rect_dsc.bg_grad.stops[1].opa = LV_OPA_50;
  rect_dsc.border_width = 2;
  rect_dsc.border_opa = LV_OPA_90;
  rect_dsc.border_color = lv_color_white();
  rect_dsc.shadow_width = 5;
  rect_dsc.shadow_offset_x = 5;
  rect_dsc.shadow_offset_y = 5;

vwheeler63, 2024-10-28:
Hi, Gábor!  ( @kisvegabor )

Kevin [astutely](https://www.oxfordlearnersdictionaries.com/us/definition/english/astute) mentions the need for documentation about how to create new types of Widgets.  Given my past (a year ago, creating custom `gauge` to more closely mimic behaviors of gauges on automotive dashboards + my study of the inheritance mechanism as part of that), I was toying with the idea of possibly addressing this need....   But first I have a question:  is the inheritance mechanism fairly stable at this point in time?

Kind regards,
Vic

kisvegabor, 2024-10-28:
@kdschlosser 
What level of details would be required for to cover the rendering API? Would it be required to document all the fields? (Some of them are quite trivial (`bg_color` is the color of the background, but others are more complex, e.g. `text_local`)).  So what kind of docs would be the most useful?
- More API comments in the structs and enums
- Examples (FYI, the canvas examples show case all the draw task types, but obviously not all the possible combinations)
- General description of the philosophy (see [here](https://docs.lvgl.io/master/details/main-components/draw.html))

The gradients would really require a good documentation. 

Most of the these `draw_dsc_t` properties are hugely overlapping with style properties which makes it harder to document them correctly, in a non-redundant way. 

@vwheeler63 
> But first I have a question: is the design of the inheritance mechanism fairly stable at this point in time?

Yes, there is no plan to change it.

kdschlosser, 2024-10-28:
OK so as an example...

lv_draw_triangle_dsc_t you can set 
opacity
color
gradient

Since I am looking at this I am wondering as to why the same drawing parameters that are seen in a rectangle are not available for the triangle. I would have thought there should be.

with `lv_draw_rect_dsc_t` what does `shadow_width` and `shadow_spread` do. they almost seem to be redundant in purpose seeing as how the wider the shadow is the more it is going to be spread out no? shadow_offset_x and shadow_offset_y are offset from where? the center of the object? the top left corner? how do you set the direction of the shadow, meaning if I wanted the light source to come from the top right so the shadow would be at the bottom left. Or if I wanted to have the light source coming from the bottom right so the shadow is at the top left. How is that done?

Is there no ability to set an outline only to specific sides like the border is able to have done to it?

Here is a big one. bg_image_recolor.. Now what on earth would you use that for? Me personally I have used the recolor feature for images that were used for for complicated shadowing but I wanted to provide the ability to change the color so I made the shadow using white and alpha and I could recolor the image and set the background color of an object behind it to get the desired color changes without LVGL needing to render a gradient..

see this example. 
https://www.youtube.com/watch?v=3Knze9cjU-0

giving use cases for the different bits is a huge part of a person being able to understand the purpose to something. To my point, why is there an outline and also a border? that seems off to have. how do they differ from one another?

what is `lv_draw_box_shadow_dsc_t`, `lv_draw_border_dsc_t` and `lv_draw_fill_dsc_t`?? There are no examples of what these are for.

This documentation is a wee but skinny on useful information'

```c
/**
 * Initialize a fill draw descriptor.
 * @param dsc       pointer to a draw descriptor
 */
```

You also have the ability to draw masks but the most important part of it which is `lv_draw_mask_rect_dsc_t` is in the private scope. No documentation on how to use that at all. 

`lv_draw_line_dsc_t`, what is `blend_mode`?? the docstring for `raw_end` doesn't really make any sense. I am guessing this has top do with seaming 2 lines together that are of different colors possibly? IDK.

There is nothing in the way of documentation for `lv_draw_label_dsc_t` and a lot of the naming you can kind of guess but not really.. Things like `sel_start` and `sel_end`. Do I enter the character number to start and end at? or do I enter a pixel number? It takes a uint32_t data type for those fields so I am leaning to it needing to be filled with the start and stop pixel positions and if that is the case then how do I go about collecting that information? With text you start and end the selection on a character so to convert that to pixels I would need to be able to calculate the positions on the boundaries of the characters and because not all characters take up the same space it would be a pretty complex bit of code no? My assumption to it being a pixel number is because who would have 4,294,967,295 characters all in a single label running on a micro controller?? That's a pretty large number and for some reason I don't think it would perform very well having it load that much text in a single go.  what is `decor` and `hint`??

`lv_draw_image_dsc_t` clip_radius, blend_mode, pivot, skew_y, skew_x are a few that could be explained better than what they are.

That's just a quick look over just the structures. Lots of things that are undocumented, have no examples and no use cases.

kdschlosser, 2024-10-28:
I think as far as the drawing API goes it would be a good idea to have a flow chart of the process. When I hear the term layer when drawing think if being able to change blend modes and opacity at the layer level. I think of being able to draw multiple things to a single layer and have that blend mode and opacity effect everything that is rendered on that layer. Am I able to draw multiple things to a single layer? or can I only render a single thing to a layer? Can I hold references to the layers and use them later on?

lots and lots of questions and none of them are answered in the documentation.

I am not able to answer any of those questions because frankly I have no clue how to even use any of those things. If I knew how to use them I would submit PR's adding information to the documentation that would be helpful. But there is such a large amount iof the draw API that doesn't have any documentation I wouldn't know where to begin. I have spent hours messing around with gradients and adjusting the `frac` and it's behavior seems to be completely random. Number is too low it doesn't render anything. set the number a little higher and i don't get a gradient I get a solid color. frac I am assuming is a fraction but a fraction of what? the number of colors in the gradient? the total number of pixels the gradient spans? I have no clue.... how would I go about making this kind of a gradient? The transitions don't seem to feather together like that. they are more abrupt. 

[image]

how come I am not able to create a radial gradient with the focal point being in the center of the bounding rectangle? How come the gradient extends past the bounding rectangle on 2 of the edges where it gets cut off, it should be contained within the rectangle no?

kisvegabor, 2024-10-29:
Now it's much cleaner from where the trouble comes from. I think we should do this:
- Add more docs to the [style properties](https://docs.lvgl.io/master/details/base-widget/styles/style-properties.html). E.g. to explain what shadow spread is.
- In the [drawing docs page](https://docs.lvgl.io/master/details/main-components/draw.html):
   - Extend it with a section for each draw task types. 
   - Also add a ~5 examples for each draw_dsc type used in an event.
   - Show examples for `LV_EVENT_DRAW_TASK_ADDED`  
- Add comments for all elements of draw_dsc-s. 

I can allocate time for next week.

kdschlosser, 2024-10-29:
I only gave a few examples. There is a whole lot more that could really benefit from additional documentation or actually being documented. 

The drawing API is quite large and without documentation that's pretty detailed with use cases, examples and explanations of how things work (frac) it would be exceedingly difficult for some to learn it by trial and error. I cannot imagine the plethora of time that would need to be invested in order to learn it that way.

kdschlosser, 2024-10-29:
You don't have to go crazy with it. Get the documentation to the point where I would be able to read it and be able to get it working properly and then I can go and update the documentation to provide more information and use cases. Things like that. Add pictures showing some effects that are able to be achieved by setting things a specific way. That kind of thing.

vwheeler63, 2024-11-17:
Not stale.

kisvegabor, 2024-12-22:
Let's continue in https://github.com/lvgl/lvgl/pull/7241

closed 2024-12-22 by kisvegabor