lvgl/lvgl #7165: Parallel rendering is in fact not running in parallel the majority of the time.

kdschlosser, 2024-10-28:
### LVGL version

9.2.x

### What happened?

When using the FreeRTOS LVGL OS on the ESP32 series of MCU's the parallel rendering may or may not occur on different cores.

The reason why this occurs is because the wrapper function that is written into the ESP-IDF is not being used to create new tasks. That means the option to direct FreeRTOS as to what core the task should run on is not available to be set so FreeRTOS makes the decision as to what core to run the task on which more often than not is going to place the tasks on the same core so parallel rendering doesn't actually happen. If you look at the ESP-IDF documentation located [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_additions.html#_CPPv423xTaskCreatePinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC12TaskHandle_tK10BaseType_t)

In order to use the wrapper function you would need to include `freertos/idf_additions.h`.

The other thing that I noticed is in the documentation there exists the ability to set how many parallel tasks are running but this seems to be be unused as the number of threads that get created is dictated by the `LV_DRAW_SW_DRAW_UNIT_CNT` macro and not how many tiles are set to the display using the `lv_display_set_tile_cnt` function. The informatiomn about setting the tiles is located [here](https://docs.lvgl.io/master/details/main-components/display.html#tiled-rendering)

Unless you specify the processor core to run the rendering on FreeRTOS is going to pick whatever core hasd the lower amount of load at the time the task is created. So if there happens to be more load at the time the code is being executed in LVGL to create the tasks then the task can all end up getting piled onto the same core.

There is also no information omn setting the number of parallel draw units... If I set it to 2 which would be equal to the number of cores and the MCU has 2 cores what happens with the main task that created the 2 tasks since there are only 2 cores available does the main task end up sitting there twiddling it's thumbs waiting or is the main task doing one of the draw units work? I am guessing that the main task would be blocked until the rendering has completed by all of the draw units and that ultimately doesn't make much sense to stall the main task and also introduce additional overhead by starting a task to do the work the main task could be doing. 

as it turns out the pthread OS also suffer from the same issue as the core that the thread is running on is not being set. You need to use the `pthread_set_affinity_np` function to specify the core the thread should be running on. if it is not set the thread is going to run on whatever one the hardware OS decides to initially start the thread on based on the load at the time the threads are created. It's not going to shuttle the threads around from core to core because of memory regions being mapped to specific cores when the thread is started.

Windows will always run a thread on the same core the parent process is running on unless the `SetThreadAffinity` function is used to specifically set a different core. so the parallel rendering in Windows doesn't work at all because all threads are currently being created on the same core.

kisvegabor, 2024-10-30:
> Unless you specify the processor core to run the rendering on FreeRTOS is going to pick whatever core hasd the lower amount of load at the time the task is created.

That's very important. I didn't know it. In light of that probably we should add an `lv_thread_set_core` function. 

---

What will happen if you have 2 cores and 2 draw units?
You will have 3 threads
   - main thread: which collects the draw tasks
   - rendering threads

It might sound stupid, but the main thread usually finishes faster than the the rendering. So e.g.
- for 5 ms you have 3 thread for 2 cores (not optimal)
- for the next 15 ms of rendering you will have 2 rendering thread for 2 cores

kdschlosser, 2024-10-30:
I don't know what the cost is to start a thread on a micro controller. I have no clue how much in the way of resources it uses but as we do know resources like memory are a precious commodity on MCU's and keeping memory use as low as possible was the sole reason why you created LVGL in the first place. Don't loose sight of that primary objective you had when you first started writing LVGL. On a desktop PC having the threads overlap for the 15 or so milliseconds with one of them telling the other to do work and then it waits until that other thread finishes before it proceeds means that other thread is really not needed because the work is able to be done in the primary thread. I cannot imagine it would be that difficult to have the main thread process one of the drawing units. instead of waiting.

With respect to the CPU core issue. It is something that is an easy thing to overlook. The OS is not always the best thing to have handle the assignment because it only looks at what is happening now and not what could happen. due to the nature of an MCU typically the core that is running the main thread is going to have a higher load than the other so the "OS" is going to place both of the threads onto the other core. This negates the parallel rendering, that makes us come full circle back to the main thread not being used to process one of the draw units.  LOL...

There needs to be several functions added to the OS code in LVGL. One would be to collect the total number of cores, another to collect the core that the current thread/task is running on and the function that us used to create a new thread/task needs to have a parameter added to identify what core to run the new thread on. You need to have those things so you don't go out of bounds with the core ID, and you don't want tom create a task/thread on the same core that is running the main task if you decide to have the main task process one of the draw units.

kdschlosser, 2024-10-30:
The use of setting the number of tiles in the display to specify the number of cores to run on is a great thing to use. as a default it should automatically set that to the number of cores the CPU has and the user has the ability to override it but only to a lower number. perhaps adding a uin8_t field somewhere to store what cores a user wants to use for those threads/tasks. Simple bit masks to identify the core numbers. That would allow identifying 8 different cores which I think would be sufficient but perhaps  uint16_t would be better to allow for future use as time passes and hardware advances. If the user has not set specific cores then that same field would be used as the way to identify what cores have running tasks\threads on them.

kdschlosser, 2024-10-30:
> It might sound stupid, but the main thread usually finishes faster than the the rendering. So e.g.

This is not 100% true. It depends. A core is only able to run one thing at a time. so until the main thread yields to allow another thread that is running on the same core to start working that other thread is not going to do anything at all except sit there and wait. Threads running on the same core do not run in parallel. FreeRTOS and similar type OS's don't switch contexts automatically. They have to be instructed to using a mutex or other locking type mechanism or by telling the OS it is yielding.

kisvegabor, 2024-11-14:
> This is not 100% true. It depends. A core is only able to run one thing at a time. so until the main thread yields to allow another thread that is running on the same core to start working that other thread is not going to do anything at all except sit there and wait. 

Yes, but imagine this:
- collecting draw tasks takes 5 ms
- rendering on 1 core takes 20ms (let's assume that rendering can be done fully parallel)

So:
- 0-5ms: 
  - core 1: all draw tasks are collected, main task is idle
  - core 2: 5ms is rendered from the 25ms, meaning the rendering is 20% ready
- 5-10ms:
  - core 1: can do 5ms rendering (+20% progress) 
  - core 2: can do 5ms rendering (+20% progress) 
- 10-15ms:
  - core 1: can do 5ms rendering (+20% progress) 
  - core 2: can do 5ms rendering (+20% progress) 
  
And we are done. 

---

For core count `lv_draw_sw_set_draw_unit_count(int x)` could work too. It creates new tasks or deletes some if needed.

kisvegabor, 2024-11-17:
> You are yielding blindly. you do not have any idea what task FreeRTOS is going to start to run when you yield the main thread. 

Actually that's why there are priorities. SW render threads are created wit `LV_THREAD_PRIO_HIGH`. If something is really that important that its thread is created with a higher priority then it's correct to run it instead of rendering.

kdschlosser, 2024-11-17:
unless the main thread is already set to priority max and the new thread is set to priority max and if there are other threads/tasks that are set to priority max by say an MCU SDK or some other library. You cannot rely on the priority settings to delegate what is going to be run first because if a user is using any external libraries or SDK's they might not have control over the priorities that are being set for tasks in those libraries or SDK's

kdschlosser, 2024-11-20:
> None of the libs should grab the whole CPU time saying it's the most important.

in an ideal world it would be that way. But we both know it is not an ideal world and people do things like this.

And if you have 2 tasks that have the same priority the context does switch between them as well as lesser priority tasks. any time a task yields while waiting on a mutex or semaphore the context will switch. The point is why have time wasted switching contexts when the work is able to be done right there and then. no additional resource use for a separate task to do what can be done using the task that is already in context.

kdschlosser, 2024-11-22:
I honestly don't even know how the code doesn't deadlock.

see here..
https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/lv_draw.c#L190

`lv_draw_dispatch` gets called and that then makes a call to `lv_draw_wait_for_finish`

https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/lv_draw.c#L166

and the `lv_draw_wait_for_finish` function is a spinning wheels function. It doesn't even yield to other tasks. The only renderer that uses the `wait_for_finish_cb` is vglite. It should be used by all renderers and the function that gets set to it should be dependent on the LV OS that is being used. This way the proper mechanics can be put into place to properly stall the main thread allowing it to yield.

kdschlosser, 2024-11-22:
so the main task actually does nothing at all until those draw tasks are finished.

kdschlosser, 2024-11-22:
The other thing is it looks like the draw tasks are being created over and over again. are the actual threads/tasks also being created over and over again? If they are would it be best to create them a single time at application start and use mutexes to handle when they should render and to have them yield when there is nothing to render? This would reduce CPU time needed to create the tasks/threads as well as reduce the heap fragmentation that would occur as a result.

kisvegabor, 2024-11-22:
> I honestly don't even know how the code doesn't deadlock.

The main thread yields here.

https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/lv_draw.c#L255

> The other thing is it looks like the draw tasks are being created over and over again. are the actual threads/tasks also being created over and over again?

Where?

closed 2024-12-17 by lvgl-bot