lvgl/lvgl #7176: Unable to draw outside the non-rotated bounds of `lv_layer_top()` with software rotation

the-programmer, 2024-10-29:
### LVGL version

v8.4.0

### What happened?

Since my current hardware does not support hardware rotation of the screen I have to use the software rotation (thanks for adding that). However, when trying to draw the keyboard on the top layer I am unable to get that working. (Additional reference [here](https://forum.lvgl.io/t/using-the-keyboard-in-the-menu/18234))

As a test I made the code below. The results are as follows:
```c
    disp_drv.sw_rotate = 1;
    disp_drv.rotated = LV_DISP_ROT_NONE;
```
[image]

```c
    disp_drv.sw_rotate = 1;
    disp_drv.rotated = LV_DISP_ROT_90;
```
[image]

Note that all the other functionality works as intended. Only keyboards on the top layer fail to work.

### How to reproduce?

```c
//Blue
static lv_point_t line_points1[] = { {5, 5}, {480-5, 5}, {480-5, 800-5}, {5, 800-5}, {5, 5} };
//Red
static lv_point_t line_points2[] = { {10, 10}, {800-10, 10}, {800-10, 480-10}, {10, 480-10}, {10, 10} };

void main()
{
    lv_obj_t * mainScreen = lv_obj_create(lv_scr_act());
    lv_obj_t * toplayer2 = lv_layer_top();

    /*Create style*/
    static lv_style_t style_line1;
    lv_style_init(&style_line1);
    lv_style_set_line_width(&style_line1, 8);
    lv_style_set_line_color(&style_line1, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_rounded(&style_line1, true);

    /*Create style*/
    static lv_style_t style_line2;
    lv_style_init(&style_line2);
    lv_style_set_line_width(&style_line2, 8);
    lv_style_set_line_color(&style_line2, lv_palette_main(LV_PALETTE_RED));
    lv_style_set_line_rounded(&style_line2, true);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(toplayer2);
    lv_line_set_points(line1, line_points1, 5);     /*Set the points*/
    lv_obj_add_style(line1, &style_line1, 0);
    lv_obj_set_pos(line1, 0, 0);

    /*Create a line and apply the new style*/
    lv_obj_t * line2;
    line2 = lv_line_create(toplayer2);
    lv_line_set_points(line2, line_points2, 5);     /*Set the points*/
    lv_obj_add_style(line2, &style_line2, 0);
    lv_obj_set_pos(line2, 0, 0);
}
```

the-programmer, 2024-10-29:
I created a notebooklm with all the relevant sources (tip: although you can't upload .c files, raw github links work fine)

With that I was pointed to the `LV_OBJ_FLAG_OVERFLOW_VISIBLE` flag. So I added that.

```c
lv_obj_t * toplayer2 = lv_layer_top();
lv_obj_add_flag(toplayer2, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
```

Now I at least have a work-around but the code still needs to be checked.

kisvegabor, 2024-11-14:
Hi,

Sorry for the late reply. Is the problem related to the top_layer and does it work if the line are on the `mainScreen`?

the-programmer, 2024-12-03:
The issue is still relevant.

This week I should be able to start on some more experiments with this. However I am unable to remove the label.

the-programmer, 2024-12-05:
Ok, so I have time to start some tests now. 
First off, I realized that my original code contained a bug for creating the `mainScreen`. This should just be `lv_obj_t * mainScreen = lv_scr_act();`. (This has been updated)

If I use `mainscreen` or `toplayer2` as parent when creating the line, the result is the same.
So, this is the original (yes, now as propper screenshot)
[image]

And if I rotate everything get the following (the screenshot is rotated after capture)
[image]

the-programmer, 2024-12-06:
So, for the keyboard I found a workaround.
The steps I took.

First I create a screen where the `LV_OBJ_FLAG_OVERFLOW_VISIBLE` flag is set. (this is later loaded by `lv_scr_load_anim()`)
```c
    lv_obj_t * screen = lv_obj_create(NULL);

    lv_obj_set_size(screen, lv_pct(100), lv_pct(100));
    lv_obj_set_layout(screen, LV_LAYOUT_FLEX);
    lv_obj_set_flex_flow(screen, LV_FLEX_FLOW_COLUMN);
    lv_obj_add_flag(screen, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
```

Next I create the keyboard (when it's required) with the floating flag set.
```c
        lv_obj_t * parent = GetRootParent(object);

        _keyboard = lv_keyboard_create(parent);
        lv_obj_set_size(_keyboard, lv_pct(100), lv_pct(50));
        lv_obj_align(_keyboard, LV_ALIGN_BOTTOM_MID, 0, 0);
        lv_obj_add_flag(_keyboard, LV_OBJ_FLAG_FLOATING);
        lv_obj_move_foreground(_keyboard);
```

This way I have a working keyboard both in landscape and portrait.

As an additional reference, here is my `GetRootParent()` method
```c
lv_obj_t * GetRootParent(lv_obj_t * current)
{
    lv_obj_t * parent;
    while(1)
    {
        parent = lv_obj_get_parent(current);
        if(parent == NULL)
            return current;
        
        current = parent;
    }
}
```

the-programmer, 2024-12-22:
Well, here is the required comment to keep the issue open.

Is anything required from my side?

kisvegabor, 2024-12-22:
I'm sorry, but I didn't have time to look into it earlier. Happy to hear that it's working now. 

I think `lv_obj_add_flag(screen, LV_OBJ_FLAG_OVERFLOW_VISIBLE);` is not required. 

It seems `GetRootParent` is the same as `lv_obj_get_screen`.

closed 2025-01-18 by lvgl-bot