lvgl/lvgl #7178: lv_display_set_rotation and ST LTDC driver - rotation incorrect/not working.

LoyalServant, 2024-10-29:
### LVGL version

9.3

### What happened?

Code I am working with is below. The screen is a "bar" type 320x820 screen found on aliexpress with ST7701 driver, and adafruit sells one like it.

Mostly a question of whether ot not LVGL supports rotation with the LTDC peripheral/driver on the stm32?
I am working with the STM32U5 series.

The issue:
When the screen is rotated by 90 degrees, the spinner is not rotated, and its shoved all the way to what would be the top of the display, and the bottom 2/3 are black.
When rotating 180, the screen is never rotated. Its the same as if it was at 0.
When rotating to 270 - the screen appears the same as it does at 90.
I will try to take some pictures to illustrate this behaviour.

### How to reproduce?

```
lv_init();
uint32_t my_ltdc_layer_index = 0; /* typically 0 or 1 */
lv_display_t * disp = lv_st_ltdc_create_direct(framebuffer, framebuffer1, my_ltdc_layer_index);
lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);

lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x003a57), LV_PART_MAIN);
lv_obj_set_style_text_color(lv_screen_active(), lv_color_hex(0xffffff), LV_PART_MAIN);

/* Create a spinner */
lv_obj_t * spinner = lv_spinner_create(lv_screen_active());
lv_obj_set_size(spinner, 128, 128);
lv_obj_align(spinner, LV_ALIGN_CENTER, 0, 0);
```
```
#define DISPLAY_WIDTH  320
#define DISPLAY_HEIGHT 820
__attribute__((aligned(16))) uint16_t framebuffer[DISPLAY_WIDTH * DISPLAY_HEIGHT];
__attribute__((aligned(16))) uint16_t framebuffer1[DISPLAY_WIDTH * DISPLAY_HEIGHT];
```
```
void MX_LTDC_Init(void)
{

  /* USER CODE BEGIN LTDC_Init 0 */

  /* USER CODE END LTDC_Init 0 */

  LTDC_LayerCfgTypeDef pLayerCfg = {0};

  /* USER CODE BEGIN LTDC_Init 1 */

  /* USER CODE END LTDC_Init 1 */
  hltdc.Instance = LTDC;
  hltdc.Init.HSPolarity = LTDC_HSPOLARITY_AL;
  hltdc.Init.VSPolarity = LTDC_VSPOLARITY_AL;
  hltdc.Init.DEPolarity = LTDC_DEPOLARITY_AL;
  hltdc.Init.PCPolarity = LTDC_PCPOLARITY_IPC;
  hltdc.Init.HorizontalSync = 0;
  hltdc.Init.VerticalSync = 0;
  hltdc.Init.AccumulatedHBP = 30;
  hltdc.Init.AccumulatedVBP = 16;
  hltdc.Init.AccumulatedActiveW = 350;
  hltdc.Init.AccumulatedActiveH = 836;
  hltdc.Init.TotalWidth = 356;
  hltdc.Init.TotalHeigh = 837;
  hltdc.Init.Backcolor.Blue = 0;
  hltdc.Init.Backcolor.Green = 0;
  hltdc.Init.Backcolor.Red = 0;
  if (HAL_LTDC_Init(&hltdc) != HAL_OK)
  {
    Error_Handler();
  }
  pLayerCfg.WindowX0 = 0;
  pLayerCfg.WindowX1 = 320;
  pLayerCfg.WindowY0 = 0;
  pLayerCfg.WindowY1 = 820;
  pLayerCfg.PixelFormat = LTDC_PIXEL_FORMAT_RGB565;
  pLayerCfg.Alpha = 255;
  pLayerCfg.Alpha0 = 0;
  pLayerCfg.BlendingFactor1 = LTDC_BLENDING_FACTOR1_CA;
  pLayerCfg.BlendingFactor2 = LTDC_BLENDING_FACTOR2_CA;
  pLayerCfg.FBStartAdress = (uint32_t)framebuffer;
  pLayerCfg.ImageWidth = 320;
  pLayerCfg.ImageHeight = 820;
  pLayerCfg.Backcolor.Blue = 0;
  pLayerCfg.Backcolor.Green = 0;
  pLayerCfg.Backcolor.Red = 0;
  if (HAL_LTDC_ConfigLayer(&hltdc, &pLayerCfg, 0) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN LTDC_Init 2 */

  /* USER CODE END LTDC_Init 2 */

}

liamHowatt, 2024-10-30:
No conscious effort was made to make rotation work with this driver. Good catch. I'll take a look next week :slightly_smiling_face:

liamHowatt, 2024-11-07:
A PR is open @LoyalServant. Please try it out: #7254

Fyi, you must use partial mode for rotation to work, i.e. `lv_st_ltdc_create_partial`.

LoyalServant, 2024-11-07:
Working - mostly.
There is some distortion on the top (either left or right side depending on rotation) when rotated 90 or 270.
I dropped the specs to the display in here (in case they help)  but I have used it with the ST framework and it's working fine there.
it's also fine when using lv_st_ltdc_create_direct() - just not rotated.

A picture is worth a thousand words - hope that helps.

[Specification+TL032FWV01-I1440A（3.2+320X820+IPS+RGB)+(2).pdf](https://github.com/user-attachments/files/17664120/Specification%2BTL032FWV01-I1440A.3.2%2B320X820%2BIPS%2BRGB.%2B.2.pdf)
[image]

liamHowatt, 2024-11-12:
I'm having some trouble reproducing it. Are you using NemaGFX?

Are these correct?

```
  hltdc.Init.AccumulatedHBP = 30;
  hltdc.Init.AccumulatedVBP = 16;
  hltdc.Init.AccumulatedActiveW = 350;
  hltdc.Init.AccumulatedActiveH = 836;
  hltdc.Init.TotalWidth = 356;
  hltdc.Init.TotalHeigh = 837;
```

LoyalServant, 2024-11-12:
Not using NemaGFX. It's just LTDC. No DMA2D, etc.
The porch values are worth exploring. I had to infer those values from whatever information I could find and some rough guesses in the datasheet based on the timings.
When it displayed correctly in the correct orientation I figured I had it.
My only thing that casts doubt is that I was also fiddling with TouchGFX. I can do a 90 degree rotation only, not 270, and it displays correctly in either orientation.
At some point today I can probably pull that board out and try moving those values back and forth a little.

LoyalServant, 2024-11-12:
Taken another look.
No amount of fiddling with timings resolves this or even changes its behaviour.
It breaks the display, yes - but the same corruption appears in some way or form when rotated.

What i also noticed and it's not visible in the image that that there are 1 pixel lines missing from the word "test"
Also, the pixels on the top of the display move and change colour with the spinner rotation.

This only appears when using lv_st_ltdc_create_partial();
when using lv_st_ltdc_create_direct() everything is correct - just not rotated.

I could be wrong here but I don't think this is a problem with LTDC.
I created a test pattern that also displays correctly - but it writes directly to the LTDC framebuffer:

```void LTDC_DisplayAlignmentScreen(void) {

	// horizontal line down the center.
	for (int x = 0; x < DISPLAY_WIDTH; x++) {
	    framebuffer[(DISPLAY_HEIGHT / 2) * DISPLAY_WIDTH + x] = 0xFFFF;
	}

	// vertical line down the center
	for (int y = 0; y < DISPLAY_HEIGHT; y++) {
	    framebuffer[y * DISPLAY_WIDTH + (DISPLAY_WIDTH / 2)] = 0xFFFF;
	}

	// borders - top/bottom
	for (int x = 0; x < DISPLAY_WIDTH; x++) {
	    framebuffer[x] = 0xF800;
	    framebuffer[(DISPLAY_HEIGHT - 1) * DISPLAY_WIDTH + x] = 0xF800;
	}

	// borders - left/right
	for (int y = 0; y < DISPLAY_HEIGHT; y++) {
	    framebuffer[y * DISPLAY_WIDTH] = 0xF800;
	    framebuffer[y * DISPLAY_WIDTH + (DISPLAY_WIDTH - 1)] = 0xF800;
	}

	// red
	for (int y = 0; y < DISPLAY_HEIGHT; y++) {
	    int x = (y * DISPLAY_WIDTH) / DISPLAY_HEIGHT;
	    framebuffer[y * DISPLAY_WIDTH + x] = 0xF800;
	}

	// blue
	for (int y = 0; y < DISPLAY_HEIGHT; y++) {
	    int x = DISPLAY_WIDTH - 1 - (y * DISPLAY_WIDTH) / DISPLAY_HEIGHT;
	    framebuffer[y * DISPLAY_WIDTH + x] = 0x001F;
	}

}
```

kisvegabor, 2024-11-13:
Could you show how you create an `lv_display` and how you declare the buffers?

LoyalServant, 2024-11-14:
@kisvegabor 
Sorry about the late reply.

It's basically the same code as above except with the one line being changed.
This is what that code looks like at the moment:

```
uint32_t my_ltdc_layer_index = 0; /* typically 0 or 1 */
//lv_display_t * disp = lv_st_ltdc_create_direct(framebuffer, framebuffer1, my_ltdc_layer_index);
lv_display_t * disp = lv_st_ltdc_create_partial(framebuffer, framebuffer1, 65536, my_ltdc_layer_index);
lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);
```

kisvegabor, 2024-11-17:
Just to make sure, in
```c
lv_display_t * disp = lv_st_ltdc_create_partial(framebuffer, framebuffer1, 65536, my_ltdc_layer_index);
```
`framebuffer` and `framebuffer1` are not the frame buffers that are used by LTDC, right? So you should have:
- a frame buffer for the whole display, LTDC reads from here
- 1 or 2 independent buffers for rendering (these are passed to `lv_st_ltdc_create_partial`)

So it should look like:
```c
static uint8_t buf1[64*10124];
/*When rotated we can't make us of 2 buffers*/
lv_display_t * disp = lv_st_ltdc_create_partial(buf1, NULL, sizeof(buf1), my_ltdc_layer_index);

```

LoyalServant, 2024-11-20:
no, 
'framebuffer' is the one used by LTDC. 'framebuffer1' is an additional framebuffer I made for lvgl, but it is the same size.
I will yank the second one out and give it another try.
I see the comment there that you made regarding not being able to use two buffers - was not aware of that.

LoyalServant, 2024-11-20:
Ok, so..
`
lv_display_t * disp = lv_st_ltdc_create_partial(framebuffer, framebuffer1, 65536, my_ltdc_layer_index);
`

to

`
lv_display_t * disp = lv_st_ltdc_create_partial(framebuffer, NULL, sizeof(framebuffer), my_ltdc_layer_index);
`

Makes things much worse. There are artifacts over much of the screen doing it this way.
It's not the removal of the second framebuffer ('framebuffer1') but changing the buf_size.
I can get back to the original issue with 65535, instead of sizeof(framebuffer).
So, the size of the framebuffer should be 524800, and I did try that instead of using sizeof() in case there was something going wrong there but that didn't change anything.

What I have noticed is that the second framebuffer being passed in has _absolutely no influence_ on the issue, whether it's specified or not, but buf_size absolutely does make a difference. A buffer size of like '2050' almost eliminates the issue. (524800/256)

So, noticing that a smaller buffer seems to clear it up, i tried to get cute and use 1280 (320*2*2) but i got nothing on the screen.
1920 was the lowest I was able to get working.(320*3*2)

Not knowing the internals - I don't know if this helps. Just that 1920 for buf_size results in what looks like only 2 rows of moving pixels at the top of the screen that coincides with the animation of the spinner.
The other side effect is that performance takes a massive hit - but I get that with such a small buffer.

I wondered if this was something possibly with ICACHE on the stm32, so I disabled that but it made no difference.

kisvegabor, 2024-11-28:
If `framebuffer1` is the one you have created for LVGL then you should do
```c
lv_display_t * disp = lv_st_ltdc_create_partial(framebuffer1, NULL, sizeof(framebuffer), my_ltdc_layer_index);
```

It means LVGL will render to `framebuffer1` and it will be rotated to `framebuffer`.

closed 2024-12-02 by FASTSHIFT

userganeshr, 2025-03-10:
Hi @kisvegabor ,
     I am also facing the issue while try to rotate the screen 90 degree by using stm32 ltdc and lvgl. I have created display using lvgl partial method like below,

__attribute__((section(".sdram_data"), aligned(32))) static lv_color_t buf_1[800 * 480];
__attribute__((section(".sdram_data"), aligned(32))) static lv_color_t buf_2[800 * 480];

  lv_display_t * disp = lv_st_ltdc_create_partial(buf_2, NULL, sizeof(buf_1), 0);
  lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);     /* set landscape orientation */

[image]

PR #7254 (2024-12-02): feat(drivers/st_ltdc): add rotation support to LTDC driver
Fixes #7178

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`ast [... cut]