lvgl/lvgl #7179: ARC Events Documentation Out of Date?

vwheeler63, 2024-10-29:
### LVGL version

v9.3

### What happened?

On the documentation page  https://docs.lvgl.io/master/details/widgets/arc.html#events  all the events listed after `LV_EVENT_VALUE_CHANGED` appear to have been removed from the Widget's code, in which case, I should remove them from the documentation.  @kisvegabor can you confirm?

[image]

### How to reproduce?

n/a

kisvegabor, 2024-10-30:
I confirm that `LV_DRAW_PART_BEGIN/END` are leftovers from v8 and doesn't exist in v9.

vwheeler63, 2024-10-30:
> I confirm that `LV_DRAW_PART_BEGIN/END` are leftovers from v8 and doesn't exist in v9.

Thank you!  I have removed them and will include in batch 4.

vwheeler63, 2024-10-30:
> Thank you! I have removed them and will include in batch 4.

cc:  @kisvegabor 

Correction -- I have made these changes here locally already, but I will wait until the first `arc.rst` changes are merged before re-submitting so I don't cause a conflict.

vwheeler63, 2024-11-17:
Will be submitting a "wrap-up" PR on all the widgets (Arc included) once the existing PRs are committed, to avoid merge conflicts.

kisvegabor, 2024-12-09:
@vwheeler63 Is the fix already merged?

vwheeler63, 2024-12-09:
> @vwheeler63 Is the fix already merged?

Indeed it is!  PR #7405 took care of it.

closed 2024-12-09 by vwheeler63