lvgl/lvgl #7182: Placeholder Text Not Centered For lv_textarea_t

stachurr, 2024-10-29:
### LVGL version

v9.2.0

### What happened?

Setting a placeholder for a `lv_textarea_t` doesn't center the text. Note however, the normal text IS centered -- just not the placeholder text.
I also tried `lv_obj_set_style_align(m_textarea, LV_ALIGN_CENTER, LV_PART_TEXTAREA_PLACEHOLDER);` with no luck.
In all screenshots attached, I focused the textarea so the centered cursor would render, giving more visual cues.
[image]

Here are some screenshots using different fonts (1 built-in, 1 custom):
[image]
[image]

For now, I'm appending 4 spaces to the placeholder which shifts the text left enough to look centered...
[image]
[image]

### How to reproduce?

```
lv_obj_t* screen = lv_screen_active();
lv_obj_t* textarea = lv_textarea_create(screen);
lv_obj_set_size(textarea, 200, 100);
lv_textarea_set_placeholder_text(textarea, "Enter Something...");
//lv_obj_set_style_align(textarea, LV_ALIGN_CENTER, LV_PART_TEXTAREA_PLACEHOLDER);
```

liamHowatt, 2024-10-30:
It's a bit of an edge case. Would you prefer the cursor to be at the left edge of the placeholder text while the placeholder text is visible?

kisvegabor, 2024-10-30:
You should use `LV_TEXT_ALIGN_CENTER` instead of `LV_ALIGN_CENTER`.
```c
lv_obj_set_style_text_align(textarea, LV_TEXT_ALIGN_CENTER, LV_PART_TEXTAREA_PLACEHOLDER);
```

stachurr, 2024-10-30:
Hi @kisvegabor, I have already tried this and it does not work.

kisvegabor, 2024-10-30:
There was a related fix in `master`. Please use the latest commit from the `master` branch.

closed 2024-11-05 by stachurr