lvgl/lvgl #7184: chart widget break. when one series have one data, it will be break!!

xinsuinizhuan, 2024-10-30:
### LVGL version

all version

### What happened?

when one series have one data, it will be break! Code here break:
[image]
it should be change as:
[image]

there are so many division, should judge the divisor is zero, in code!!!

liamHowatt, 2024-10-30:
I encountered this divide-by-zero bug as well. Please open a PR.

kissa96, 2024-10-30:
Hi @xinsuinizhuan, do you want to open a PR, or should I handle it?

xinsuinizhuan, 2024-11-01:
you handle it!

xinsuinizhuan, 2024-11-01:
https://github.com/lvgl/lvgl/issues/7007 this is also the divide-by-zero bug.

liamHowatt, 2024-11-05:
Closing as #7211 has been merged.